letscontrolit/ESPEasy #2472: esp32test P016_IR plugin compilation platformio failed

dony71, 2019-05-16:
I need to include P016_IR so I comment out below in define_plugin_sets.h
//    #define USES_P016   // IR
However compilation in platformio reporting error below
Anybody can guide me how to compile it successfully?

Compiling .pioenvs\esp32test_1M8_partition\src\ESPEasy.ino.cpp.o
F:/GitHub/ESPEasy-mega-20190226/src/_P016_IR.ino:11:20: fatal error: IRrecv.h: No such file or directory

TD-er, 2019-05-16:
Not sure if the IR library is usable on ESP32.
If plugins are not yet included on ESP32, while they are present  for ESP8266 in the mega branch, then it is disabled for reason of incompatibility.

dony71, 2019-05-20:
I manage to get IR remote compile successfully with this library below which should support for ESP32
https://github.com/ExploreEmbedded/Arduino-IRremote
I also have to adjust P016_IR plugin code to use function in that library
However once plugin being executed, I have error below
It seems cache is disable in ESP32 code. How to enable this in the code?
==============================================================
IR Init
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IlleGuru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (loopP        4)
Guru Meditation Error: Core  1 panic'ed (Guru Meditation Error: Core  1 panic'ed (Guru Meditation Error: Core  1 panic'ed (Unknown reason)
Guru Meditation Error: Core  1 panic'ed (Double exception)

TD-er, 2019-05-20:
That library you linked is about 2 years old, so I guess it is not really compatible (anymore?) with the current ESP32 core libraries.

tonhuisman, 2022-04-18:
This has been resolved, so issue can be closed.

closed 2022-04-18 by TD-er