letscontrolit/ESPEasy #2475: Implement proper debounce (general GPIO handler)

TD-er, 2019-05-21:
It as fixed in here: https://github.com/letscontrolit/ESPEasy/pull/2451
But that fix will now (temporarily) be reverted in favor of the fix of interrupt calls with the IRAM attribute set.

Previous discussions on the debounce:
- https://github.com/letscontrolit/ESPEasy/issues/2366
- https://github.com/letscontrolit/ESPEasy/issues/2463
- https://github.com/letscontrolit/ESPEasy/pull/1880
- https://github.com/letscontrolit/ESPEasy/issues/2034
- https://github.com/letscontrolit/ESPEasy/pull/1942

See also #1620

Marty, 2019-12-23:
Hi @TD-er, do you have any idea when this will be re-implemented? I just ran into this issue myself and am wondering whether working around it in software (or learning to compile the firmware with custom plugins) is worth the effort.

TD-er, 2019-12-23:
Last months I was mainly occupied with the stability issues.
@giig1967g is working on moving GPIO handling into the code, so this can also be a part of that.

giig1967g, 2019-12-24:
@Marty:
what is exactly your request?

@TD-er 
GPIO software debounce is implemented in the single plugin code, in the PLUGIN_WRITE section, as it depends on the switch settings.

Marty, 2019-12-24:
I'm facing the exact same issue as @oerweber described [here](https://github.com/letscontrolit/ESPEasy/issues/2366#issuecomment-471934945) in #2366.
If I understood correctly, his fix broke (or conflicted with) something and was reverted to be implemented in the core at a later stage with a reference to this issue here.

giig1967g, 2019-12-24:
@Marty 
I understand that you are talking about P003, right?
My development on debounce was for PLUGIN 001, 009, 019 (switch, mcp and pcf extenders).

I actually didn't touch P003 regarding debounce.

Marty, 2019-12-24:
@giig1967g 
I just started with espeasy a week ago, so I do not get all the technical terms yet I'm afraid.
As far as I understood the discussion over the debouncing, @oerweber fixed the debouncing issue in #2366 for the P003 (P222 in playground) but a more general solution should be implemented so that not every plugin has to it on its own (?). It is my understanding that this issue here is meant for this said general solution.

Please correct my if I understood this wrong and this issue here has nothing to do with reliability of the P003.

TD-er, 2019-12-25:
If you can implement a de-bouncer in hardware (e.g. add some strong pull-up and a capacitor and not too long wires too) then that is by far the preferred way as ESP8266 will never be able to do a perfect de-bouncer in software.
Still I think there is a lot to improve on the software de-bouncer as we have it now.
Also the GPIO handler changes we have in mind are still a work-in-progress.

@Marty 
What terms are still unclear? Maybe we can help in making them more clear to you?

Marty, 2020-01-13:
Hi @TD-er and thanks for your kind response. I tried to solve this issue with hardware and put in an inverted schmitt trigger with a 100nf capacitor as he did [here](https://nachbelichtet.com/2019/06/23/so-funktioniert-der-tcrt5000-mit-espeasy-als-pulse-counter/), but that did not solve it for me. What worked is [the plugin](https://github.com/oerweber/ESPEasyPluginPlayground/blob/master/_P222_DebouncedCounter.ino) from @oerweber. I now get correct results :)

oerweber, 2020-01-16:
> If you can implement a de-bouncer in hardware (e.g. add some strong pull-up and a capacitor and not too long wires too) then that is by far the preferred way as ESP8266 will never be able to do a perfect de-bouncer in software.

Even every low performance 8 bit controller is able to do a proper software debounce. The problem is mostly, that the wrong software approach is used. Like delay loops etc. One (there may be others) low cost (no hardware, less cpu load) solution is the timer based sampling method. There are many discussions about this topic (e.g. https://www.mikrocontroller.net/topic/tasten-entprellen-bulletproof) but this turned out to be one of the best solution.

TD-er, 2020-01-16:
@oerweber I fully agree with most of your post.
But the sad truth is that ESPEasy does perform more tasks than just watching a pin and it does not really allow for extensive functions executed during ISR calls.  (very limited amount if iram which is needed for those functions)
I will read the topic you linked as it is always good to be (or remain) informed about these things.

tonhuisman, 2023-09-02:
This can be closed, as it is working 'good enough' at this time.

closed 2023-09-02 by TD-er