letscontrolit/ESPEasy #2477: Debug timing

s0170071, 2019-05-23:
@TD-er Did you know that you can have your own software-watchdog handling routine ? You could use that send/store/display messages if the SW watchdog triggered. Or just halt the system. Or just serve the webserver...

Found it [here](https://mongoose-os.com/blog/esp8266-watchdog-timer/).

Example sketch

```
void INT(void *arg) // this is the ISR for the WD
{
  Serial.println();
  Serial.println("A");
  yield(); // feed the WD
}

void setup() {
  Serial.begin(115200);
  ets_isr_attach(8, INT, NULL);  // my own ISR for the SW watchdog.
}

void loop() {
  for (;;) Serial.print("*");  // This should trigger the WD- but it does not.
}
```
This sketch makes the software WD feed the hardware WD :-)
I am not saying that we should do this in the normal builds, but this code here could go into a debug option or debug plugin to catch rare restarts.

TD-er, 2019-05-23:
Hmm, did not know that.
It would indeed be great for debugging.
Thanks.  Does it matter when the callback function is being registered?  (does it have to be registered in the `setup()` call?)

What is the argument for the call back function?

TD-er, 2019-05-23:
I'm thinking what endless possibilities this will bring.

- Mark what statistics recorded event was active when the watchdog was triggered (store a counter at start of the statistics call and compare with the stop call + increase watchdog counter in the call back function)
- Reduce some timeout settings if watchdogs are catched
- "hold" or "forward" the watchdog event (e.g. decide when to reboot or restart wifi for example)
- Store important stuff before actually crashing

And lots of other possibilities.
This is really great !!!!

TD-er, 2019-05-23:
Do you know what the scope is of this callback routine?
Is it part of the cont stack or the sys stack?

s0170071, 2019-05-23:
Nope. Is there way to test? I would assume so. Need to research first.

uzi18, 2019-05-23:
thx to @s0170071 link, we can find explanation of serial bootloader [issue](https://mongoose-os.com/blog/why-esp8266-cannot-be-soft-rebooted-after-first-flashing/)

s0170071, 2019-05-24:
@TD-er got one more for you: 
https://github.com/esp8266/Arduino/issues/1152#issuecomment-450362922
It is a library that registers a callback for crashes, similar to postmortem.c In the event of a crash, it saves the stack to spiffs. That would mean that every crash report could include a stack trace. Or two ;-) Repo is here:
https://github.com/krzychb/EspSaveCrash

TD-er, 2019-05-24:
That one I already saw once, but may also be nice to have a look at indeed.

s0170071, 2019-05-24:
That Wdt ISR and EspSaveCrash code could go entirely into a plugin.

One thing I noticed: The Wdt ISR can be triggered only 20 times before it needs to go through the main loop once. Otherwise the HW WD will trigger. This should be no big deal though.

tonhuisman, 2023-09-03:
This has been implemented, though maybe in some different form. Can be closed.

closed 2023-09-03 by TD-er