letscontrolit/ESPEasy #2479: Please add VEML7700 sensor

ways, 2019-05-25:
### Summarize of the problem/feature request

Please add VEML7700 sensor. It's a i2c calibrated light sensor that gives LUX values. Docs at https://www.adafruit.com/product/4162

Probably pretty similar to VEML6070 added in 2440.

TD-er, 2019-05-26:
I have still not received the other VEML sensors I ordered myself, so I guess I can add that one too to the list of parts to re-order because of missing packages from China.

ways, 2019-05-27:
$5 at adafruit. Only found more expensive options at aliexpress.

(68 SEK at my closest supplier, which is about $7. https://www.electrokit.com/produkt/adafruit-veml7700-ljussensor-i2c/ )

tonhuisman, 2024-05-18:
It's been a while, but today I added support for VEML6030 and VEML7700 (nearly identical chips in specs)
Test build available from this [GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/9140693778).

Please report your findings here or in the PR.

PR #5055 (2024-06-30): [P168] Add VEML6030/VEML7700 Light/Lux sensor
Resolves #2479 
And a [Forum request](https://www.letscontrolit.com/forum/viewtopic.php?t=10156)

Features:
- Support for VEML6030 and VEML7700 Light/Lux sensors (nearly the same specs, same data interface)
- Using a modified version of the [Adafruit VEML7700](https://github.com/adafruit/Adafruit_VEML7700) library (we don't want to use delay calls while waiting for the conversion to complete)
- Implement the AutoLux mode as provided by the library
- Included in `Collection G`, `Climate` and `MAX` builds

TODO:
- [x] Add documentation
- [x] Testing  [... cut]
closed 2024-06-30 by TD-er