letscontrolit/ESPEasy #2480: Code review request _P012_LCD.ino for an LCD1602

Domosapiens, 2019-05-25:
Subject: Is the PLUGIN_READ code correct for an LCD1602 ?

May I ask for help for a code review?
My coding skills are from the '80ties.

When using an LCD1602  ... I see huge "READ" statistics values
https://github.com/letscontrolit/ESPEasy/issues/2463#issuecomment-494737478

Is the PLUGIN_READ code correct for an LCD1602 ??
https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P012_LCD.ino

Line 14:
LiquidCrystal_I2C *lcd=NULL;
int Plugin_012_cols = 16;
int Plugin_012_rows = 2;

Line 24:
#define P12_Nlines 4        // The number of different lines which can be displayed
#define P12_Nchars 80

Line 189
case PLUGIN_READ:
      {
        char deviceTemplate[P12_Nlines][P12_Nchars];
        LoadCustomTaskSettings(event->TaskIndex, (byte*)&deviceTemplate, sizeof(deviceTemplate));

        for (byte x = 0; x < Plugin_012_rows; x++)
        {
          String tmpString = deviceTemplate[x];
          if (lcd && tmpString.length())
          {
            String newString = P012_parseTemplate(tmpString, Plugin_012_cols);
            lcd->setCursor(0, x);
            lcd->print(newString);
          }
        }
        success = false;
        break;
}

My question ... is the PLUGIN_READ loop limited to the possibilities of the LCD1602 ?
Thanks for checking

TD-er, 2019-05-26:
I edited your post to add some formatting blocks to it to make it easier to read.

I thing the check for `lcd` can be placed wrapping the whole read call.
Then it will not doing anything if the plugin is not initialized.
But that's not the main cause of the delays I guess.

There is also not any limit check on the size of tmpString. So it may be possible that it will take a lot more than needed if the customtasksettings are not starting with a 0 on the first byte of a line.

The parseTemplate function can also be quite intensive.

I don't know how optimized the LCD library is.
You could add a call to `delay(0);` after each `lcd->print......` to prevent wifi timeout issues.

Domosapiens, 2019-05-26:
@TD-er , thanks for your effort to clean-up my post and answering.
But your answer goes beyond my ability of code understanding.
My role is limited to debugging user.

The module _P012_LCD.ino needs to handle displays with 4 lines and 20 char (2004) but also displays with 2 lines with 16 char (1602).

I use the 1602.
I was wandering if there was in the code something that tried to read 4 lines where only 2 lines are available, or the same for 20/16 char, or any other mismatch between the 20x4 versus 16x2 display

uzi18, 2019-05-27:
should be better if schedule every line in time

TD-er, 2019-05-27:
@Domosapiens It does first read the entire block of data from flash (if it isn't already present in memory)
But that's rather efficient and doesn't take a lot of time.

Then it does check for every line possible if there is something to write by checking the length of the string it should write:
`if (lcd && tmpString.length())`

If that string is empty (length = 0) then it will not go into the time consuming routine of looking for string markers to replace (e.g. `[name#value]` or `%ip%` )
The only thing that could be wrong here is when the binary data blob is not properly initialized with 0 values and thus may contain "noise" data for lines that cannot be displayed (because display does not have enough lines to show)

This leaves the question whether the LCD library is coded efficient.
Like I said, if you're experiencing stability issues due to these timings, you can add a `delay(0)` after the print statement. That will make sure the WiFi part is also given time to handle stuff and thus make it less of a problem the display routine is taking so much time.

The reason I marked the lines > 100 msec in green is to have a quick glance on where to look for things that might cause issues, since the wifi should not be left alone for long.
It should be handled every 10 msec at most, but I've seen that up-to 100 msec can also work if there is not really a lot of traffic.

Another thing you need to realize is that rendering the web page also takes time (few 100 msec), so if that's being done, the call to the loop will likely be quite high after a few page loads. (800'ish msec)

s0170071, 2019-05-27:
I propose a change like this: 
1. only set an update flag on PLUGIN_READ
2. in the 100ms task, update the lines one by one until 10ms are over. 
3. in the next 100ms time slot continute updating lines...

like this: (not tested, just hacked in here:)

```
uint8_t linePtr=0; 
boolean doUpdate = false;
uint32_t timeout; 
case PLUGIN_READ:
      {
     timeout = millis(); 
     doUpdate = true; 
     success = false;
     break;
     }

...
case 10_PerSecond:  // or so. not sure off the top of my head
      {
        char deviceTemplate[P12_Nlines][P12_Nchars];
        LoadCustomTaskSettings(event->TaskIndex, (byte*)&deviceTemplate, sizeof(deviceTemplate));  // is this really needed ?

        for (byte x = linePtr; x < Plugin_012_rows; x++)
        {
          linePtr++;
          String tmpString = deviceTemplate[x];
          if (lcd && tmpString.length())
          {
            String newString = P012_parseTemplate(tmpString, Plugin_012_cols);
            lcd->setCursor(0, x);
            lcd->print(newString);
            if (timePassedSince(timeout) > 10) break;  // timeout. Skip remaining lines.
          }
        }
        
       doUpdate = false;   // update complete.
        linePtr=0;   // reset line pointer.
        break;
}
```

TD-er, 2019-05-27:
@s0170071 Well it might work, to split its work.
But I would first look into the LCD plugin to see why it does take so much time.
If there's not much to be done there, then this might be a good idea.

s0170071, 2019-05-27:
@TD-er [here](https://forum.arduino.cc/index.php?topic=286837.msg2009827#msg2009827) some guys evaluated the timing of a 128x64 SSD1306 I2C display. Seems like clearing it takes 40ms plus another 40ms for each line of text. 
Thats already 200ms for four lines of text- pretty much what @Domosapiens found in his timing stats.

TD-er, 2019-05-27:
@s0170071 The OLED and LCD are different.
The LCD has its own fonts present, so you should only have to send a few bytes, right?
But if you need to wait for things to clear, you're right in that it should be split in multiple stages.

TD-er, 2019-05-27:
I just took a look at the LiquidCrystal_I2C library and it seems the way it is doing a print is not really fast.
For each character in the string, it is calling a `write`, which does a lot of work.

- `write4bits` is called 2x per character
- `expanderWrite` is called 3x per call of `write4bits` + 1 usec + 50 usec delay
- `expanderWrite` does a complete I2C beginTransmission + endTransmission.

This does look a bit elaborate.
Maybe it is also a good compromise to keep a buffer in the plugin to keep track of what characters are already displayed and only update those characters which are to be changed.
Then it only has to do all this work if there is something to change, which is perhaps not that often and not all characters.

Domosapiens, 2019-05-27:
@TD-er 
> Like I said, if you're experiencing stability issues due to these timings, you can add a delay(0) after the print statement. 

Now I understand .... delay(0) ... as a rule command.

But in my case, I just use the "Display LCD2004" task GUI  for Line 1 and 2:

`
[Flow#Count]L/m [MUC21kW#MUC21_kW]kW
H[TempH#TemperatureH]L[TempL#TemperatureL]d[DeltaT#Delta_T]
`

In the GUI, no empty spaces before or after.
The lines are displayed correctly. (Text&Value)
Lines does not exceed the display size (15char)

Line 3 and 4 are empty

(useless hint?) Any blame on this Feb 13, 2018 commit?
https://github.com/letscontrolit/ESPEasy/commit/b293e8fee0eb8d5764433710d4ed6e23a456e4f6#diff-ed46efd25058647d0b08e70302ed1621

TD-er, 2019-05-28:
@Domosapiens 
> Now I understand .... delay(0) ... as a rule command.

Nope, in the C++ code.
The delay command in the rules should not be used.

About the linked commit.
I don't see any striking issues with that commit, when quickly looking at it.

closed 2020-04-15 by Domosapiens

uzi18, 2020-04-15:
@Domosapiens @s0170071 @TD-er maximum known size is 4x20chars = 80
It can be buffered and splitted by lines to send.
Have understand we want to tweak something here?

TD-er, 2020-04-15:
@uzi18 It looks like @Domosapiens  is cleaning old issues he once opened.
The other ones he just closed were already fixed, so I assume this one is also no longer an issue or maybe replaced by other changes in the code base.

uzi18, 2020-04-15:
just want to know because it is a good idea with this buffering

Domosapiens, 2020-04-25:
> When using an LCD1602 ... I see huge "READ" statistics values

This was in the hunt for the "watchdog"
And ... the dog is gone now!
Thanks

TD-er, 2020-04-25:
> > When using an LCD1602 ... I see huge "READ" statistics values
> 
> This was in the hunt for the "watchdog"
> And ... the dog is gone now!
> Thanks

Do you have anything changed, which could be included in a pull request?

Domosapiens, 2020-04-27:
No @TD-er 
Using now 20191116 and running rock-solid.
Thanks