letscontrolit/ESPEasy #2481: Failed to compile on Arduino IDE

s0170071, 2019-05-27:
I fail to compile ESPEasy 20190523 with the Arduino IDE.

first I was missing StringProviderTypes.h
After changing #include <StringProviderTypes.h> to #include "StringProviderTypes.h"
 I get a whole bunch of new errors: 
```

StringProvider:3: error: 'LabelType' has not been declared
 String getInternalLabel(LabelType::Enum label) {
                         ^
StringProvider:7: error: 'LabelType' has not been declared
 String getLabel(LabelType::Enum label) {
                 ^
StringProvider:98: error: 'LabelType' has not been declared
 String getValue(LabelType::Enum label) {
```

and many many more...

Anyway, the webserver.ino has 7608 lines. When did that happen ? :-O

sracing, 2019-05-28:
Uff..... Yes. I can confirm. Same error by using Arduino IDE (it was my first try to compile ESP Easy so at least it seems it was not on my side.)

uzi18, 2019-05-28:
it is so much easier with platformio =)

TD-er, 2019-05-28:
I did not try to build it myself in Arduino IDE.
Maybe we should rename the src directory in the repository to `ESPEasy`, to match the name of the main ino file.   (this is a relatively new feature of PlatformIO to rename the directory)
Then it is easier to also test it using Arduino IDE.

PC-Eggsberde, 2019-06-06:
ist da schon eine Lösung in Sicht, bekomme es mit platformio auch nicht zum laufen.

TD-er, 2019-06-06:
I just added a PR to help fix this.
At least I could build it in Arduino IDE. Let's hope it still builds in PlatformIO.
If so, then I will merge it.

s0170071, 2019-08-12:
not working again...
AddrList.h: No such file or directory

TD-er, 2019-08-12:
That one is wrapped in defines to only have it used with core 2.5.2 or newer.
Arduino IDE should have those also defined, right?

TD-er, 2019-08-12:
```c++
/******************************************************************************\
 * Detect core versions *******************************************************
\******************************************************************************/

#ifndef ESP32
  #if defined(ARDUINO_ESP8266_RELEASE_2_4_0) || defined(ARDUINO_ESP8266_RELEASE_2_4_1)  || defined(ARDUINO_ESP8266_RELEASE_2_4_2)
    #ifndef CORE_2_4_X
      #define CORE_2_4_X
    #endif
  #endif

  #if defined(ARDUINO_ESP8266_RELEASE_2_3_0) || defined(ARDUINO_ESP8266_RELEASE_2_4_0) || defined(ARDUINO_ESP8266_RELEASE_2_4_1)
    #ifndef CORE_PRE_2_4_2
      #define CORE_PRE_2_4_2
    #endif
  #endif

  #if defined(ARDUINO_ESP8266_RELEASE_2_3_0) || defined(CORE_2_4_X)
    #ifndef CORE_PRE_2_5_0
      #define CORE_PRE_2_5_0
    #endif
  #else
    #ifndef CORE_POST_2_5_0
      #define CORE_POST_2_5_0
    #endif
  #endif

  #ifdef FORCE_PRE_2_5_0
    #ifdef CORE_POST_2_5_0
      #undef CORE_POST_2_5_0
    #endif
  #endif
#endif // ESP32
```

These `ARDUINO_ESP8266_RELEASE_2_3_0` should be defined in the Arduino core.
What core are you using?

s0170071, 2019-08-15:
I was using the freshly downloaded  zip from GitHub. Never mind, I got it sorted out by re-installing the Arduino IDE. Was using 1.8.5 portable, now using 1.8.9 non-portable.

TD-er, 2019-08-15:
OK, that makes me curious to what the difference may be between those versions.
Building in ArduinIDE now works without having to change the code? (adapting paths is still a bit of work, sadly)

closed 2019-10-10 by TD-er