letscontrolit/ESPEasy #2483: Feature request: espeasy as modbus-rtu slave (RS485)

Andrey2509, 2019-05-31:
I want to receive data using the modbus-rtu protocol when espeasy is modbus-rtu slave. This is required in order for the PLC (modbus-rtu master) to receive data from the sensors connected to the ESP8266 directly without intermediaries (WIFI, MQTT). Is it possible to modify modbus-rtu.ino for such a possibility? Or develop a new controller - the gateway modbus?

TD-er, 2019-05-31:
Strictly speaking is a controller (current design) something that will output data to another recipient.
And plugins do fetch data from some sensor or something that will produce data.

I am not really sure if this should be a controller or a plugin.
A controller will just output data when it is ready (push) and this is something that will be polled for data to pull from the ESP.
Also it is quite specific, since you must define what data should be mirrored/collected to what register address.

I guess a controller does make sense.
I have to look into the Modbus_RTU code to see what must be changed for it.
Please note that I will change the existing code for Modbus RTU to split the process command loop into parts that can be called from the 50/sec loop and thus sending a command will not return a result immediately. This also means that the 2 plugins using that code must be changed. So if you're using it in one of your own plugins, you're now aware of the upcoming changes :)

Andrey2509, 2019-05-31:
In the software on the modbus-master side (industrial plc), we can raise the waiting time for a response from a modbus-slave to 200 ms or more, so a frequency of 1/50 sec is sufficient.
It would be ideal to have espeasy controller (modbus-slave), which creates holding registers sequentially, starting with 40,000 numbers. The number of registers created must correspond to the amount of data transmitted from plugins and configured on this controller.

TD-er, 2019-05-31:
It is no guarantee it will reply in such short time. The 1/50sec loop is just the intentional (scheduled) interval. But if another part of the code does keep the unit occupied, it will not react in due time.
ESPeasy isn't a real-time environment like a PLC.

I'm working as hard as I can to make all plugins and controllers behave as real-time as possible.
Just to let you know that you should take timeouts into account when interfacing with ESPeasy.
On an ESP82xx you simply cannot make it absolute real time.
An ESP32 might be able to, since it has a second core that could be assigned real-time tasks.

Andrey2509, 2019-05-31:
Another way to get data for PLC is to use modbus-tcp if this is easier. In any case, the timeout can be increased to 1 sec.

pruwait, 2019-12-01:
Modbus RTU controller for EasyESP - it's a good idea. I wont to collect data from sensors (pirs, switches etc) and send it to hassio (modbus RTU master). More reliablity without WiFi and routers... I am ready to remove some plugins for more stability.

Git-ChrisSc, 2023-02-08:
This is a nice Idea:
ESPHome has an unofficial Plugin for that - but never try it.
i also tried to build it by my own on plaine MCUs - but i have not enough know-how. 
( I get only TTL-Modbus working not with RS485 - i think it is bug in the lib)

But using ESPEasy as an Modbusslave with the option to deactivate/aktivate WLAN would be pretty nice!
I love to use ESPEasy for building easy and reliable Sensors for quick Projekts.