letscontrolit/ESPEasy #2484: APDS-9960 Gesture Sensor is not listed in ESPEasy control panel

artafakhari, 2019-06-01:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

I can't see the APDS-9960 in the devices list.
I tried all bin files and also compiled my own formware.bin in platformio.
But can't see this component in the device section lists in ESPHome control panel.
I2c Scan in Tools get the following results:
Supported Devices:
PCF8574A
TSL2561
APDS9960
Which means the device is recognized by the firmware
### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

TD-er, 2019-06-01:
What build did you use?  (N.B. this is one of the lines asked to fill in when using the issue template)

It should be present in the testing builds.

See:
https://github.com/letscontrolit/ESPEasy/blob/5702efdcb2f332b70657e5d5df6ce9fe96a39167/src/define_plugin_sets.h#L579-L597

artafakhari, 2019-06-02:
Thank you Gijs
Yes it worked for me
As you said I've used test version of the build 20103 - Mega
How can I make a rule based on APDS gestures?

artafakhari, 2019-06-02:
Thank you Gijs
Yes it worked for me
As you said I've used test version of the build 20103 - Mega
But it seems that the device became unstable
Pages don't open sometimes

TD-er, 2019-06-02:
Do you have a log on the memory usage when that happens?
I think the test and dev builds are becoming too big for the memory usage.

I also do advice to use core 2.5.2 build. That core version is way more stable than the others.

artafakhari, 2019-06-03:
Hello Gijs
 ESPEasy is a great project but based on my needs I had to switch to Sonoff/Tasmota after testing some firmware.
But it doesn't mean that espeasy is weak.

Therefore I don't have access to log file anymore

closed 2019-06-03 by artafakhari