letscontrolit/ESPEasy #2486: Mega firmware on NodeMCU-wroom-esp32

liklev, 2019-06-06:
Hi, finally i succeeded to flash 
ESP_Easy_mega-20181026_esp32dev.bin on my nodemcu-wroom-esp32 board.
(but not the test kind) and its running very nicely.
i found some other thread around here solving it.
in the mega firmware package there is no
boot_app0.bin
bootloader.bin
partitions2.bin
i dont know if these bins are built in or not into the full firmware.
and i burnt them from commandline with the esptool.exe (under windows)

esptool.exe --chip esp32 --port COM15 --baud 256000 --before default_reset --after hard_reset erase_flash

esptool.exe --chip esp32 --port COM15 --baud 256000 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x8000 partitions2.bin 0xe000 boot_app0.bin 0x1000 bootloader.bin

esptool.exe --chip esp32 --port COM15 --baud 256000 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x10000 ESP_Easy_mega-20181026_esp32dev.bin

after giving the commands then it was trying to connect,then  i had to press and hold the boot button a little to catch the board.

i'am not familiar with it, but this process ended up with a good-working board. the same manner as my older esp8266 boards work with a single bin fw.

maybe the gui hands over wrong parameters?

thanx, lev

TD-er, 2019-06-07:
Thanks for the write-up to get it fixed.

I usually flash the ESP32 from PlatformIO, so had not noticed it wasn't working as binary download.

liklev, 2019-06-07:
sorry i'm not familiar with this issues thing.
should i close it ?

TD-er, 2019-06-08:
Nope, leave it open, or else it will not be looked at.

dpal80, 2019-11-27:
Hi There... I would like to ask you one thing.... I am trying to flash this NodeMCU with Esp Easy
https://www.amazon.it/gp/product/B074RGW2VQ/ref=ppx_yo_dt_b_asin_title_o07_s00?ie=UTF8&psc=1
I do not understand if the CHIP is a normal ESP32 or something different (Wroom32)

I have followed instructions of @liklev but my board does not boot up
this is the error at startup:

```
INIT : Booting version: (custom)
INIT : Cold Boot
FS   : Mounting...
CRC  : No program memory checksum found. Check output of crc2.py
CRC  : Settings CRC           ...OK
CRC  : SecuritySettings CRC   ...OK 
INIT : Free RAM:168836
INIT : I2C
INIT : SPI not enabled
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction)
. Exception was unhandled.
Register dump:
PC      : 0x4010b5fc  PS      : 0x00050031  A0      : 0x40005232  A1      : 0x3ffc0600  
A2      : 0xe7ffffff  A3      : 0x3ffc0c30  A4      : 0x40081bb0  A5      : 0x6000e050  
A6      : 0x00000000  A7      : 0x6000e0c4  A8      : 0x8000522d  A9      : 0x40087120  
A10     : 0x00000960  A11     : 0x08c5b249  A12     : 0x00000000  A13     : 0x00000394  
A14     : 0x0000004a  A15     : 0x6000e138  SAR     : 0x00000019  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x40089ec4  LEND    : 0x40089efc  LCOUNT  : 0x00000000  

Backtrace: 0x4010b5fc:0x3ffc0600 0x4000522f:0x00000000

Rebooting...
```

Any IDEA on what's happening and why I cannot manage to write this firmware to my board?

I was used to write ESP Easy oin ESP8266 Esp12E boards, and usually do not have any problem, but with this board I am going crazy!!!

TD-er, 2019-11-27:
Can you try flashing it from the editor (so not using the pre-compiled binary) ?
So then you compile it from the IDE and flash it from the IDE.

liklev, 2019-11-28:
hi dpal80,
this attached zip files includes all the components i used to burn my wroom32 nodemcu boards.
the cmd files i did not attachm so have to copy/paste from above (make only change to com port no)

[flash_firmware.zip](https://github.com/letscontrolit/ESPEasy/files/3900798/flash_firmware.zip)

liklev, 2019-11-28:
dpal80,
o gosh, (i comment too early)
my boards are power hungry. if them dont get enough through the microusb plug them go on boot-loop or something. have some hardware issue on nodemcu-wroom32 that something does not get power (usb controller chip or what).
i have some beefy variable power-supply module to experience how much volt cause stable run-up in case i feed the chip on Vin pin (5v/1A) is not enough. 6.5v sure going stable but the onboard voltage-regulator get a bit hot.

dpal80, 2019-11-28:
Thanks @liklev I Will ty your package.

So you think that should be a problem of power feeding this boot loop?

dpal80, 2019-11-28:
thanks @liklev with external power supply the board boots up regularly.
[image]

liklev, 2019-11-28:
hi dpal80,
i'm happy to hear your problem "rectified".
i've tumbled into this power-supply trouble once i tried to build a battery
operated unit, and to my surprise small power-modules of 5v/2amp claimed
capacity was unable to push through it of boot-loop.
it is a design flaw of those nodemcu 1.1 modules, a PCB trace is missing
between two components.

TD-er, 2019-11-28:
>  it is a design flaw of those nodemcu 1.1 modules, a PCB trace is missing between two components.

That's is very interesting information.
Do you have more info on that?

liklev, 2019-11-28:
oh, sorry. that thing happened to me at least 4-5 months ago. i cannot
recover links or solid facts, just my memory.
something like from the Vin pin power was going to regulator then to the
esp32 itself but nothing back to the usb chip, then on booting the usb chip
was unable to give ready signal and the boot was cycling and waiting for
it.
since then i had a full clean-up laptop-wise. no history just my messy
synapses.

uzi18, 2019-11-28:
this is known ch340 issue

liklev, 2019-11-28:
but opposing to the previous fairy-tale somehow a sufficient 6-7-8volt
feeding on Vin used to kick it up into good running. all this design flaw
stuff i collected from forums.

PR #2967 (2020-04-06): ESP32: add subtarget for genuine units firmware-factory.bin
Fixes: #2964
Fixes: #2941
Fixes: #2486
closed 2020-04-06 by TD-er