letscontrolit/ESPEasy #2491: BME280 i2c no output to OLED i2c

semaf, 2019-06-12:
Using the same board as here https://github.com/letscontrolit/ESPEasy/issues/1415

Build:⋄ | 20103 - Mega
System Libraries:⋄ | ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support
Git Build:⋄ | mega-20190607
Plugins:⋄ | 47 [Normal]
Build Md5: | 66e9fc3eb52e77ec6b166ca1c6847b0
Md5 check: | passed.
Build Time:⋄ | Jun 7 2019 02:15:21
Binary Filename:⋄ | ESP_Easy_mega-20190607_normal_ESP8266_4M.bin

Connected is a BME280 over I2C (D4/D5) and I am getting results from the sensor. Forwarding them to the OLED screen keeps the screen dark. Nothing to see.

Tested without BME280 to show the IP and can see it. As soon as BME280 is connected the OLED is off.

OLED Address: 0x3C
BME280 Address: 0x76

A regular sketch over Arduino is running fine with OLED and BME280.

semaf, 2019-06-12:
I think I could find the issue. As soon as I set the value for the Interval higher than 0 the screen goes off after the give values time. The screen comes only back if I disconnect cable and reconnect.

TD-er, 2019-06-12:
Sounds familiar.
There have been more reports of boards with an OLED which does need a separate GPIO for resetting the OLED.

This issue has a lot of information about those boards: https://github.com/letscontrolit/ESPEasy/issues/1031
The last post also does mention a fix including adding a resistor: https://github.com/letscontrolit/ESPEasy/issues/1031#issuecomment-498376535

semaf, 2019-06-12:
Wouldn't it be easier just to add the option to select the reset pin to set it high after each interval?

`U8G2_SSD1306_128X32_UNIVISION_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ 16, /* clock=*/ 4, /* data=*/ 5);`

is doing it well without a resistor

TD-er, 2019-06-12:
It may be, but I don't have those boards which have these issues and I don't know what is the exact problem here.
I just read your issue and it looked familiar to me so I linked the issue which may have some additional info regarding your problem.

ShardanX, 2019-07-02:
Not to forget the marginal issue that GPIO's are a rare ressource....

tonhuisman, 2022-05-07:
This doesn't seem relevant any longer, and can probably be closed

closed 2022-05-07 by TD-er