letscontrolit/ESPEasy #2492: Code review request: ICACHE_RAM_ATTR for _P059_Encoder.ino ?

Domosapiens, 2019-06-13:
This is a code review request for  _P059_Encoder.ino.
(still hunting for watchdog reboots)

Since https://github.com/letscontrolit/ESPEasy/releases/tag/mega-20190522 the 
ICACHE_RAM_ATTR attribute is used for Time functions and _P003_Pulse.ino.
And I remember a discussion of ICACHE_RAM_ATTR before or after the ISR definition.

Is the ICACHE_RAM_ATTR attribute also needed for the _P059_Encoder.ino interrupt routine?
https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P059_Encoder.ino
https://github.com/letscontrolit/ESPEasy/blob/mega/lib/MechInputs/QEIx4.h
https://github.com/letscontrolit/ESPEasy/blob/mega/lib/MechInputs/QEIx4.cpp

Thanks for help
Ref: https://arduino-esp8266.readthedocs.io/en/latest/faq/a02-my-esp-crashes.html#other-causes-for-crashes

TD-er, 2019-06-13:
Just a first tip in hunting down the watchdogs.
Try to focus on builds based on core 2.5.2
That one does appear to have some of the crashes and reboots fixed.
Doesn't mean they are all fixed, but at least we're then narrowing down the number of possible causes and maybe find some consistent explanations.

TD-er, 2019-06-13:
I've looked through the code and it looks like you may be on to something.
The function called from the ISR should be declared with the ICACHE attribute and only use variables declared volatile.

Neither the ISR() function, nor the __instance array is declared volatile.
Also the ISR function does too much here.
It should just set a flag to see what kind of interrupt was triggered and maybe when and then exit.
In the loop call the set flag(s) should be processed and set back to their "unset" or "handled" state.

From a programmer point of view it is a nice solution, but it can lead to undefined behavior on this specific platform.

What does bother me, is that this code has not been touched in a while and I don't recall seeing reports about this plugin.  That's an indication it was working fine at some point and maybe at some core lib version.
It could be that in older versions the resource usage was a lot less which allowed this code to remain in RAM all the time?

Domosapiens, 2019-06-13:
Thanks @TD-er 

> Try to focus on builds based on core 2.5.2

Will do. Next week no time, but after that I will upgrade all 7 units to 2.5.2

> What does bother me, is that this code has not been touched in a while and I don't recall seeing reports about this plugin

I just finished my second Thermostat unit with rotary encoder (without any other interrupts) and saw some WDR's while using the encoder.
But later also WDR's without touching the encoder. So it could be an Effect instead of Cause.
I will inform you after next week.

Domosapiens, 2019-06-26:
Following your advice, using ESP_Easy_mega-20190522_normal_core_252_ESP8266_4M.bin.
After a number of reboots the Rotary Encoder task is disabled (b.t.w. a good function!).

The error message: **ISR not in IRAM!**

The debug log shows:
`
INIT : Booting version: mega-20190522 (ESP82xx Core 2_5_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.1.2 PUYA support)
81 : INIT : Free RAM:33888
82 : INIT : Warm boot #101 - Restart Reason: Software Watchdog
84 : FS   : Mounting...
109 : FS   : Mount successful, used 77057 bytes of 957314
505 : CRC  : program checksum       ...OK
534 : CRC  : SecuritySettings CRC   ...OK 
620 : INIT : Free RAM:30832
622 : INIT : I2C
622 : ISR not in IRAM!

Abort called

Then a stack dump follows
`

This problem was not in my previous release 20181128 normal

TD-er, 2019-06-26:
Well the abort due to incorrect used ISR function calls has only recently been added to the core library.
And personally I really like the feature, since it will give a clear indication on what is wrong, instead of crashing later at some totally non reproducible way.

TD-er, 2019-07-30:
Just a note to myself to help me find it later this evening.
Should fix now :)

commit 472d004f (2019-08-12): [Encoder] ICACHE_RAM_ATTR added for encoder plugin

Fixes #2492
closed 2019-08-12 by TD-er