letscontrolit/ESPEasy #2493: PCA9685 on ESP32

alancilia, 2019-06-17:
ESP32-WROM-32  Crashes and restarts when adding PCA9685 in devices 
and PCA9685 will not Work at all.

Tried:   20190425_esp32test_1M8 & 20190425_normal_WROOM02_2M.bin 
same problem only works with the old ESPEasy32_R20100.bin 

Thanks.

alancilia, 2019-07-02:
Hello 

Still no Fix for this in the last Release :(

TD-er, 2019-07-02:
That's correct, it has become part of another PR which was not yet merged.

alancilia, 2019-09-23:
hello 

Any chance this will be fixed pls :(

TD-er, 2019-09-23:
I have been too occupied with other stability issues.
Only the last few weeks, I've been testing with ESP32 (again), but not with this one.

I just did have a look at the plugin and noticed there is some very strange code in there to mark which addresses are in use.

I changed it and here it is not crashing on ESP32 when I add the plugin and enable it.
Should that be enough to make it crash?

alancilia, 2019-09-23:
Hi TD-er 

Thanks very much for finding some time to look at this bug , i will wait until the the bin file is released cos I don’t know how to compile it . 
Hope the fix will be in the next release . 

Thanks again 

Regards 
Alan

TD-er, 2019-09-23:
Could you test [this test build](https://www.dropbox.com/s/ko2vzsu06g8j60a/esp32test_1M8_partition_issue2493.bin?dl=0) ?
It has a lot of other commits in it, so don't run it on a node that has to be in operation or is hard to reach :)

alancilia, 2019-09-23:
Hi thanks very much for your work i will test tomorrow and let you know

alancilia, 2019-09-24:
Good Morning  Td-er

I just tested the bin file and it is working great here not crashing anymore :) .
in the old ESPEasy32_R20100.bin there was a problem that after a few minutes 
i2c will stop working, i have build a small setup here I will leave it running for 
a few hours and let you know .

Thanks again 

Alan

TD-er, 2019-09-24:
Thanks for testing.

closed 2019-09-24 by TD-er

alancilia, 2019-09-24:
Nice :) 

Thanks and keep up the good work

TD-er, 2019-09-25:
Just to let you know.
The fix is not yet included in last night's build, but it has been merged now, so it will be included in tomorrow's build.