letscontrolit/ESPEasy #2494: Problem tcrt5000 + Pulse counter.

ded000, 2019-06-22:
Problem tcrt5000 + Pulse counter.
When using tcrt5000 + Pulse counter, ESP periodically reboots on Watchdog.
When switching tcrt5000 there is a strong level bounce, which causes ESP to reboot.
On tcrt5000 entered feedback, result 0.
How to achieve stable work?

TD-er, 2019-06-22:
What build do you use?
About 2 weeks ago a user reported an issue with the interrupt handling of the pulse counter plugin.
See https://github.com/letscontrolit/ESPEasy/pull/2469
This was merged on May 21st

ded000, 2019-06-22:
Currently Git Build: mega-20190607.
Also this problem is observed on other assemblies.
I did not find the assembly without this problem.
I can simulate this problem if I place the sensor on the black-white border.

TD-er, 2019-06-22:
OK, good to know it is still a problem.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-03 by TD-er