letscontrolit/ESPEasy #2496: ESP32 WROOM TEST firmware - To much plugins!

henkiejan1, 2019-06-26:
I have flashed my new ESP32 Wroom with ESP32 test firmware with the instructions found here: https://www.letscontrolit.com/forum/viewtopic.php?t=6354

Bootlog:
`
INIT : Booting version: mega-20190626 (ESP32 SDK v3.3-beta1-179-ge931fe9f5-dirty)
58 : INIT : Free RAM:292880
59 : INIT : Cold Boot - Restart Reason: CPU0: Vbat power on reset CPU1: for APP CPU, reseted by PRO CPU
61 : FS   : Mounting...
101 : CRC  : No program memory checksum found. Check output of crc2.py
112 : CRC  : SecuritySettings CRC   ...OK
183 : INIT : Free RAM:287644
184 : INIT : I2C
184 : INIT : SPI not enabled
184 : System: Error - To much Plugins
185 : System: Error - To much Plugins
193 : INFO : Plugins: 75 [Normal] [TEST ESP32] (ESP32 SDK v3.3-beta1-179-ge931fe9f5-dirty)
194 : Programming error! - Increase PLUGIN_MAX
198 : WIFI : No valid wifi settings
201 : WIFI : Could not connect to AP!
204 : WIFI : Set WiFi to AP
1049 : WIFI : AP Mode ssid will be TEST ESP32_0 with address 192.168.4.1
1054 : OTA  : Arduino OTA enabled on port 3232
2388 : WD   : Uptime 0 ConnectFailures 0 FreeMem 243216 WiFiStatus 0
`

But when i look in the serial information there is some error with 'to much plugins!'. Is there a fix?

Btw: i see the focus on ESP Easy is a lot on ESP8266. Very nice. But the ESP32 has a lot more features. Why not more on the ESP32? :)

TD-er, 2019-06-27:
Hmm, that's something that should (now) also happen on ESP82xx test builds.
The fix is mentioned in the log ;) Increase the value for `PLUGIN_MAX`

v-a-d-e-r, 2019-06-27:
Typo: To much Plugins => Too much Plugins

v-a-d-e-r, 2019-06-27:
Typo in __Plugin.ino: To much Plugins => Too much Plugins

TD-er, 2019-06-27:
Well if it is going to be changed, change it to "too many Plugins" :)

v-a-d-e-r, 2019-06-27:
Yeah, sounds also not bad. ;)

closed 2019-10-10 by TD-er