letscontrolit/ESPEasy #2497: Add support formaldehyde sensor ZE08-CH2O

ofinto, 2019-07-02:
Hi, I want use a formaldehyde sensor ZE08-CH2O. Is support planned for this?

TD-er, 2019-07-02:
I have never heard of the sensor, so until now there was no support planned.

ShardanX, 2019-07-02:
This sensor has two outputs: DAC 0,4...2 V and UART.
Using the analog output it should be possible to use it with ESPeasy's analog input (Voltage divider 1:2 needed) if using directly with ESP-12 or similar.
UART is 3.3V so it is voltage compatible to ESP8266. It will need a specific plugin with ESPeasy.

Datasheet can be found here:
https://www.winsen-sensor.com/d/files/PDF/Gas%20Sensor%20Module/Formaldehyde%20Detection%20Module/ZE08-CH2O%20V1.0.pdf

marsic-84, 2019-11-06:
So... Does support planned? And when?

TD-er, 2019-11-06:
> So... Does support planned? And when?

I don't have the sensor, so there is no planned timeline to support it.
As @ShardanX pointed out, if you use the analog output of the sensor, you can already use it, but it will not be as exact as when using the UART interface.

llegovich, 2020-10-03:
HI there, I'm also interested in ze08 interface.
Source code can be found here https://github.com/rsalinas/ze08-ch2o-arduino
I have the sensor and can test on ESP32 if this helps.