letscontrolit/ESPEasy #2504: Could not find a version that satisfies the requirement '2.20402.4-puya' for your system 'windows_amd64'

SSilverHawk, 2019-07-07:
visual studio code 1.36 x64

I downloaded the latest Mega(mega-20190630) code without any modifications.
The following error occurred when I compiled the code.

espressif8266 @ 9ffc857 has been successfully installed!
The platform 'https://github.com/TD-er/platform-espressif8266.git#patch/v1.8.0_Puya' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
PackageManager: Installing framework-arduinoespressif8266 @ 2.20402.4-puya
Error: Could not find a version that satisfies the requirement '2.20402.4-puya' for your system 'windows_amd64'
The terminal process terminated with exit code: 1

TD-er, 2019-07-07:
Do you have a directory resembling this core lib version in your platformio directory?
In Windows it should be something like this:
`C:\Users\gijs\.platformio\packages\framework-arduinoespressif8266@2.20402.4-puya`

If this is the only one you tried to build, it will probably not have the `@2.20402.4-puya` part in the directory name.

What PlatformIO version are you running?

I there any reason why you want to build the core 2.4.2 version?
Core 2.5.2 does seem to be more stable.

TD-er, 2019-07-08:
I ran a test for it on my computer too an I can reproduce it here:
```
> Executing task: C:\Users\gijs\.platformio\penv\Scripts\platformio.exe run --environment normal_core_242_ESP8266_1M <

DEPRECATED! A legacy library storage `c:\GitHub\TD-er\ESPEasy\.piolibdeps` has been found in a project. 
Please declare project dependencies in `platformio.ini` file using `lib_deps` option and remove `c:\GitHub\TD-er\ESPEasy\.piolibdeps` folder.
More details -> http://docs.platformio.org/page/projectconf/section_env_library.html#lib-deps
Processing normal_core_242_ESP8266_1M (framework: arduino; platform: https://github.com/TD-er/platform-espressif8266.git#patch/v1.8.0_Puya; board: esp01_1m)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
PackageManager: Installing toolchain-xtensa @ ~1.40802.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
toolchain-xtensa @ 1.40802.0 has been successfully installed!
PackageManager: Installing framework-arduinoespressif8266 @ 2.20402.4-puya
Error: Could not find a version that satisfies the requirement '2.20402.4-puya' for your system 'windows_amd64'
The terminal process terminated with exit code: 1
```

TD-er, 2019-07-08:
Also found the cause, it is something with my domain that hosts the json file for the patched version.

TD-er, 2019-07-08:
It should be fixed now, or at least I cannot do anything more now.
The problem was in the SSL certificate on my domain td-er.nl which hosts the patched version of the core 2.4.x libraries.
I moved the domain to another VM, since I was not able to generate new Let's Encrypt certificates on the old one. This requires a DNS update, which may take up-to 24h (TTL was set to 86400 sec).
On my computer I was able to download the patched libraries again so if it is not yet working, please try again in a few hours.

Building core 2.5.2 should not be affected by this.

stefbo27, 2019-07-08:
I had the same problem. Switching to 2.5.2 solved this issue, but now I have to cope with #2492. :-(

Christian-Me, 2019-07-09:
Had the same problem, after @TD-er fixed the certificate you need to perform a „clean“ before build. Then the dev 4M version builds again.

stefbo27, 2019-07-09:
@Christian-Me you are right, after a "clean" the dev 4M version builds again!
 
@TD-er: is there a reason, why dev 4M version is still on core 2.4.2? You stated 2.5.2 is more stable...

TD-er, 2019-07-09:
@stefbo27 No there is no reason anymore other than that I did not yet merge the change in the platformIO.ini file.

Last week I was almost exclusively occupied with the WD reboot issues and have not done a lot of other stuff on ESPeasy. (the last 2 or 3 builds only had changes/fixes made by others)

stefbo27, 2019-07-12:
No problem, as all works fine again...
Thanks a lot!

closed 2019-07-14 by TD-er