letscontrolit/ESPEasy #2505: powering ESP32 only on USB

megamarco833, 2019-07-11:
hi , i have ESP32 esp-wroom32 with yellow pins:
https://www.aliexpress.com/item/32845423155.html?spm=a2g0s.9042311.0.0.27424c4dYMzqRH
i upload succesfull the bin file of last release for ESP32

but i find an issue to power up the ESP:
1) if i use the USB cable, it boot up normally, and i have no problem on using it.
2) if i use the 5v and GND pins the ESP continue to reboot and red led continuos blinking.

I guess that there is a problem about Wifi connnection in case i use pins intead of usb port.

If i connect using usb port, then wait around 20seconds, and the connect the 5v and GND pin, and after that disconnect the usb cable, the ESP will not reboot end become stable.

It's a "known issue" ?
it's related in someway to firmware?

thanks

TD-er, 2019-07-11:
How do you power it?
What kind of power supply and cables I mean.
Also is there anything else connected to the ESP32 board?

Especially during WiFi connection the ESP may draw quite a bit of current (short peaks of up-o 500 mA)
If the power supply you're using is not really powerful or the cables used are quite thin, then the voltage to the ESP board may drop.
Such a voltage drop may cause all kind of undefined behavior.

The reason why I mention any other connections is that if there is somehow some indirect connection to other nodes or devices then there may be current flowing through places where you don't want.

Can you show a picture of how it is connected?

megamarco833, 2019-07-11:
> How do you power it?
> What kind of power supply and cables I mean.
> Also is there anything else connected to the ESP32 board?
i power it with a power supply 230v-5v stabilized using dupont cables. i'm confident that is not here the problem because i use same stuff to power on esp8266 (nodemcu or wemoss d1 mini) without any problem.
ESP32 has nothing connected
 
> Especially during WiFi connection the ESP may draw quite a bit of current (short peaks of up-o 500 mA)
> If the power supply you're using is not really powerful or the cables used are quite thin, then the voltage to the ESP board may drop.
> Such a voltage drop may cause all kind of undefined behavior.
ok, but i use the same stuff to power on ESP8266.
ESP8266 is different compared to ESP32 ?
 
> The reason why I mention any other connections is that if there is somehow some indirect connection to other nodes or devices then there may be current flowing through places where you don't want.
> 
> Can you show a picture of how it is connected?
> Also try to measure the voltage over the pins where you power it. (preferably at the pins on the ESP board)

i measure 5,1V and it can be ajusted from 5,5V to 4.5V, nothing change in this range.

thanks for your support.
p.s. anyway i bet that the problem could be inside the wifi connection, because if i use usb cable to power on ESP32 (5v and gnd disconnected) i wait 15-20seconds (wifi in this time will connect) then connect 5v and GND, and after that wait 5seconds and disconnect usb cable.
in this case the ESP32 still connect with 5v and GND 
if i use directly 5v and GND ESP continue to boot in a loop.

TD-er, 2019-07-11:
Just a warning about the power switch you just tested.
It is good to know there seems indeed to be something fishy here (might be the USB to serial does behave badly and does perform a reset), but be aware that supplying external 5V to an USB cable connected to your PC may cause damage to your PC or USB hub if you're using one.

Do you have some USB power bank you can test?
Some of these power banks do also set some defined state on the data pins to allow the device to pull 500 mA or more. Maybe powering it via the USB port but not sending any data can help to test if the USB to serial chip is doing something strange here.

megamarco833, 2019-07-12:
thanks again for reply, but sorry maybe i explain in not a good way.
if i use only usb, i do not have any issue. all is working and boot correctly.

if i do not use usb, but only 5v pin the esp continue to boot and i never succeed on login to its web page.

the only way to use 5v pin is:
1) connect only usb cable and not 5v pin
2) wait 20-25seconds with  only usb cable connected
3) after this 25seconds connect the 5v pin and still mantein connected usb cable
4) wait 5seconds
5) disconnect usb cable and still mantein connected 5v pin
6) now the esp32 will be poweron only by 5v and it still connected without any problem

TD-er, 2019-07-12:
I understand it, but I just want to eliminate the possibility the PC does something to the USB=>serial chip on the ESP.
If you're using an USB power bank (or just about any USB power plug), it will not communicate to the USB=>serial chip, but is does set some resistors to the data lines indicating the USB device can switch to > 100 mA. That's the only communication then. It also does use the USB power lines.
I know these USB=>serial chips can perform a reset on the ESP. You can easily verify that by opening a serial console to the ESP and see it will reboot.

A next test could be to use some "charge only" cable, meaning one that does not allow data communication. But that's something not generally available.

My assumption is that the USB=>serial does perform a reset when it powers on, or maybe even keeps the ESP reset triggered when it is not powered.
I can imagine the USB=>serial chip is not powered when using the 5V lines instead of powering it via the USB port.

megamarco833, 2019-07-15:
> A next test could be to use some "charge only" cable, meaning one that does not allow data communication. But that's something not generally available.

ciao, i did some tests using different power charge and connect them directly to ESP32, with this the ESP is not booting and continue goes to a boot loop
only way to power one is using cable connected to USB computer.
 
> My assumption is that the USB=>serial does perform a reset when it powers on, or maybe even keeps the ESP reset triggered when it is not powered.
> I can imagine the USB=>serial chip is not powered when using the 5V lines instead of powering it via the USB port.

do you have an idea how to solve it?
thanks

TD-er, 2019-07-15:
You can try running the basic blink sketch in the Arduino toolkit to make sure it is not an issue with ESPeasy.
But I am afraid it may be a (hardware) design flaw on the board you have.

megamarco833, 2019-07-15:
> 
> 
> You can try running the basic blink sketch in the Arduino toolkit to make sure it is not an issue with ESPeasy.
> But I am afraid it may be a (hardware) design flaw on the board you have.

ok thanks, this evening i'll try to make this test.
anyway i ordered a new one ESP32 wroom to test if it's the hardware problem or not :-)

p.s.
i see the latest build on github: Release mega-20190715
there are not inside the bin file compiled?
why?
thanks again

TD-er, 2019-07-15:
About the build issues.
I also noticed there were no binaries inside, so I just fixed the build script.
The update to PlatformIO 4 did move the location where the files are built, so I had to update the build script.

megamarco833, 2019-07-15:
hi, related to module, no, i chanege the vendor.
mine module has also pins that are yellow. the new one should be black.

Related to build issue, can you please substitute the zip file on github, or i'll wait for tomorrow build?
thanks

TD-er, 2019-07-15:
There will be a build tomorrow anyway.
I don't think anything related to your issue has changed in the nightly builds.
If you like, you can test [this build](https://www.dropbox.com/s/dr9rn7hfv26yjzu/ESPEasy_mega-20190715-2-PR_2511.zip?dl=0) I made for testing the build script, but I don't think it will make any difference.

manmoncang, 2019-07-15:
Hi guys. Sorry disturb both of you.

i also have same problem. 

Powered by USB no problem. But using 5V pin has problem which wifi always reset. 

i read your discussion above but dont have any solution or suggestion to solve. Hope somebody can help this.

plin2, 2019-08-12:
I'm running ESPeasy on an ESP32 board with OLED and battery. I have the same problem. Connected to USB the wifi starts fine, connected to a strong power supply I don't get wifi, but the display shows 'ESPeasy'.

uzi18, 2019-08-12:
Did anyone try to connect external rs232/ttl converter?

TD-er, 2019-08-12:
> I'm running ESPeasy on an ESP32 board with OLED and battery. I have the same problem. Connected to USB the wifi starts fine, connected to a strong power supply I don't get wifi, but the display shows 'ESPeasy'.

@plin2 This looks like the ESP does boot, but crashes.
Maybe you can start a ping from one node, to see if it actually does get an IP-address?
Via rules you may also try different things like acting on [events](https://espeasy.readthedocs.io/en/latest/Reference/Events.html) as system#boot to set a LED.
Maybe we can then see how far it does get into the startup process.

I've re-read this issue again and I think it may be possible to power the unit by powering over 3.3V instead of 5V.
This way the USB controller chip does not get power.
My theory is that the USB controller chip does create some pin state at startup which may trigger the reset-circuit.
Such a circuit is meant to help putting the node into flash mode, but it may now trigger a reset.

uzi18, 2019-08-12:
@TD-er it is also possible that usb converter chip mess up something.

uzi18, 2019-08-12:
is converter chip CP2102 or CP2104 ?

uzi18, 2019-08-12:
[image]

@TD-er it could be related to auto program feature

plin2, 2019-08-12:
@TD-er: I had a ping running while connecting the ESP32 to the power supply. No response. Just tried it again and it works. I think I had given the device a fixed IP address and changed 'Restart WiFi Lost Conn:' to ON. Just trying to revoke these changes and test again. Can't reproduce the connection probem right now. Will have to try later again.

TD-er, 2019-08-12:
> 
> @TD-er it could be related to auto program feature

That;s indeed my hypothesis.
The circuit on the lower left of your posted part of the schematic.

plin2, 2019-08-12:
I had my device running in the power supply until now and it still had the wifi connection and DHCP-based IP address.

megamarco833, 2019-08-12:
> 
> 
> I had my device running in the power supply until now and it still has the wifi connection and DHCP-based IP address.

what's happen if you remove the power supply 5V and after a while reconnect again the 5V external?
the ESP32 will work again, or become in continius boot loop?
what's the working configuration that are you using now? (eg:'Restart WiFi Lost Conn: is on or off ?; etc....)

plin2, 2019-08-12:
State
Restart WiFi Lost Conn:
no fixed IP adress
detached device from power, after reattaching it doesn't connect to wifi, 'ESP Easy' on the display is coming/going in an app.r 4.5 secs loop

attaching it via USB
immediate wifi connect

attaching it to power
no wifi connect

setting a fixed IP address
attaching it to power
doesn't connect to wifi, 'ESP Easy' on the display is coming/going in an app.r 4.5 secs loop
after a while 2 pings went through
device doesn't respond to ping again

now the display is permanently showing 'ESP Easy', accordin to my access point the device is connected, pings are not successful

using the on-board power switch off->on
steady display of 'ESP Easy', connection to access point, no ping

TD-er, 2019-08-12:
How are you "attaching it via USB" ?
Is that on a:
- computer
- USB hub
- "smart" charger

If one of those, can you check on the simplest USB charger you can find with at least 500 mA (preferably more) power

plin2, 2019-08-12:
Connection via USB is to my computer. That way I cann use the serial monitor in Arduion IDE to check the state etc.

Well, since I have as working display: is it possible to show the wifi connections state and IP address on the display?

plin2, 2019-08-12:
accidenatially plugged it into the other USB port of my computer => no ping response
effect is reproducable

TD-er, 2019-08-12:
If you're using the Framed OLED plugin, it does update every second the WiFi connection state via the bar-icon in the upper right corner.
When not connected, nothing is shown, otherwise 5 bars indicating the RSSI.

TD-er, 2019-08-12:
That "other port" is that the difference between USB2 and USB3?

plin2, 2019-08-12:
Im using the 'Display - OLED SSD1306' device. I have this board: https://de.aliexpress.com/item/32905558379.html?spm=a2g0s.9042311.0.0.2be14c4dGSWZdq
It only shows 'ESP Easy' in line 1.
On the right USB port the device appears as COM6

well, things keep changing. Attached to the löeft port it's still COM6 and it just got a wifi connection and responds ti pings

plin2, 2019-08-12:
would that give us some help?

on System#Boot do
    Let,1,0                              // set user variable to initial state of 0         
    oled 2,1,'reboot'
EndOn
 
on Wifi#Disconnected do              	 // if no wifi, wait 3 minutes
  if [VAR#1]=0
    timerSet,1,180
    Let,1,1                              // set user variable 1 to 1 to prevent re-set of var1 on each event.
    oled 3,1,'disconnected'
  endif
endon
 
on Wifi#Connected do     	 	 // if wifi returns, cancel reboot
  timerSet,1,0                    	 // cancel timer
  Let,1,0                                // set user variable 1 to 0
  oled 3,1,'connected'
endon
 
On Rules#Timer=1 do                    	 // if still no Wifi, reboot
  reboot
endon

plin2, 2019-08-12:
I'm just seeing some rebooting loops while attaache to the power supply. The cycle still is appr. 4.5 secs.
The loop can start after showing
'ESP Easy'
'ESP Easy' + reboot 
'ESP Easy' + reboot + disconnected, '', ''
on the display.

TD-er, 2019-08-12:
Can you see the reboot reason right after it did successful connect to wifi?
It should be on the sysinfo page.

plin2, 2019-08-12:
I had id attached to power, no connection. Then switched it to USB and without reboot a got a connection:

Reset Reason: | CPU0: Vbat power on reset CPU1: for APP CPU, reseted by PRO CPU
-- | --

TD-er, 2019-08-12:
OK, that's the expected reset reason.
The reason I somewhat "hoped" for was this one:
`case RTCWDT_BROWN_OUT_RESET : return F("Reset when the vdd voltage is not stable");`

plin2, 2019-08-12:
well, it's not what it looks like:
ANKER 5 port power supply: problems
USB port of computer: ok
smaller 3 port power supply: ok
small 1 port power supply: ok

TD-er, 2019-08-12:
OK, so it seems the Anker power supply does some communication with the serial port controller.
That's not really a surprise, since Anker is known for their quality products. So it does make sense they do support something more than just 2 resistors on the data lines of an USB port to signal "you may take 500 mA".
The PC port obviously does also do something, but that's not going to crash the USB->serial controller since it will be some kind of USB init sequence..

Can you see what USB->serial adapter is used (as asked by @uzi18 https://github.com/letscontrolit/ESPEasy/issues/2505#issuecomment-520336827 )

plin2, 2019-08-12:
>Can you see what USB->serial adapter is used (as asked by @uzi18 #2505 (comment) )
I don't have any documentation for the board. I took a photo and found a TP5400 chip near to the USB port, other one further away an SS54. Is there any technical way to find out?

TD-er, 2019-08-12:
You can see in the device manager of Windows what driver is used.
The CH340 is a rectangle shaped SMD chip.
The CP210x ones are smaller and square shaped.
I think it is not the CH340, since that one almost never gives issues with rebooting the node.
The CP210x ones do reboot the device by just opening the serial port or rebooting the computer.

plin2, 2019-08-12:
According to Windows it's a Silicon Labs CP210x USB to UART Bridge

TD-er, 2019-08-12:
And on the chip itself?

plin2, 2019-08-12:
I didn't find the chip yet. There are two without printing on them
- a little silver one (unknown number of pins)
- a little black square one with 28 pins

TD-er, 2019-08-12:
Yes, that pin count should be right.

- CP2102-GM* QFN28
- CP2109-A01-GM* QFN28
- CP2104 QFN24

TD-er, 2019-08-12:
I was thinking, it would not be the first time those popular chips are cloned.
And given it has no marking on them does suggest these adapter chips may not be the real Silicon Labs ones.

plin2, 2019-08-12:
I enlarged the photo of the board but still can't recognise anything like printing on that 28 pin chip...

uzi18, 2019-08-12:
sometimes photo with flash do the job

if you have some experience you can try to remove both small transistors

plin2, 2019-08-12:
@uzi18: I did use a flash, photo manipulation (contrast, btightness,...), no luck

uzi18, 2019-08-12:
if it is clone or even broken chip, thats why no marking, do you have another module to test?

plin2, 2019-08-12:
no, that's the only one

yonirom, 2019-12-24:
This may be irrelevant, but I encountered the same issue and think my solution may be applicable to others.
The Yellow pinned devkit c sold by aliexpress has incorrect silk screen markings as seen below.
(The pin marked GND is actually CMD).
If GND to a different GND pin, the chip boots as expected.
[image]

megamarco833, 2019-12-25:
> This may be irrelevant, but I encountered the same issue and think my solution may be applicable to others.
> The Yellow pinned devkit c sold by aliexpress has incorrect silk screen markings as seen below.
> (The pin marked GND is actually CMD).
> When connecting GND to a different GND pin, the chip boots as expected.
> [image]

Hi, thanks for the info. 
But if i do not understand well, if the GND pin close to Vin pin are marked wrong, if i need to power on this esp32 where i connect vin and gnd?
I guess vin (+5v) is correct pin, so i will put here 5v
But for gnd there is only another one pin marked as gnd: close to 3v3, i need to use this one?
Do you know if maybe there is another pin that it is gnd not marked?
Thanks

yonirom, 2019-12-25:
See image below. There are two real ground pins:
(Image borrowed from: https://components101.com/microcontrollers/esp32-devkitc)
[image]

megamarco833, 2019-12-25:
> See image below. There are two real ground pins:
> (Image borrowed from: https://components101.com/microcontrollers/esp32-devkitc)
> [image]

Thanks, but mine module seams different... 
I bought on aliexpress as esp32 nodemcu 
It's like the image fron here
https://circuits4you.com/2018/12/31/esp32-devkit-esp32-wroom-gpio-pinout/
I have two gnd pin, at the bottom, one close to vin and one close to 3v3 like the link that I post above

yonirom, 2019-12-25:
I understand, so these are probably different issues. 
The symptoms you described were very similar to what I experienced, so I thought this might help others.
Good luck.

megamarco833, 2019-12-25:
I think that you are right maybe different module has the same issue : pins are not marked well.
The problem is how i can discover where is the right gnd in my version?

megamarco833, 2020-02-22:
hi, i have three different ESP32 bought on aliexpress.
i fash them with espeasy without any problem, but i have a strange behavier when i power on them.
if i use usb cable, no problem at all.
if i use pin Vin and GND with a poer regulator of 5V i can see the web page of espeasy.
for sure is not the power suppy problem, i use this one to power up 6 esp8266, so i can exclude it

i see that if i connect before the usb cable, and after that connect the Vin and GND, the webpage will be visible and ESP32 do not crash.

i would like to use esp32 to connect 8ch relay and 8 switch
i create the task but if i use esp32 powered by USB cable i can sitch on/off the relays, if i use Vin and GND power (plus usb cable, otherwise as i explained usin only pins is not working the ESP32 power up) the relays are working.

these are my device:
https://ibb.co/nb1mH3L

you can see same ESP32 wroom32
i have two of those with yellow pin and one with black pins, but the behavier is the same for all my three ESP32
has someone this problem?
can someone suggest me on aliexpress a link where i can buy an ESP32 that can be programmed with espeasy that can use pins as power-up and can use relays+switch ?
thanks

aliusc, 2020-05-01:
Hi, I may confirm issue. When powering Esp32 devboard (board look like upper board of previous comment) with USB port it starts fine and works as expected. USB port may be powered from PC, Phone charger or powerbank - it works always. When powering with 5V through VIN and GND it fails connecting to WiFi network due it do not find it or because authentication error. Scanning for networks returns from zero to two results when powered from VIN. When powered from USB it returns always 4-5 networks. I live in semi-urban area.

I added capacitors to power lines etc, and do not imagine how I could improve startup of device without attaching USB cable.

larryn46, 2020-05-04:
I have two different devkit1 boards. I can't recall from whom I bought them, but the one on the left, with the white letters on the black background worked on USB only. The one on the right with (small) black letters ESP32S on a white background works on USB **and** battery connected to VIN. The battery voltage at VIN was measured at 4.8V
[image]
The second image is of the same boards in the same position. The front looks identical.
[image]
The power regulator (you can't see it in the above images, but with a magnifying glass, they both use AMS1117 3.3v chips) are identical.
I hope this saves everyone a lot of trouble avoiding the  same problem that has been bugging a lot of us.

megamarco833, 2020-05-04:
Do you try to power on the black one using vin pin with 5.6 volt? Mine work in that condition...

larryn46, 2020-05-04:
I made a mistake in the above comment. The voltage supplied to both boards was 3.8v, not 4.8v.

larryn46, 2020-05-04:
To megamarco833, I have a power supply that has a voltage of 5.03 volts when nothing is attached. The voltage drops to 4.6 volts when connected to the devkit1 board that is black on the back with white letters. It boots without any problems when the power is supplied to the VIN pin using the GND next to the 3V3 pin. Did you see the above comment that the so-called GND pin next to the VIN pin is really CMD, not VIN? On my board, the two GND pins are electrically connected, so I don't think it matters.

[42 more replies]