letscontrolit/ESPEasy #2507: NaN values cause crash in rules execution.

TD-er, 2019-07-13:
See also #1843 

For example the output of a DS18b20 Dallas sensor may output a NaN value.
```
if [Temperature#IntTemp] < 10
```

This is enough to cause a crash of ESPeasy.
In my test setup, the unit reboots several times until it disabled the Dallas sensor.

jimmys01, 2019-07-13:
Perhaps make the "NaN " value of the DS18b20 to be 85 degrees C, as this is the error reading (check datasheet) ? This at least will "fix" this particular sensor.

Grovkillen, 2019-07-13:
Or a given negative of -8266°C ✌️

jimmys01, 2019-07-13:
well the 85 is the default error reading for this particular sensor :-p

uzi18, 2019-07-13:
about 85deg, it is also possible to have such readings not as error! how to know it is an error?
maybe we should know last value and compare it before assign

I think we should have also option to ommit assign value if we think it is wrong

jimmys01, 2019-07-13:
85 is the max the sensor can go and also the power-on reset value of the temperature register. It is in the datasheet. If you ever see 85.00 as a reading then this is an "error" reading

uzi18, 2019-07-13:
no, max is 125deg in the datasheet on first page

jimmys01, 2019-07-13:
Anyway 85.00 is the way that the sensor tells you that something is wrong

TD-er, 2019-07-13:
As you may have seen (at least @uzi18  did ;) ) I did make some selector for setting a value to signal the error state in this commit: https://github.com/letscontrolit/ESPEasy/pull/2501/commits/48559142846d7c0f39d3cf7dfa11ce13df7859d0

85 degree is IMHO a very confusing value as error state, since it is in the range of the sensor itself.
In the commit I made, I included -55, 0 and +125 as error state value options. (and NaN which is the default)

@uzi18 suggested to also add an option for "ignore".

uzi18, 2019-07-13:
thats why dallas temperature library choose - 127 as error state

TD-er, 2019-07-15:
Maybe the files inserted in the rules before execution should use `doFormatUserVar` to fetch (and format) the value.
The `mustCheck` flag must then be set.

uzi18, 2019-07-15:
@TD-er you did remove delay(800) in Dallas plugin, maybe it will not break anything, sure it chould be different solution here

about doFormatUserVar - why?

jimmys01, 2019-07-15:
https://github.com/TD-er/ESPEasy/blob/1aba4286be4e87a6c88d26fac9d8c97ba29b5d72/src/_P004_Dallas.ino#L280

```
  if (DSTemp == 0x550)      // power-on reset value

 return false;
```

0x550 is 80 dec. So there is that.
Also uzi18 is right, you need the delay to be there to make the measurment, or else you will get the previous measurement

[image]

TD-er, 2019-07-15:
Good point, will add it again.  Probably forgotten while debugging.
But I will also make some changes later since I don't want any plugin to cause a delay that long.

TD-er, 2019-07-15:
> about doFormatUserVar - why?

Since then we have some check for a valid float/long at a single place in the code.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-03 by TD-er