letscontrolit/ESPEasy #2510: No binaries included in the last 2 releases

ironataerial, 2019-07-15:
https://github.com/letscontrolit/ESPEasy/releases

-D

TD-er, 2019-07-15:
The previous one was a build error which I did fix yesterday. But indeed there are no binaries in the last one.
I will look into it.

TD-er, 2019-07-15:
I made a test build on my own Linux machine and it should be fixed now.

Sasch600xt, 2019-07-15:
same here....i can´t find binaries in last release

TD-er, 2019-07-15:
It is already fixed and tomorrow will be a new build.
It was caused by the upgrade to PlatformIO 4, which does build the binaries in another directory.

As soon as the new nightly build is ready, I will remove the last two release marks (and copy the release notes)

Sasch600xt, 2019-07-15:
ok,
thank you for the fast reply :)
i am looking forward to the new release :)
Have a great day !

uzi18, 2019-07-16:
@TD-er it looks like with this PR https://github.com/letscontrolit/ESPEasy/pull/2509 (reduce size to minimal) we have more space in release file for new features =)

TD-er, 2019-07-16:
@uzi18 That was what I was aiming for ;)
I even set the IR part to be extended (including AC codes)

Also I deleted the incomplete nightly builds.
Last night's build looks successful, so I guess this issue can be closed.

closed 2019-07-16 by TD-er

uzi18, 2019-07-16:
compressed code included ;)

TD-er, 2019-07-16:
@uzi18 Sorry, I don't get it (may be nice benefit from PIO 4?)

uzi18, 2019-07-16:
@TD-er it is just joke PR said "reduce size to minimal" and we have almost empty release archive ;)

TD-er, 2019-07-16:
Ah OK :)