letscontrolit/ESPEasy #2515: JSON to num

Quart666, 2019-07-19:
I want to use weather data sent på Weather-Displays application MQTT client to my Mosquitto.
But Weather-Display can only publish messages with the values inside { } and ESP Easy dont like that.
I think more people want to use tags sent as JSON. 
So I request a JSON to NUM with MQTT import or the possibility to run script like this to fetch just numbers.
var MyNum = parseFloat(args[0].match(/[\d.-]+/)[0]);

await setTagValue(“RetNum”, {type: “number”, title: “RetNum”}, MyNum);

return true

uzi18, 2019-07-19:
link to application please
also please provide/show json data you want to support and how it looks like now

Quart666, 2019-07-19:
Weatherdisplay.com
Link to forum and thread with mqtt Info and examples.
https://www.weather-watch.com/smf/index.php/topic,67853.0.html

Just let me know if you want more information

PR #3424 (2022-04-02): [P037] Add mapping, filtering and json support
Extension to P037 MQTT Import plugin:
- Mapping string to numeric/other string (on -> 1, off -> 0), or numeric to percentage of a value (0..255 -> 0..100)
- Filtering to limit/select processed data, support for = and (numeric) ranges, can select idx=123 to get only sensor 123 from Domoticz/out
- json support to process complex messages, can select sub-value from f.e. Domoticz svalue="16.4;28.7;2;990.6;0", "svalue;4" selects the 4th value: 990.6
Generates event for all or selected json attribute(s)
- Generic topic-prefix [requested in the forum](ht [... cut]
closed 2022-04-02 by TD-er