letscontrolit/ESPEasy #2516: PMS5003 free D0, or other than D0 (test version plugin for espeasy)

pdrobek, 2019-07-19:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

pdrobek, 2019-07-19:
D0 is needed to enter deep-sleep mode, in this case D0 is used for SET pin of PMS5003. Deep sleep is essential for dust sensor powered by batteries

TD-er, 2019-07-20:
Not only for battery usage, but also for the lifetime of the internal laser in the sensor.
For example the laser in the SDS011 dust sensor is rated for 8000 hours. In continuous use, that's about a year.

ShardanX, 2019-07-24:
Are you refering to the schematic I've published in the forum a while ago?
If so - and if the plugin wasn't changed meanwhile - the D0 is not driven by the plugin but by a rule.
It is necessary to stop the sensor between measurements as the lifetime of the laser is very limited.
Anyways, as it is driven by rule the GPIO can easily be changed to another free GPIO pin.

pdrobek, 2019-07-24:
@ShardanX how to check for rule which determinates usage of D0? I would like to change it for another not used GPIO In my project i would like to power off whole set, after certain time aprox 10 minutes power on ... wait ... measure and repeat. I use 18650 cells to power it up.

ShardanX, 2019-07-25:
The Rules i use are quite simple:
`on System#Boot do
   gpio,16,0
   timerSet,1,20
endon

On Rules#Timer=1 do
   LongPulse,16,1,5
   timerSet,1,20
endon`

At bootup GPIO16 (D0) is set to "0" and a timer is startet.
If the timer triggers, a longpulse of 5 Sec is set to GPIO16, it's long enough for giving a set of values from the sensor. Then the timer is started again.
That's all.  So just change the "16" to another appropriate GPIO.

chemmex, 2019-07-26:
There is a little problem with that setup: unless you pulldown the gpio you are going to use for SET pin, it will float in high level during deepsleep. I had to put an N-FET to invert the logic, so 0 for PMS active and 1 for sleeping.

ShardanX, 2019-07-26:
The design I published includes a power supply as schematics and PCB show.
It was never meant for using deep sleep.
For deep sleep I'd suggest to use another GPIO and a pulldown resistor.

mackowiakp, 2019-08-11:
Another solution is using Nettigo NPMS-5 instead PMSx003. It is modification of Plantower sensor. They change laser diode to red one so lifespan is 40000 h of continuous work. Moreover, they added temp/hum buid-in sensor for software compensation. But FW of this sensor is still in prototype state.
[](https://nettigo.eu/products/nettigo-npms-5-air-quality-humidity-and-temperature-sensor)
I use it with RPi with own Python driver, written based on Github avaliable one.

TD-er, 2019-08-11:
@mackowiakp does that sensor need any changes in the software to be supported?

mackowiakp, 2019-08-11:
I think so. As I wrote. I use it with RPi with own Python driver, based on (modifying) PMSx003 driver avaliable on Github. It is because whole communication protol is identical to PMS but they add some more fields at the end of transmission string. 
Below full description of this protocol. But it is in Polish. I think they have it in English. Just ask them.
[Protokół_komunikacyjny_NPMS-5_i_NPMS-6.pdf](https://github.com/letscontrolit/ESPEasy/files/3489627/Protokol_komunikacyjny_NPMS-5_i_NPMS-6.pdf)

ShardanX, 2019-08-25:
Sorry for a late answer, I'm a bit off atm.

As I was asked where to find livetime in the datasheet of the sensor - just nowhere.
Plantronics themselves said it has a lifetime of 100.000 hours... well, I belive that's a bit too optimistic.
The MTBF (Meantime between Failures) for these laser diodes is 100K hours in permanent work, sure, that's correct. But the MTBF is not the problem. 

Laser diodes have a so called "degradation". Simplified the light of the diode gets darker over time.
As the sensor measures the light scattered by dust particles the "getting lower" of the laser has massive influence to the measurement.
The point is the degradation, not the MTBF. Data about that can be found in the datasheets of these laser diodes.
I've built two sensors when the PMS7003 came here. They were running in same location, one with laser permanent running, one with laser on and off by rule.
These sensors have some tolerances, that was expected when looking at the price.
The interesting point is that the permanent running laser showed lower values over time and deviation from the pulsed sensor grew so degradation was at work most likely.
Sadly I don't have the equipment to test a bigger bunch of sensors, so this test is phenomenologic.

@makowiakp
I don't know what laser is built into the NPMS-5 so i can't say anything, I don't even have a sample.
40.000 hours sounds good on first sight but make sure it's the degradation time. If it is MTBF, it might be worse then the Plantower sensors.

mackowiakp, 2019-08-25:
Yep. Because of that (I use it only for home needs) it was necessary to re-solder cables inside. In my application, ventilator works all the time but rest of electronics is in deep sleep mode. Each 10 min I wake-up sensor for 30 sec (docu stands that it is correct time to measure) and then read sensor. Everything from rule of course. So most of the time, laser diode is off.  I hope it will work a little bit longer.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-03 by pdrobek