letscontrolit/ESPEasy #2518: esp.easy.flasher.exe window too big and fonts

saper-2, 2019-07-29:
Hello,

The esp.easy.flasher.exe is unusable because the window goes outside of screen, I have 1366x768 monitor and the main window is going outside of screen (is just too high) - not every one have 8k monitor 😞 . Here is a desktop screenshot:
[image]

Can't you use a *normal* fonts (like Tahoma or Arial or Times? with normal size (10pt?) )? Even after installing them flasher still opens Windows Fonts and complain that there is no fonts installed...

I'd like to know why flasher try to connect to github (i'm touchy whenever something try to connect to internet 🐟 )...

I'm sorry, but I'm getting really annoyed when developers write a program (or web(app)) that he/she think look great on his hardware but won't consider others. Not to mention that you, not always can install fonts (e.g sys admin restrictions, or just user don't want to make bigger scrap-heap from own computer)

Grovkillen, 2019-07-29:
If you press "CTRL + S" a "Default.ini" file is saved to the settings folder. Open up that ini file and scroll down to the last row `Pixels Per Inch=`, change that value to something smaller.

Regarding the font, it was my bad decision early on. Please understand that that was my first real app I developed. And I first did it only for personal use... I will (if we can get this rocket launched, as in income of money) make a new app which will work on Mac and Linux as well as Windows. I'm a Javascript guy and thus I will make it a web application just like Slack etc. is working.

I make the app connect to github just to be able to automatically download later releases without the user needing to manually do that... You don't need to do it if you don't want, you can set that up in the ini file as well.

saper-2, 2019-07-29:
Ok, you're excused 😁 since it was your first app 😄

saper-2, 2019-07-29:
file: ```settings/Default.ini``` : 
```
[Debug settings]
Debug level=0
Pixels Per Inch=108
```

108dpi did the trick 😄 

As the checking for new version, I just block on my FW 😁

closed 2019-07-29 by saper-2

vmath54, 2020-06-21:
I have the same problem, under windows 10, with the latest version of ESP.Easy.Flasher.exe
It is almost unusable, even by changing 'Pixels Per Inch' in Default.ini. I'm giving up.

Grovkillen, 2020-06-22:
I will try to find time to see if I can fix it. I'm currently doing some development for a company and that is why I haven't been active in the ESP Easy community.

CybrGost69, 2025-02-03:
5 years later...any progress?

tonhuisman, 2025-02-03:
> 5 years later...any progress?

This software has been deprecated, and development has stopped a long time ago, as it's not usable for ESP32 flashing.
We advise to use either the ESPEasy WebFlasher (using Chrome or Edge browsers): https://td-er.nl/ESPEasy/
Or the Espressif Flash Download tool, to be downloaded from here: https://docs.espressif.com/projects/esp-test-tools/en/latest/esp32/production_stage/tools/flash_download_tool.html (a version of this tool is also included in the official [Release](https://github.com/letscontrolit/ESPEasy/releases) .zip files)