letscontrolit/ESPEasy #2521: Double Login

fly74, 2019-07-30:
Why it is necessary to double athenticate since a while?

[image]

[image]

TD-er, 2019-07-30:
A while ago we 'fixed' the authentication.
Previously it wasn't really authentication. If someone would login, the "logged in" flag was just active for some time, no matter if another client would connect.
This change lead to some issue with IOS browsers which appeared not following the standards, so for IOS we added a work-around.
If you try another browser (Chrome/Opera/Firefox) you'll see you only have to enter the authentication once.
To be honest, I never tried with Edge.

fly74, 2019-07-30:
Yes, it's only with edge. But why ask for username when it is alway admin? The password should be enough.

TD-er, 2019-07-30:
The thing is, I didn't fix it, and I also don't know for sure if there is a function present to only ask for a password.
Also later we may need several layers of authentication.

fly74, 2019-07-31:
But maybe there should be an only password option for people who need only one layer.

TD-er, 2019-07-31:
I guess so, but like I said, I don't know if the core libs allow for such a login prompt.
See here [ESP8266WebServer](https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WebServer)

fly74, 2019-07-31:
But what is the use case for multilayer auth? I think the main use is for dumb sending sensor data device. SSL would be more intersting.

TD-er, 2019-07-31:
One use case can be to have an admin part and one part that only allows to send commands or view only a subset of pages.
For example if you connect a device to the internet to have some JSON output to be loaded via some dashboard, but you don't want people to change stuff.
Or if you plan to install units in some area where not every user can be trusted to not trying to mess around in the settings, but also don't want to expose too much info.
So a read only mode and a installer mode.

And SSL is a bit out of the question on an ESP8266/8285 due to memory requirements of SSL.

fly74, 2019-07-31:
But any login without encryption isn't save, even multi layer

closed 2019-07-31 by fly74

TD-er, 2019-07-31:
That's true, but at least better that nothing.