letscontrolit/ESPEasy #2522: ESPEasy P2P Networking

fly74, 2019-07-30:
ESPEasy P2P Networking ist not working, even if port is set and controller is enabled. I don't use it since a long while, but it's now not syncing dummy devices to other ESP and in the main view the ESP only see itself. Has something changed or removed?

TD-er, 2019-07-30:
Did you reboot the unit after setting the port number?
That's one bug I didn't have time yet to fix.

At least you should see the other nodes in the network when the p2p function is active.
For example:
[image]

And in order to 'share' plugins among other nodes, you must make sure to have the p2p controller active and also the same slot (in devices page) free for the plugin to be installed via the other host.

fly74, 2019-07-31:
it looks like:

[image]
[image]
[image]
[image]
[image]

fly74, 2019-07-31:
slots free on the other devices

TD-er, 2019-07-31:
And the other device is also having the p2p network controller active? (not only the port set)

fly74, 2019-07-31:
Yes, all on, maybe another static IP issue?

TD-er, 2019-07-31:
I have to test it myself to see what's wrong.
I do know that the p2p controller isn't that intuitive. I have to look into the code myself to see how it should be working.

fly74, 2019-07-31:
I think this was looking different as it was working, maybe you can compare it to your setting.

[image]

TD-er, 2019-08-06:
I have tested it and it is working as I thought it should work.

- Make sure both nodes see eachother in the overview page (share the same UDP port and are rebooted after enabling/changing the port)
- Enable p2p networking controller on receiving node
- Make sure the receiving node has the spot free which is being used on the 'sending' node (in my example slot 12)
- Enable p2p networking controller on sending node
- Set the plugin you want to share to use the p2p controller

See the result on the receiving end:
[image]

But this is rather counter intuitive, so it definitely must be changed into a more intuitive procedure.
For example by either explicitly sending it to a specific node, or by setting some kind of dummy p2p receive plugin with its own purpose to receive data from another node.
Then you make sure it is only added to the node you want it to have.

TD-er, 2019-08-06:
One last note. You must make sure the build on the receiving end also has the plugin included.

fly74, 2019-08-06:
It is all set as you described. Can ist be an network-UPD issue? every esp only see itself.

TD-er, 2019-08-06:
If you set them all to the same UDP port and they are on the same network (subnet) then they should see each other.
A few things that I cannot stress enough:
- Make sure to reboot the node after changing the UDP port
- If you have ArduinoOTA enabled, use another port for ESPeasy p2p UDP (port suggested in most OTA examples use port 8266)
- Ping the nodes from some other host to keep their WiFi awake
- Disable "Eco mode" in the advanced settings
- Sharing a plugin to be auto installed on another node is only sent right after the plugin is set to use the p2p controller. So if you don't see it appear on the other node, save it again on the source node.

fly74, 2019-08-08:
last check - multible boots included

all devices
[image]

![2019-08-08 20_48_12-Microsoft Edge](https://user-
images.githubusercontent.com/38754941/62729593-8bdfdc80-ba1e-11e9-8ed3-1cad0b3b4990.png)

1st dev:
[image]

2nd dev:
[image]

in mainview all only see itself, i've no idea whats wrong

uzi18, 2019-08-08:
28266 is set on both units?

fly74, 2019-08-08:
yes alle 28266

uzi18, 2019-08-08:
are they connected to same AP ?

uzi18, 2019-08-08:
@fly74 what is dev1 and 2 IP configuration? paste from sysinfo, please

fly74, 2019-08-08:
yes all at the same AP, same subnet with static IPs.

TD-er, 2019-08-08:
Well we have port 8266 registered for this purpose, but I don't see any reason why any other port should not work.
As long as it is not being blocked by a firewall or some other rule and also isn't being used for something else.

Are the nodes connected to an AP with some kind of "guest network" functionality?
Some AP's do have a guest network feature, but one of the properties of such a feature is that they will not allow communication between clients on the same guest network.

TD-er, 2019-08-08:
Ah one other thing you may need to check...
Make sure all units have their own (unique) unit number set.

fly74, 2019-08-08:
[image]
FRITZ!Repeater 3000

I check the unit numbers

fly74, 2019-08-08:
Hi folks, it works with unique unit numbers. but how could anyone guess it? Is this written in the documentation.? THANKS a LOT.

closed 2019-08-08 by fly74

TD-er, 2019-08-08:
Well, not sure, but you can be sure it is documented very soon.

Just about all info is about it is summarized in this issue, so I will write it down now it is still fresh.

TD-er, 2019-08-08:
The current documentation on this controller is -to say the least- not written for the average user.
It is more or less a technical summary of its format and not about what you can do with it.

fly74, 2019-08-08:
The webinterface should give some hints, when p2p is going enabled. It saves a lot of time.

uzi18, 2019-08-08:
just warning about other unit with same address should be enough

PR #2537 (2019-08-08): [Docs] Document sharing plugin via ESPeasy p2p controller
Fixes #2522
commit 800e7d22 (2019-08-08): [Docs] Document sharing plugin via ESPeasy p2p controller

Fixes #2522