letscontrolit/ESPEasy #2525: Sonof OTA Flash not possible

fly74, 2019-08-03:
OTA flash from ESP_Easy_mega-20190731_minimal_core_252_ESP8285_1M_OTA.bin ist not possible:

http://ESP/update
shows only:

URI: /update
Method: GET
Arguments: 0

and ESP_Easy_mega-20190803_minimal_core_252_ESP8285_1M_OTA.bin seems to big with over 604 kb for 2way flash

uzi18, 2019-08-03:
yes as described, flash is too small in your esp, you can try 2 step flashing

fly74, 2019-08-03:
but the upload button only show the message:

URI: /update
Method: GET
Arguments: 0

so the 2 way flash cant be used.

TD-er, 2019-08-03:
Hmm, that's strange.
The build script has a check in it to not include binary files which are too big for OTA on 1M devices.
So an OTA file which appears to be too large after compiling should not be present in the download ZIP.

Can you give the values shown on the sysinfo page?
Maybe I made some error in the estimate on ESP8285 nodes.

fly74, 2019-08-03:
`System Info
Unit Number:
0
Local Time:
2019-08-03 20:00:46
Uptime:
1 days 0 hours 52 minutes
Load:
3.40% (LC=4018)
CPU Eco Mode:
false
Free Mem:
23864 (21824 - sendContentBlocking)
Free Stack:
3600 (1956 - sendContentBlocking)
Heap Max Free Block:
21064
Heap Fragmentation:
12%
Boot:
Cold boot (0)
Reset Reason:
Power on
Last Task:
Background Task
SW WD count:
0
Network ?
Wifi:
802.11N (RSSI -36 dB)
IP Config:
Static
IP / Subnet:
XXXXXX / 255.255.255.0
Gateway:
XXXXXXXXXX
Client IP:
XXXXXXXXXXX
DNS:
XXXXXXXXXX / (IP unset)
Allowed IP Range:
XXXXXXXXXXXXXXXXXXXX
STA MAC:
XXXXXXX
AP MAC:
XXXXXXX
SSID:
XXXXXXXXXXXXXXXXXXXXXX
Channel:
1
Connected:
1d00h51m
Last Disconnect Reason:
(1) Unspecified
Number Reconnects:
0
WiFi Settings
Force WiFi B/G:
false
Restart WiFi Lost Conn:
false
Force WiFi No Sleep:
false
Periodical send Gratuitous ARP:
false
Connection Failure Threshold:
0
Firmware
Build:?
20103 - Mega
System Libraries:?
ESP82xx Core 2_5_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.1.2 PUYA support
Git Build:?
mega-20190731
Plugins:?
3 [Normal] [Minimal 1M OTA]
Build Md5:
a1889d2cbd1fcd2d21e9e75497aea93a
Md5 check:
passed.
Build Time:?
Jul 31 2019 02:10:00
Binary Filename:?
ESP_Easy_mega-20190731_minimal_core_252_ESP8285_1M_OTA.bin
System Status
Syslog Log Level:
None
Serial Log Level:
None
Web Log Level:
None
ESP board
ESP Chip ID:
1225965 (0x12B4ED)
ESP Chip Frequency:
80 MHz
ESP Board Name:
PLATFORMIO_ESP8285
Storage
Flash Chip ID:
Vendor: 0x5E Device: 0x4014
Flash Chip Real Size:
1024 kB
Flash IDE Size:
1024 kB
Flash IDE Speed:
40 MHz
Flash IDE Mode:
DOUT
Flash Writes:
0 daily / 0 boot
Sketch Size:
626 kB (248 kB free)
SPIFFS Size:
110 kB (28 kB free)
Page size:
256
Block size:
4096
Number of blocks:
27
Maximum open files:
5
Maximum path length:
32`

fly74, 2019-08-03:
Hope it helps.

TD-er, 2019-08-03:
I updated your post to make it more readable.
Just a tip for any next time.
If you paste the text from the web page directly into the Github text window you will get proper formatting.
After that you can wipe out any info you think needs to be removed.

About the reported sketch size.
It is way more than 604 k which is about the max ota images may be

TD-er, 2019-08-03:
Found the bug, will fix it.

closed 2019-08-04 by TD-er

userosos, 2020-10-12:
Hi! I use son off ch4. If i pressed to Tools/Update firmware i can see:
URI: /update
Method: GET
Arguments: 0
I want update the sonoff use ESPEasy_2step_UploaderMega_1024.bin but i can't select the file for update because i has the page after press Update firmware button.

Unit Number: | 0
-- | --
Local Time: | 2020-10-12 09:17:27
Uptime: | 0 days 0 hours 5 minutes
Load: | 8.59% (LC=2400)
CPU Eco Mode: | false
Free Mem: | 15800 (34144 - )
Free Stack: | 3664 (3824 - )
Heap Max Free Block: | 11928
Heap Fragmentation: | 24%
Boot: | Manual reboot (28)
Reset Reason: | Software/System restart
Last Task: | Const Interval timer, id: 3
SW WD count: | 0
Network ❔
Wifi: | 802.11N (RSSI -40 dB)
IP Config: | DHCP
IP / Subnet: | 192.168.89.152 / 255.255.255.0
Gateway: | 192.168.89.1
Client IP: | 10.0.0.5
DNS: | 192.168.89.1 / (IP unset)
Allowed IP Range: | All Allowed
STA MAC: | DC:4F:22:CA:72:37
AP MAC: | DE:4F:22:CA:72:37
SSID: | CNC (84:16:F9:6D:F8:10)
Channel: | 11
Connected: | 4 m 56 s
Last Disconnect Reason: | (1) Unspecified
Number Reconnects: | 0
WiFi Settings
Force WiFi B/G: | false
Restart WiFi Lost Conn: | false
Force WiFi No Sleep: | false
Periodical send Gratuitous ARP: | true
Connection Failure Threshold: | 0
Firmware
Build:⋄ | 20107  - Mega
System Libraries:⋄ | ESP82xx Core a5432625, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ | mega-20200608
Plugin Count:⋄ | 46 [Normal]
Build Origin: | Travis
Build Time:⋄ | Jun  8 2020 21:23:13
Binary Filename:⋄ | ESP_Easy_mega_20200608_normal_ESP8285_1M
Build Platform:⋄ | Linux-5.0.0-1031-gcp-x86_64-with-glibc2.2.5
Git HEAD:⋄ | HEAD

TD-er, 2020-10-12:
You're missing the most crucial part of the system information, the "sketch size".
I guess the size of that bin is too large to perform an OTA.
The filename of the current firmware doesn't have "minimal OTA" in the name, so it is very likely too large to fit along with the 2-step OTA bin.

userosos, 2020-10-12:
> 
> 
> You're missing the most crucial part of the system information, the "sketch size".
> I guess the size of that bin is too large to perform an OTA.
> The filename of the current firmware doesn't have "minimal OTA" in the name, so it is very likely too large to fit along with the 2-step OTA bin.

But i not selected the file. I press the button and after it i see the error.

TD-er, 2020-10-12:
Hmm I don't understand what you mean.
If a 2-step OTA will not fit, there is also no "update firmware" button to press (it is not enabled).
So can you post a screenshot of where you press the button and also the exact error?

userosos, 2020-10-13:
> 
> 
> Hmm I don't understand what you mean.
> If a 2-step OTA will not fit, there is also no "update firmware" button to press (it is not enabled).
> So can you post a screenshot of where you press the button and also the exact error?

Ok. I make an video.
https://dropmefiles.com/D2fKx

TD-er, 2020-10-13:
Ah OK.
Just a last check, can you show the reported sketch size (and flash size) from the sysinfo page?  (almost at the bottom of the page)

userosos, 2020-10-13:
>  Just a last check, can you show the reported sketch size (and flash size) from the sysinfo page? (almost at the bottom of the page)

Storage
--
Flash Chip ID: | Vendor: 0x51 Device: 0x4014
Flash Chip Real Size: | 1024 kB
Flash IDE Size: | 1024 kB
Flash IDE Speed: | 40 MHz
Flash IDE Mode: | DOUT
Flash Writes: | 0 daily / 254 boot
Sketch Size: | 853 kB (20 kB free)
Max. OTA Sketch Size: | 599 kB (614384 bytes)
OTA possible: | false
OTA 2-step Needed: | true
SPIFFS Size: | 110 kB (27 kB free)
Page size: | 256
Block size: | 4096
Number of blocks: | 27
Maximum open files: | 5
Maximum path length: | 32

TD-er, 2020-10-13:
OK, so the real bug here is that the upgrade firmware button should have been disabled as it is not possible to perform an OTA update on this setup with the current active build.

userosos, 2020-10-13:
> OK, so the real bug here is that the upgrade firmware button should have been disabled as it is not possible to perform an OTA update on this setup with the current active build.

I want flash - ESPEasy_2step_UploaderMega_1024.bin after it i want flas - ESP_Easy_mega_20201009_normal_ESP8285_1M.bin
I can't do it?

TD-er, 2020-10-13:
Nope, as the size of the 2nd bin you want to flash is too big for 1M flash.

for the 2-step OTA, you need to store both simultaneous on the flash (which will not fit) and also keep the SPIFFS filesystem.

So the max. size of a bin to perform a 2-step OTA is 600 kB.

userosos, 2020-10-13:
> 
> 
> Nope, as the size of the 2nd bin you want to flash is too big for 1M flash.
> 
> for the 2-step OTA, you need to store both simultaneous on the flash (which will not fit) and also keep the SPIFFS filesystem.
> 
> So the max. size of a bin to perform a 2-step OTA is 600 kB.

I can't update it using wireless methods?

TD-er, 2020-10-13:
That is correct.
It simply will not fit using the build you are currently running.

userosos, 2020-10-13:
> That is correct.
> It simply will not fit using the build you are currently running.

What kind of firmware should I fill up to have a wireless upgrade and normal everyday work?

TD-er, 2020-10-13:
The 2-step update image is only capable of performing an update.
Nothing else.

Use case:
- Install an image like the "minimal OTA" builds.
- Perform OTA with the 2-step uploader image
- Perform OTA with the new "minimal OTA" build.

The 2-step uploader is 276 kB, the file system slightly less than 128k.
So there is 1020k - 276k - 128k => roughly 600k free with the 2-step image active.
If the "minimal OTA" bin is active, then there is roughly 276k free.

TL;DR
- With "minimal OTA" bin active, you can ONLY flash the 2-step updater bin.
- With the 2-step updater bin there is only room for "minimal OTA" bin files.
- With any other flashed bin file, there is NO ROOM for any known existing bin image (on an 1M flash)

userosos, 2020-10-14:
> The 2-step update image is only capable of performing an update.
> Nothing else.
>
> Use case:
>
>     * Install an image like the "minimal OTA" builds.
>
>     * Perform OTA with the 2-step uploader [...]

What diffirents bitween minimal and normal builds? Where minimal builds?
Thx for helipng!

TD-er, 2020-10-14:
They should be included in the ZIP files of the nightly build.
However, there is a check in the build script to see if they will fit anyway and if they don't, then they will not be included in the ZIP.

Also recently there is an issue with building variants using an older Arduino core library version (which is smaller) so those are not included for the last month or so (core 2.4.x builds)
Just checked and at least this one has a "minimal OTA" build included: https://github.com/letscontrolit/ESPEasy/releases/tag/mega-20200829

PR #2526 (2019-08-04): [OTA] Fix check 1M filesize during build
fixes #2525
commit fc4033a7 (2019-08-04): [OTA] Fix check 1M filesize during build

fixes #2525