letscontrolit/ESPEasy #2528: web-UI: wrong labels for network configuration?

es-ka, 2019-08-04:
### Summarize of the problem/feature request
If not using DHCP, the given network configuration on http://device_ip/config is not the same as shown on http://device_ip/sysinfo

### Expected behavior
Should be shown identically on both pages.

### Actual behavior
Please see summary. Maybe the labels on the config page are wrong/mixed up?

### Steps to reproduce
1. enter a static network configuration
2. go to sysinfo page
3. see some strange network configuration info

The problem persists after powering off and on.

### System configuration
Hardware:
WeMos D1 mini

ESP Easy version: 
ESP_Easy_mega-20190803_test_core_242_ESP8266_4M.bin

BoekGitHub, 2019-08-04:
ESP Easy version: 
ESP_Easy_mega-20190803_normal_ESP8266_4M.bin
Hardware:
Node MCU v.3

IP settings on page /sisinfo same to page /config. 
Good.

TD-er, 2019-08-04:
@es-ka can you show a screenshot of both pages?
@BoekGitHub Did you recently change the IP settings or upgrade from other settings?

BoekGitHub, 2019-08-04:
**> Did you recently change the IP settings or upgrade from other settings?**
I'll try to explain. 
I used a pure Node MCU V. 3 (from the package) and ESP.Easy.Flasher.exe (0.04.007)
First flash blank_4Mb.bin then, flash ESP_Easy_mega-20190803_normal_ESP8266_4M.bin
Not reset, not disconnected the power Node MCU. 
Then ESP.Easy.Flasher.exe (0.04.007) configure and flash:
Unit name: ESP
Number: 20
WiFi SSID: ######
WiFi Password:###########
Fixed IP (check): 192.168.1.200
Admin password, WiFi SSID (fallback) and WiFi Password:(fallback) left blank.

Then through the browser connected to the Node MCU (ESP_Easy_mega-20190803_normal_ESP8266_4M) and checked the IP settings. I've disconnected the power to the Node MCU several times and several times I checked the page /config /sysinfo.  That's all right.

**>have you also tested core 2.4.2?**
No. Testing only ESP_Easy_mega-20190803_normal_ESP8266_4M with Node MCU v.3.

TD-er, 2019-08-04:
The reported issue is with core 2.4.2
I recently fixed some issues regarding static IP, but I did only test those fixes on core 2.5.2 and the latest stage core version.
So maybe something is/was different with core 2.4.2 regarding static IP

BoekGitHub, 2019-08-04:
For information.
Now I checked ESP_Easy_mega-20190803_test_core_242_ESP8266_4M.bin with my Node MCU v.3.  No additional settings.

Flash blank_4MB
Flash ESP_Easy_mega-20190803_test_core_242_ESP8266_4M.bin
Not reset, not disconnected the power Node MCU.

Programm:
Unit name: ESP
Number: 20
WiFi SSID: ######
WiFi Password:###########
Fixed IP (check): 192.168.1.200
Admin password, WiFi SSID (fallback) and WiFi Password:(fallback) left blank.

Disconnect/connect (on/off) power.

Then through the browser connected to the Node MCU (ESP_Easy_mega-20190803_test_core_242_ESP8266_4M) and checked the IP settings. 
I've disconnected the power to the Node MCU several times and several times I checked the page /config /sysinfo. Strange, but all well. Maybe this is a special case.
[image]

TD-er, 2019-08-04:
@BoekGitHub Thanks very much for testing.
Let's see if @es-ka can show some screenshots to see if there's something wrong.
Just a note, make sure all fields of the IP config are filled in.

es-ka, 2019-08-04:
Screenshots will be provided on Tuesday when I have access to the device again.

BTW: I first flashed that Wemos D1 more than a year ago (did some OTA updates later) and it has been working well with DHCP. After upgrading to ESP_Easy_mega-20190630_test_core_242_ESP8266_4M.bin I decided to switch to static IP config and noticed the mismatch.

es-ka, 2019-08-04:
Meanwhile I tested with a brandnew NodeMCU. Same firmware - same strange behaviour.
[image]
[image]

es-ka, 2019-08-04:
As it turns out the label "ESP Subnet" should read "Net mask" instead. As soon as I changed "ESP Subnet" from 192.168.10.0 to 255.255.255.0 everything works as expected.

TD-er, 2019-08-04:
OK, glad we could get this cleared out.

closed 2019-08-04 by TD-er