letscontrolit/ESPEasy #2535: /timingstats page not protected by password?

ghtester, 2019-08-07:
I don't know if it is by design or if it's an issue similar to #1707 but currently (see the build info below) the /timingstats page is accessible without password entering even after device reboot.

Build:⋄ | 20103  - Mega
System Libraries:⋄ | ESP82xx Core 2_5_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.1.2 PUYA support
Git Build:⋄ | mega-20190731
Plugins:⋄ | 50 [Normal] [IR_Extended]
Build Md5: | 97cfac5af4187c94a9dc83e7069c8
Md5 check: | passed.
Build Time:⋄ | Jul 31 2019 02:21:44
Binary Filename:⋄ | ESP_Easy_mega-20190731_normal_IRext_ESP8266_4M.bin

It's also interesting that this webpage is not so sensitive to "F5 crash" already discussed somewhere a long time ago and which is still there - i.e. ESP_Easy_Mega  node can be easily crashed (rebooted with exception or watchdog reason) by keeping pressed F5  (= refresh) in web browser on any ESP_Easy webpage.

TD-er, 2019-08-07:
Well I don't think the timingstats page really has sensitive information.
The worst that can happen is that it does reset the stats by accessing the page.

What pages are more sensitive to "F5 crashing" ?
The rendering of this timing stats page is rather simple and although it is quite big in size, it really doesn't take long to render it.
Also it doesn't need to load anything from SPIFFS, which also may make a difference.

jimmys01, 2019-08-07:
by the way `/update` is also not protected

TD-er, 2019-08-07:
@jimmys01 That may be more of an issue :)

ghtester, 2019-08-07:
@TD-er  Well, unfortunately I would say any pages quickly refreshed from Firefox browser... :-) Am I alone who is experiencing this issue?

ghtester, 2019-08-28:
FYI the same issues including "F5 crashing" on ESP_Easy_mega-20190827_test_core_260_sdk222_alpha_ESP8266_4M.bin

TD-er, 2019-08-28:
I would be surprised if it has changed, since no related code for this has been changed recently to fix these issues.
And I am not sure "F5 crashing" is easy to fix anyway, since it is very likely the memory is being exhausted by packets not yet served.

ghtester, 2019-08-30:
I know but there was a small hope that core 2.6.0 build could be more resistant to "F5 crashing" ;-)

TD-er, 2019-08-30:
Indeed I don't think it will ever be "fixed" for this platform.
Maybe there is a way to get events for every request and there add some check to create some DoS attack check.
But I guess this may also take a lot of resources and also create new issues.

ghtester, 2019-08-30:
OK, Gijs, thanks for the details, I understand and I know there are more important things to do. But it's a pity that is so easy to crash the node using Firefox browser.

TD-er, 2019-08-30:
Not only a Firefox browser.
It can even be a simple JavaScript fetching the same JSON at an interval which is way too fast for the ESP to handle.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-03 by TD-er