letscontrolit/ESPEasy #2538: Receiving a HTTP GPIO command triggers PN532 Bus error

es-ka, 2019-08-08:
### Summarize of the problem/feature request
Immediately after receiving a HTTP GPIO command the I2C bus with attached PN532 RFID reader crashes.

### Steps to reproduce
1. set I2C configuration to SCL->D5 and SDA->D6
2. connect PN532 via I2C to D5/SCL and D6/SDA
3. send HTTP command: `http://ESP_IP/control?cmd=GPIO,5,0`
4. communication to PN532 breaks

The problem does persists after powering off and on.
Without sending HTTP commands the connection to PN532 is stable and RFID tags are read as expected.

### System configuration
Hardware: WeMos D1 mini
ESP Easy version: ESP_Easy_mega-20190803_test_core_252_ESP8266_4M.bin

The PN532 is the only device connected to the WeMos, wires are about 15 centimeters long.

ESP Easy settings/screenshots:
- Rules: enabled
- Rules old engine: enabled

### Rules
```
on RFID#Tag do
  Pulse,15,1,3000
  SendToHTTP 192.168.16.200,80,/status/api.php?task=auth&source=%sysname%&key=%eventvalue%
endon

on Notaus#State=0 do
  GPIO,5,1
  GPIO,4,1
  GPIO,0,1
  GPIO,2,1
  Pulse,15,1,3000
endon
```
### Log data
Web Log Level: debug dev
unfortunetly no messages about incomming HTTP request...
```
27778: EVENT: Clock#Time=Thu,23:11
32367: WD   : Uptime 1 ConnectFailures 0 FreeMem 18128 WiFiStatus 3
42867: PN532: Found chip PN532 FW: 1.6
43097: PN532: BUS error
43167: PN532: Found chip PN532 FW: 1.6
43397: PN532: BUS error
43467: PN532: Found chip PN532 FW: 1.6
43697: PN532: BUS error
```

uzi18, 2019-08-08:
What is your i2c confirmation?

TD-er, 2019-08-08:
OK, it should not crash, but I think this is also something you really should not want to do.
You don't want to mess with pins assigned to any plugin other than a switch plugin.

At least this command should check for valid pins and those are pins present in the monitoring struct.

es-ka, 2019-08-08:
> ok why you want to set gpio used by i2c?

I don't want. i2c is set up to use D5/GPIO-14 and D6/GPIO-12.
The command `GPIO,5,0` should set GPIO-5 which is called D1.
Is there any problem to use D5 and D6 for i2c instead of D1 and D2?

es-ka, 2019-08-08:
I did some more testing:
`http://ESP_IP/control?cmd=GPIO,4,0` -> i2c crash
`http://ESP_IP/control?cmd=GPIO,2,0` -> OK
`http://ESP_IP/control?cmd=GPIO,0,0` -> OK

TD-er, 2019-08-08:
> Is there any problem to use D5 and D6 for i2c instead of D1 and D2?

See [Best pins to use on ESP8266](https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#best-pins-to-use-on-esp8266)

es-ka, 2019-08-13:
Finally I changed the board layout for my project to use D1 and D2 for i2c to make it work again.
If it's not reliable to use other GPIOs than D1 and D1, the config should not be changeable in Web UI to avoid confusing the user, I think.

TD-er, 2019-08-13:
It depends on the used I2C devices whether it will work on other ports.
Also on some boards the on-board I2C devices are routed to other pins.

For example a board with OLED display I bought about 2 years ago, that one has the I2C pins swapped.
And on some boards (like some Sonoff and Shelly devices) these pins are already used for other purposes. So then you should at least have the option to disable I2C, but like I said not all I2C devices are that timing critical and thus may work on other pins too.

es-ka, 2019-08-13:
OK, thanks for clarification!
I didn't notice the hidden (but detailed) [documentation](https://docs.google.com/spreadsheets/d/1pQuBocanzCZJ3KPUpu3VvpOOBzboqXDuwQDVP5UYcXc/edit#gid=0) on GPIO pins.

closed 2023-09-03 by TD-er