letscontrolit/ESPEasy #2542: current distribution crashes Atom Editor

s0170071, 2019-08-12:
Since ESPEasy does not compile with the Arduino IDE (again) I use (tried to)

ATOM 1.39.1 @Linux Mint 19.1
Platformio 2.2.0 Core 4.0.0

I download the ESPEasy zip from GitHub (version from 12th of august 2019) 

Atom Editor crashes when I try to open the ESPEasy folder. 
When I open the src folder it does not crash but does not know how to compile.

uzi18, 2019-08-12:
@s0170071 please use git to fetch repository

TD-er, 2019-08-12:
These crashes also happen every now and then on Atom in Windows.
I really got fed up with those, since it may take up-to 10 crash/restart cycles of Atom to get it to startup completely.
That's why I decided to move to VS code (also available in Linux).
And I should have done that much earlier, since I now have the option to navigate to "Go to Implementation" or "Go to Declaration"
[image]

Yesterday I have made a start of documenting using VS code (not yet merged)
See here: https://github.com/letscontrolit/ESPEasy/blob/16bcd6220d2d0e0319302cda3e64fdf76d610069/docs/source/Participate/PlatformIO.rst

uzi18, 2019-08-12:
@s0170071 @TD-er In facts VS Code works more reliable

tonhuisman, 2022-04-18:
This seems to be quite out-dated (and solved), so can be closed.

closed 2022-04-18 by TD-er