letscontrolit/ESPEasy #2545: Found that the same plugin cannot be reused

qiudecai, 2019-08-14:
Found that the same plugin cannot be reused

TD-er, 2019-08-14:
Can you be a bit more elaborate on this issue?
You didn't use the issue template and the title is the same as the issue description.

I have no clue what build, what plugin etc. you are using and also not a clue what you mean by "reused".

qiudecai, 2019-08-15:
sorry！
### list
  - [] plugin cannot be reused
  - [] Custom plugins can be used normally, multiple can not be used
  - []Energy - Soil_RS485 [TESTING]
  - []Release mega-20190809

### I have got...
  - [] Use the full filename of the version (eg `ESP_Easy_mega-20190809_test_ESP8266_4096_VCC.bin`)
  - [] gives a list of active plugins or controllers (if applicable).
  - [] Fill in all applicable fields below.

###Steps have been tried...
[image]
[image]
[image]
RS485 modbus_RTU
[image]

#ifdef USES_P199
//#######################################################################################################
//############################# Plugin 199: Soil_RS485          #########################################
//#######################################################################################################
/*

  Circuit wiring
    GPIO Setting 1 -> RX
    GPIO Setting 2 -> TX
    GPIO Setting 3 -> DE/RE pin for MAX485
    Use 1kOhm in serie on datapins!
*/

#define PLUGIN_199
#define PLUGIN_ID_199 199
#define PLUGIN_NAME_199 "Energy - Soil_RS485 [TESTING]"
#define PLUGIN_VALUENAME1_199 ""

#define P199_DEV_ID         PCONFIG(0)
#define P199_DEV_ID_LABEL   PCONFIG_LABEL(0)
#define P199_MODEL          PCONFIG(1)
#define P199_MODEL_LABEL    PCONFIG_LABEL(1)
#define P199_BAUDRATE       PCONFIG(2)
#define P199_BAUDRATE_LABEL PCONFIG_LABEL(2)
#define P199_QUERY1         PCONFIG(3)
#define P199_QUERY2         PCONFIG(4)
#define P199_QUERY3         PCONFIG(5)
#define P199_QUERY4         PCONFIG(6)
#define P199_DEPIN          CONFIG_PIN3

#define P199_NR_OUTPUT_VALUES   2
#define P199_QUERY1_CONFIG_POS  3

#define P199_QUERY_T       0
#define P199_QUERY_H       1

#define P199_NR_OUTPUT_OPTIONS   2  // Must be the last one

#define P199_DEV_ID_DFLT   1 //Modbus communication address
#define P199_MODEL_DFLT    0 // AcuDC24x
#define P199_BAUDRATE_DFLT 4 // 19200 baud
#define P199_QUERY1_DFLT   P199_QUERY_T
#define P199_QUERY2_DFLT   P199_QUERY_H

#define P199_MEASUREMENT_INTERVAL 60000L

#include <ESPeasySerial.h>

struct P199_data_struct : public PluginTaskData_base {
  P199_data_struct() {}

[... 203 lines cut ...]
      static_cast<P199_data_struct *>(getPluginTaskData(event->TaskIndex));
  if (nullptr != P199_data && P199_data->isInitialized()) {
    switch (query) {
    case P199_QUERY_T:
    {
      float intvalue = P199_data->modbus.readHoldingRegister(0); 
      if (intvalue < 0x8000){                      
        return intvalue = intvalue/10.0;
      }
        return intvalue = (0x8000 - intvalue)/10.0; 
    }    
    case P199_QUERY_H:
      return P199_data->modbus.readHoldingRegister(1)/10.0;

    }
  }
  return 0.0;
}

void P199_showValueLoadPage(byte query, struct EventStruct *event) {
  addRowLabel(Plugin_199_valuename(query, true));
  addHtml(String(P199_readValue(query, event)));
}

#endif // USES_P199

TD-er, 2019-08-15:
The P199 plugin is one from the playground.
The plugins in there are untested and not really maintained.
These are plugins that were once written by someone and some were never updated again, while a lot has changed in the ESPeasy core.

If I understand you well, you mean to run multiple instances of the same plugin active on a single ESP board?
That's a problem with a lot of plugins and I am working on adding those options to several plugins.
For some plugins (like Modbus) that may be quite some work to do, since Modbus does use the same pins for multiple devices. That does need a big change in code for all modbus related plugins, since I need to make a single modbus handler for this that can be used by all modbus plugins.
Also the Modbus_RTU implementation is still very much in "testing" phase.
As far as I know, only 3 plugins in the mega branch use Modbus RTU and those will be the first that will support multiple instances. As soon as that's working, I will migrate other plugins to be designed the same way.
But this may take a while to do so.

qiudecai, 2019-08-15:
Thank you, I hope to have a modbus_RTU plugin that supports multiple instances.
I think the problem lies in the same GPIO

uzi18, 2019-08-15:
@qiudecai what is your type/model of sensor? maybe you have photo or link where to get it?

uzi18, 2019-08-15:
> [image]
> RS485 modbus_RTU
> [image]

@TD-er @qiudecai pics shows something with receiving part is wrong

TD-er, 2019-08-15:
Well if this plugin is using the same GPIO pins, anything may happen.
The problem is that it may receive data from another device (or at least read the serial buffer containing data read from another device).

These plugins are just not written with running multiple instances of the same plugin in mind.
So if it does seem to work, it is just pure luck, since there is no locking, no checking etc. in there.
The plugins just assume that any returning data is an answer to their request, but in reality it is just not clear to what device request the reply belongs.

qiudecai, 2019-08-16:
I think the modbus_RTU plugin should be similar to I2C, with a fixed GPIO.

uzi18, 2019-08-16:
@qiudecai does attached file works for you (remove .txt extension)?
[P199_Soil.ino.txt](https://github.com/letscontrolit/ESPEasy/files/3508156/P199_Soil.ino.txt)

uzi18, 2019-08-16:
Now I see it is modified copy of https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P085_AcuDC243.ino

TD-er, 2019-08-16:
The AcuDC is a very recent plugin I made, actually the one that made me start working on making a more standard Modbus implementation for ESPeasy.
But that's what it is, a work-in-progress and surely not finished.
Those AcuDC modules are also quite expensive, so that's why I did not start making them "multi-instance" from start :)

uzi18, 2019-08-16:
@TD-er what about to create more generic plugin with all sets of registers for different devices? maybe also with custom field for future use?

TD-er, 2019-08-16:
That may be a good start, but still the Modbus part should then be made aware it can be shared among plugins.
So it must know what the last task number was to send out a command and whether it has been answered (and if there should be an answer).
Otherwise the data in the serial buffer may  be part of another request.

I think the current state of the modbus implementation I made is blocking, so it may not be an issue right now.  (Note that the Eastron plugin is using its own Modbus implementation.)
This also means a read call may take 200+ msec on some devices.
That's why I wanted to make the modbus part async, so we can have a handler which just has to be called every now and then to read the serial buffer and process its content.
Just like we do for MQTT by calling the loop().
The GPS plugin has a similar approach, which calls the loop() of tinyGPS library to read a char until the sentence is finished and I'm also doing the same with the new SerialProxy plugin.

But that's going to take quite a lot of changes I'm afraid.

uzi18, 2019-08-16:
I think it is not locking issue, there are some places not correct in my opinion - like crc checking/retransmission/etc.
Will try to use dumb modbus slave to reply with correct/wrong pkts.

qiudecai, 2019-08-16:
https://www.alibaba.com/product-detail/MEC10-soil-temperature-humidity-ec-sensor_60616355945.html?spm=a2700.7724838.2017115.1.7578dd4415VDbD&s=p

I use a sensor like this, which is RS485 Modbus_RTU, each sensor has a different Modbus address, the sensor has soil temperature and humidity, and has different register addresses. I modified the AcuDC24x plugin to get a new soil sensor plugin.

uzi18, 2019-08-16:
How did you changed adress for 2nd sensor?

qiudecai, 2019-08-16:
The plugin cannot modify the address. Modify the modbus address using USB to RS485

uzi18, 2019-08-16:
I understand it work for only one sensor properly? (if you disable 1st - 2nd will work, and if you disable 2nd - 1st will work)

TD-er, 2019-08-16:
> I understand it work for only one sensor properly? (if you disable 1st - 2nd will work, and if you disable 2nd - 1st will work)

And very likely the enabled one with the highest task number (lower in the table on the devices page) is the one that will show incorrect values.

And if you can modify the address using an USB RS485 module, then it should be possible to add that kind of functionality to the plugin.

uzi18, 2019-08-16:
@TD-er what do you think? :

```
diff --git a/src/Modbus_RTU.ino b/src/Modbus_RTU.ino
index 4a249f0d..7b253f32 100644
--- a/src/Modbus_RTU.ino
+++ b/src/Modbus_RTU.ino
@@ -464,19 +464,29 @@ struct ModbusRTU_struct  {
       unsigned long timeout = millis() + _modbus_timeout;
       bool validPacket      = false;
 
+      //  idx:    0,   1,   2,   3,   4,   5,   6,   7
+      // send: 0x02,0x03,0x00,0x00,0x00,0x01,0x39,0x84
+      // recv: 0x02,0x03,0x02,0x01,0x57,0xBC,0x2A
+
       while (!validPacket && _recv_buf_used < MODBUS_RECEIVE_BUFFER && !timeOutReached(timeout)) {
         while (easySerial->available()) {
           _recv_buf[_recv_buf_used++] = easySerial->read();
         }
 
-        // Check checksum
-        crc         = ModRTU_CRC(_recv_buf, _recv_buf_used);
-        validPacket = crc == 0;
+        if (_recv_buf_used > 2) {// got length
+          if (_recv_buf_used >= (3+_recv_buf[2]+2)) {// got whole pkt
+            // Check checksum
+            crc = ModRTU_CRC(_recv_buf, _recv_buf_used-2); // 0x3984
+            unsigned int rcv_crc = _recv_buf[_recv_buf_used-1];
+            rcv_crc += (((unsigned int) _recv_buf[_recv_buf_used-2])<<8);
+            validPacket = crc == rcv_crc;
+          }
+        }
         delay(0);
       }
 
       // Check for MODBUS exception
-      if (crc != 0u) {
+      if (!validPacket) {
         ++_reads_crc_failed;
         return_value = MODBUS_BADCRC;
       } else {

```

TD-er, 2019-08-16:
I don't get this part:
```
+            crc = ModRTU_CRC(_recv_buf, _recv_buf_used-2); // 0x3984
+            unsigned int rcv_crc = _recv_buf[_recv_buf_used-1];
+            rcv_crc += (((unsigned int) _recv_buf[_recv_buf_used-2])<<8);
+            validPacket = crc == rcv_crc;
```
The whole idea of the Modbus checksum is that computing the checksum over the entire packet should be 0.
As I understand it now, you're only computing it if enough data has been received.
But on top of that you also made the checksum validation more elaborate?
Why?

uzi18, 2019-08-16:
it is like send pkt - 2 last bytes are crc16sum
if you got recv buf like: 0x02,0x03,0x02,0x01,0x57,0xBC,0x2A
slave address is 2
function is 3
num of data bytes is 2
and after data bytes you got crc16sum one more time
now some python calculations:
```
>>> import libscrc
>>> hex(libscrc.modbus(bytes([0x02,0x03,0x00,0x00,0x00,0x01])))
'0x3984'
>>> hex(libscrc.modbus(bytes([0x02,0x03,0x00,0x00,0x00,0x01,0x39,0x84])))
'0x21b1'
```

so you can see crc is ok, but when you use 2 bytes more it is different ( not 0)

TD-er, 2019-08-16:
OK, I get it now, it was failing if there is more data in the serial buffer.
So marking the right start/end of the block to check is very likely a fix.
Only thing is, how do we make sure to get "in sync" again after missing a byte? (for example buffer overflow, not reading fast enough)

And it still isn't clear to me why the CRC checksum must be checked more elaborate?
If you just mark the right range, then it should be OK to compute it over that range and compare with 0, right?

Another dangerous thing, considering we may have more bytes in the buffer than expected:
```c++
        while (easySerial->available()) {
          _recv_buf[_recv_buf_used++] = easySerial->read();
        }
```

uzi18, 2019-08-16:
@TD-er this is idea to check we have whole pkt and corect one
we can also add check for function and slave same as in sent buffer

```

diff --git a/src/Modbus_RTU.ino b/src/Modbus_RTU.ino
index 4a249f0d..c0603a13 100644
--- a/src/Modbus_RTU.ino
+++ b/src/Modbus_RTU.ino
@@ -464,19 +464,28 @@ struct ModbusRTU_struct  {
       unsigned long timeout = millis() + _modbus_timeout;
       bool validPacket      = false;
 
+      //  idx:    0,   1,   2,   3,   4,   5,   6,   7
+      // send: 0x02,0x03,0x00,0x00,0x00,0x01,0x39,0x84
+      // recv: 0x02,0x03,0x02,0x01,0x57,0xBC,0x2A
+
       while (!validPacket && _recv_buf_used < MODBUS_RECEIVE_BUFFER && !timeOutReached(timeout)) {
-        while (easySerial->available()) {
+        while (easySerial->available() && !timeOutReached(timeout)) {
           _recv_buf[_recv_buf_used++] = easySerial->read();
         }
 
-        // Check checksum
-        crc         = ModRTU_CRC(_recv_buf, _recv_buf_used);
-        validPacket = crc == 0;
+        if (_recv_buf_used > 2) {// got length
+          if (_recv_buf_used >= (3+_recv_buf[2]+2)) {// got whole pkt
+            // Check checksum
+            crc = ModRTU_CRC(_recv_buf, _recv_buf_used);
+            validPacket = crc == 0;
+            return_value = 0; // reset return value
+          }
+        }
         delay(0);
       }
 
       // Check for MODBUS exception
-      if (crc != 0u) {
+      if (!validPacket) {
         ++_reads_crc_failed;
         return_value = MODBUS_BADCRC;
       } else {

```

uzi18, 2019-08-16:
@TD-er I think it was failing because not enough data in buffer, 9600 is not so fast

uzi18, 2019-08-16:
> Only thing is, how do we make sure to get "in sync" again after missing a byte? (for example buffer overflow, not reading fast enough)

we should flush all data after send request via serial->flush()

uzi18, 2019-08-16:
```
diff --git a/src/Modbus_RTU.ino b/src/Modbus_RTU.ino
index 4a249f0d..c0c065fd 100644
--- a/src/Modbus_RTU.ino
+++ b/src/Modbus_RTU.ino
@@ -458,25 +458,36 @@ struct ModbusRTU_struct  {
       startWrite();
       easySerial->write(_sendframe, _sendframe_used);
       startRead();
+      // drop all data from buffer
+      easySerial->flush();
 
       // Read answer from sensor
       _recv_buf_used = 0;
       unsigned long timeout = millis() + _modbus_timeout;
       bool validPacket      = false;
 
+      //  idx:    0,   1,   2,   3,   4,   5,   6,   7
+      // send: 0x02,0x03,0x00,0x00,0x00,0x01,0x39,0x84
+      // recv: 0x02,0x03,0x02,0x01,0x57,0xBC,0x2A
+
       while (!validPacket && _recv_buf_used < MODBUS_RECEIVE_BUFFER && !timeOutReached(timeout)) {
-        while (easySerial->available()) {
+        while (easySerial->available() && !timeOutReached(timeout)) {
           _recv_buf[_recv_buf_used++] = easySerial->read();
         }
 
-        // Check checksum
-        crc         = ModRTU_CRC(_recv_buf, _recv_buf_used);
-        validPacket = crc == 0;
+        if (_recv_buf_used > 2) {// got length
+          if (_recv_buf_used >= (3+_recv_buf[2]+2)) {// got whole pkt
+            // Check checksum
+            crc = ModRTU_CRC(_recv_buf, _recv_buf_used);
+            validPacket = crc == 0;
+            return_value = 0; // reset return value
+          }
+        }
         delay(0);
       }
 
       // Check for MODBUS exception
-      if (crc != 0u) {
+      if (!validPacket) {
         ++_reads_crc_failed;
         return_value = MODBUS_BADCRC;
       } else {
```

uzi18, 2019-08-16:
@TD-er flush() should be before startRead(), to be sure all data is sent, so we can switch to read
I'm unable to test it with real hardware now, but will create PR

qiudecai, 2019-08-24:
problem solved. In the plugin PLUGIN_READ. Reinitialize mudbus:
  Case PLUGIN_READ: {
     Const int16_t serial_rx = CONFIG_PIN1;
     Const int16_t serial_tx = CONFIG_PIN2;
     P199_data_struct *P199_data =
     Static_cast<P199_data_struct *>(getPluginTaskData(event->TaskIndex));
     P199_data->init(serial_rx, serial_tx, P199_DEPIN, P199_storageValueToBaudrate(P199_BAUDRATE), P199_DEV_ID); //init Mudbus
     If (nullptr != P199_data && P199_data->isInitialized()) {
       For (int i = 0; i < P199_NR_OUTPUT_VALUES; ++i) {
         UserVar[event->BaseVarIndex + i] = P199_readValue(PCONFIG(i + P199_QUERY1_CONFIG_POS), event);
         Delay(1);
       }
       Success = true;
     }
     Break;
   }

qiudecai, 2019-08-24:
[image]

uzi18, 2019-08-24:
It is partial solution here - it forces delete and create serial object every time

qiudecai, 2019-08-24:
Yes! I think the best solution still hopes to be like I2C. Have a fixed GPIO

TD-er, 2019-08-24:
That's quite a heavy fix.
Can you show me the timing stats of this plugin with and without this fix? (just use 1 instance as test)
See the timing stats page for a detailed list of collected statistics.
Preferably with at least 5 samples of this read command (each time you load the timing stats page it will reset the statistics)

qiudecai, 2019-08-25:
PLUGIN_READ consumes a lot of time
[image]

uzi18, 2019-08-25:
and now check without this "init" call in PLUGIN_READ

TD-er, 2019-08-25:
It also depends on how much you try to read or have to wait between sending a command and retrieving its result.
For example the SenseAir plugin is also using too much time (too much to my liking), especially if you try to read 2 measurements (e.g. internal temp + CO2)

Description | Function | #calls | call/sec | min (ms) | Avg (ms) | max (ms)
-- | -- | -- | -- | -- | -- | --
P_52_Gases - CO2 Senseair | READ | 4921 | 0.20 | 420.833 | 421.937 | 433.806

So that's for sure one thing I like to address, but then we need to make a central Modbus controller helper.

tonhuisman, 2023-09-03:
Nearly all plugins now have support for multiple instances, and those that do not support it yet, probably have reasons it's not needed, so this issue can be closed.

closed 2023-09-03 by TD-er