letscontrolit/ESPEasy #2547: ITRX fails due to interrupts @ IRSENDAC

s0170071, 2019-08-15:
@TD-er @jimmys01  

After two full days of debugging, I found a glitch in the IRTX plugin when using IRSENDAC. Yea I have seen the comment that its preliminary test code :-)
Problem is that interrupts kick in at the very moment the IR code is sent out. This messes up the timing. 

As a fix you may want to temporarily disable them.

```
      noInterrupts();
      bool retval = Plugin_035_commonAc->sendAc(protocol, model, power, opmode, degrees, celsius, fanspeed, swingv, swingh, quiet, turbo, econo, light, filter, clean, beep, sleep, clock);
      interrupts();

```

Note: this does not happen in the example of the IR library as they to not use wifi.
Note2: IRSEND is most likely affected too, although I did not test that.

Here is the dump of what I measured using IRrecvDumpV2, first with interrupts enabled, then disabled. 

```
Timestamp : 000626.808
Library   : v2.6.4

Encoding  : UNKNOWN
Code      : 36342C81 (132 bits)
Raw Timing[263]:
   +   448, - 25180,    +  3332, -  1588,    +   466, -   378,    +   492, -   352, 
   +   466, -  1170,    +   462, -   378,    +   464, -  1170,    +   464, -   378, 
   +   462, -   378,    +   466, -   382,    +   460, -  1170,    +   466, -  1170, 
   +   466, -   374,    +   466, -   378,    +   466, -   378,    +   462, -  1170, 
   +   466, -  1170,    +   466, -   378,    +   462, -   378,    +   466, -   378, 
   +   462, -   378,    +   466, -   378,    +   466, -   378,    +   462, -   378, 
   +   466, -   378,    +   460, -   378,    +   466, -   378,    +   466, -   378, 
   +   460, -   378,    +   466, -   378,    +   466, -  1170,    +   460, -   378, 
   +   466, -   378,    +   466, -   378,    +   460, -   378,    +   466, -   378, 
   +   462, -   378,    +   466, -   378,    +   466, -  1170,    +   460, -   378, 
   +   466, -   378,    +   466, -   378,    +   460, -   378,    +   466, -  1170, 
   +   466, -  1170,    +   460, -  1170,    +   492, -  1144,    +   466, -  1170, 
   +   462, -  1170,    +   466, -  1170,    +   466, -  1170,    +   462, -   378, 
   +   466, -   378,    +   460, -  1174,    +   488, -   352,    +   466, -   378, 
   +   460, -   378,    +   466, -   376,    +   466, -   378,    +   462, -   378, 
   +   466, -   378,    +   466, -   378,    +   462, -  1170,    +   466, -  1170, 
   +   462, -   382,    +   462, -   378,    +   466, -   378,    +   466, -   378, 
   +   460, -   378,    +   466, -   378,    +   460, -   382,    +   460, -   378, 
   +   466, -  1168,    +   462, -   382,    +   460, -  1170,    +   466, -  1170, 
   +   462, -   382,    +   462, -   376,    +   466, -   378,    +   462, -   378, 
   +   466, -   378,    +   462, -   382,    +   462, -   378,    +   466, -   376, 
   +   462, -  1174,    +   462, -   378,    +   464, -   378,    +   462, -   382, 
   +   462, -   378,    +   466, -   378,    +   462, -   382,    +   462, -   378, 
   +   466, -   378,    +   462, -   378,    +   466, -   376,    +   462, -   382, 
   +   462, -   378,    +   466, -   378,    +   462, -   382,    +   462, -   378, 
   +   466, -   378,    +   462, -   382,    +   462, -   378,    +   462, -   382, 
   +   460, -   404,    +   438, -   378,    +   462, -   382,    +   462, -   404, 
   +   440, -   378,    +   462, -   404,    +   440, -   402,    +   436, -   382, 
   +   462, -   404,    +   438, -   378,    +   462, -   382,    +   460, -   404, 
   +   438, -   378,    +   460, -   408,    +   434, -   404,    +   434, -  1174, 
   +   460, -   404,    +   440, -  1196,    +   440, -  1170,    +   460, -   404, 
   +   440, -   404,    +   436, -  1200,    +   436, -   404,    +   440, -  1196, 
   +   436, -  1200,    +   434, -  1200,    +   436, - 42636,    +   558

uint16_t rawData[263] = {448, 25180,  3332, 1588,  466, 378,  492, 352,  466, 1170,  462, 378,  464, 1170,  464, 378,  462, 378,  466, 382,  460, 1170,  466, 1170,  466, 374,  466, 378,  466, 378,  462, 1170,  466, 1170,  466, 378,  462, 378,  466, 378,  462, 378,  466, 378,  466, 378,  462, 378,  466, 378,  460, 378,  466, 378,  466, 378,  460, 378,  466, 378,  466, 1170,  460, 378,  466, 378,  466, 378,  460, 378,  466, 378,  462, 378,  466, 378,  466, 1170,  460, 378,  466, 378,  466, 378,  460, 378,  466, 1170,  466, 1170,  460, 1170,  492, 1144,  466, 1170,  462, 1170,  466, 1170,  466, 1170,  462, 378,  466, 378,  460, 1174,  488, 352,  466, 378,  460, 378,  466, 376,  466, 378,  462, 378,  466, 378,  466, 378,  462, 1170,  466, 1170,  462, 382,  462, 378,  466, 378,  466, 378,  460, 378,  466, 378,  460, 382,  460, 378,  466, 1168,  462, 382,  460, 1170,  466, 1170,  462, 382,  462, 376,  466, 378,  462, 378,  466, 378,  462, 382,  462, 378,  466, 376,  462, 1174,  462, 378,  464, 378,  462, 382,  462, 378,  466, 378,  462, 382,  462, 378,  466, 378,  462, 378,  466, 376,  462, 382,  462, 378,  466, 378,  462, 382,  462, 378,  466, 378,  462, 382,  462, 378,  462, 382,  460, 404,  438, 378,  462, 382,  462, 404,  440, 378,  462, 404,  440, 402,  436, 382,  462, 404,  438, 378,  462, 382,  460, 404,  438, 378,  460, 408,  434, 404,  434, 1174,  460, 404,  440, 1196,  440, 1170,  460, 404,  440, 404,  436, 1200,  436, 404,  440, 1196,  436, 1200,  434, 1200,  436, 42636,  558};  // UNKNOWN 36342C81

Timestamp : 001036.643
Library   : v2.6.4

Encoding  : UNKNOWN
Code      : 811FFBAE (7 bits)
Raw Timing[13]:
   +   102, - 48106,    +   330, - 43476,    +   330, -  1996,    +   462, - 48352, 
   +   104, - 20246,    +   104, - 41046,    +   270
[... 23 lines cut ...]
   +   492, -  1138,    +   466, -   376,    +   492, -   352,    +   466, -   374, 
   +   466, -   378,    +   492, -   352,    +   464, -   374,    +   466, -   376, 
   +   466, -   374,    +   466, -  1170,    +   492, -  1144,    +   466, -   374, 
   +   466, -   378,    +   492, -   350,    +   466, -   372,    +   466, -   378, 
   +   500, -   342,    +   466, -   374,    +   464, -   378,    +   464, -  1166, 
   +   466, -   376,    +   492, -  1142,    +   466, -  1166,    +   466, -   378, 
   +   490, -   352,    +   466, -   374,    +   466, -   378,    +   492, -   352, 
   +   462, -   378,    +   466, -   378,    +   492, -   352,    +   462, -  1168, 
   +   492, -   352,    +   466, -   372,    +   466, -   378,    +   466, -   378, 
   +   462, -   378,    +   464, -   378,    +   466, -   378,    +   462, -   378, 
   +   466, -   378,    +   466, -   374,    +   466, -   378,    +   466, -   378, 
   +   464, -   374,    +   466, -   378,    +   466, -   378,    +   462, -   378, 
   +   466, -   378,    +   466, -   378,    +   462, -   378,    +   466, -   378, 
   +   466, -   378,    +   462, -   378,    +   492, -   352,    +   466, -   374, 
   +   466, -   378,    +   466, -   378,    +   462, -   378,    +   492, -   352, 
   +   466, -   378,    +   460, -   378,    +   466, -   378,    +   466, -   378, 
   +   460, -   378,    +   466, -   378,    +   466, -  1166,    +   466, -   378, 
   +   466, -  1170,    +   462, -  1170,    +   466, -   378,    +   466, -   378, 
   +   460, -  1170,    +   466, -   378,    +   466, -  1170,    +   462, -  1170, 
   +   466, -  1174,    +   466

uint16_t rawData[259] = {3364, 1566,  470, 372,  466, 378,  492, 1138,  470, 374,  492, 1142,  492, 346,  466, 378,  492, 352,  492, 1138,  466, 1170,  492, 352,  490, 348,  466, 378,  492, 1144,  466, 1164,  466, 378,  492, 346,  466, 378,  492, 352,  466, 372,  466, 378,  492, 348,  470, 374,  492, 352,  492, 348,  466, 378,  492, 352,  466, 374,  466, 1170,  492, 350,  492, 348,  466, 378,  492, 352,  466, 374,  466, 378,  492, 352,  466, 1164,  466, 378,  466, 374,  466, 378,  492, 352,  492, 1142,  466, 1166,  492, 1144,  492, 1144,  464, 1166,  466, 1170,  492, 1144,  466, 1164,  466, 378,  492, 352,  492, 1138,  466, 376,  492, 352,  466, 374,  466, 378,  492, 352,  464, 374,  466, 376,  466, 374,  466, 1170,  492, 1144,  466, 374,  466, 378,  492, 350,  466, 372,  466, 378,  500, 342,  466, 374,  464, 378,  464, 1166,  466, 376,  492, 1142,  466, 1166,  466, 378,  490, 352,  466, 374,  466, 378,  492, 352,  462, 378,  466, 378,  492, 352,  462, 1168,  492, 352,  466, 372,  466, 378,  466, 378,  462, 378,  464, 378,  466, 378,  462, 378,  466, 378,  466, 374,  466, 378,  466, 378,  464, 374,  466, 378,  466, 378,  462, 378,  466, 378,  466, 378,  462, 378,  466, 378,  466, 378,  462, 378,  492, 352,  466, 374,  466, 378,  466, 378,  462, 378,  492, 352,  466, 378,  460, 378,  466, 378,  466, 378,  460, 378,  466, 378,  466, 1166,  466, 378,  466, 1170,  462, 1170,  466, 378,  466, 378,  460, 1170,  466, 378,  466, 1170,  462, 1170,  466, 1174,  466};  // FUJITSU_AC
uint8_t state[16] = {0x14, 0x63, 0x00, 0x10, 0x10, 0xFE, 0x09, 0x30, 0x40, 0x03, 0x04, 0x00, 0x00, 0x00, 0xA0, 0xE9};

```

jimmys01, 2019-08-15:
Hmm it seems so in yor setup.. In my setups i condrol way a lot AC and I do not have issues. Please add the exact steps you do so maybe I can reproduse the bug. Also the plugins you use etc.
If @TD-er belives that doing the        `noInterrupts();      interrupts();` wont affect anything else we can add it as a precation anyway.
Thanks for your inputs

TD-er, 2019-08-15:
Well disabling interrupts certainly does something :)
How long do these sequences take?
I guess longer than 10 usec.
If you disable the interrupts longer than such a period, WiFi may act strange.
Also if these are needed for this plugin, then maybe we should stress not to use other plugins on the same node.

s0170071, 2019-08-15:
my setting is this: 
plain new Wemos D1 with Ir transmitter at D2 / gpio4
The only device is "Communication - IR Transmit"
Serial debug output enabled.

I received the IR burst with a second unit with IRrecvDumpV2.ino installed, also tried with my air conditioner. 

I then issue the command
`http://192.168.1.33/control?cmd=IRSENDAC,{"power":"","temp":"18","protocol":"FUJITSU_AC","model":"3","mode":"AUTO", "swingv":"OFF", "swingh":"OFF", "turbo":"OFF", "econo":"OFF", "fanspeed":"AUTO"}
`
Sometimes it used to work with interrupts enabled but this is very unreliable. Seems to be wifi related.

@TD-er disabling interrupts for more than 10us is OK if there is not much traffic going on.   The wifi stack can buffer a packet or two before it drops them. I have not measured the burst length but its a couple of ms for sure.

> Also if these are needed for this plugin, then maybe we should stress not to use other plugins on the same node.
only those that use interrupts like pulse.

s0170071, 2019-08-15:
@jimmys01 it may be that the bug only appears if you use http to issue the command as there are more wifi frames, especially the http response. This is not the case for serial commands and mqtt.

jimmys01, 2019-08-15:
For anyone that wants to try it with the interupts enabled to` sendir` and `sendirac `commands 
[normal_IRext_ESP8266_4M_With_interupts.zip](https://github.com/letscontrolit/ESPEasy/files/3505336/normal_IRext_ESP8266_4M_With_interupts.zip)

Also I made a change to send all the Stirng parameters to the IR library to the UPPER case, even thought it 'should' not make any diference. If you are kind enough to test this for me and i will push a PR asap.

For reference the IR function that checks those strings is like

```
bool IRac::strToBool(const char *str, const bool def) {
  if (!strcasecmp(str, "ON") || !strcasecmp(str, "1") ||
      !strcasecmp(str, "YES") || !strcasecmp(str, "TRUE"))
    return true;
  else if (!strcasecmp(str, "OFF") || !strcasecmp(str, "0") ||
           !strcasecmp(str, "NO") || !strcasecmp(str, "FALSE"))
    return false;
  else
    return def;
}
```

strcasecmp is not case sensitive.

s0170071, 2019-08-16:
@jimmys01 right, the ON is case corrected by the library. 

The parameter name, however, was not case insensitive in the SENDAC branch. It is checked by the plugin code. So it has to be "power":"on" , not "Power":"on". Did you fix this as well ?

jimmys01, 2019-08-16:
The JSON keys are case sensitive and the validator is implemented that way based on spec. The JSON library does not support case insesitive keys for that reason. So this cannot be fixed and not make it a hack 'fix'.

I have corrected the plugin use instractions in the top of the  P035 ino file and I will make a PR when I have other updates to do as well.

jimmys01, 2019-08-26:
@TD-er is there a way for us to detect if an iterupt has occudred while we are sending an IR signal, so we can attempt a  re send?

TD-er, 2019-08-26:
Not if the interrupts are disabled during sending (what I assume is needed to get a stable data stream)

jimmys01, 2019-08-26:
The thinking is not to disable the interupts during the signal transmision. Is to detect that an interupt happened during the transmision (so it messed up the transmision) and re send it.

TD-er, 2019-08-26:
That should be possible. Just attach an ISR function to the GPIO pin you need to monitor and only switch a boolean (volatile declared) to true if such an interrupt is happening.
Only thing I am thinking about what may go wrong here, is that you also need to detach the interrupt if you don't want to interrupt receiving data.
But I guess the IR library already is having an interrupt attached to the GPIO pin?
So maybe you can let that lib return some "received something" flag, which automatically resets the flag by reading it.
I think that's the proper place to have this handled, since that library can then evaluate the impact of such handling.
Maybe @crankyoldgit can share some profound insight on this?

jimmys01, 2019-08-26:
The RX plugin get disabled while the TX is transmitting

```
#ifdef PLUGIN_016
      if (irReceiver != 0)
        irReceiver->disableIRIn(); // Stop the receiver
#endif
```

TD-er, 2019-08-26:
Well then you could attach another interrupt to that pin to detect any activity.
But just make sure you're not trying to re-send just because you're seeing the data you are sending.
You could increase some counter instead of a flag to help detect it, but I think this may affect the 'quality' of your IR data stream..

Just as an idea, can you also mis-use the Serial1 port for this?
I don't know if you can set an arbitrary baudrate, nor if you can have a zero-usec-gap between bytes. But if that's possible, then you have a hardware buffer to send out data to an GPIO pin.
The size of that buffer is 128 bytes, so I guess long enough to send 1024 bit. Just assuming here that the encoding is not using PWM.

crankyoldgit, 2019-08-26:
> Note: this does not happen in the example of the IR library as they to not use wifi.

FYI, there are plenty of examples in the IR library that use wifi. This is the first problem I've ever heard of for interrupt activity affecting sending an IR message. Not saying this can't be the first, but I'd be wary of blaming interrupts for this based on past experience/evidence.

> How long do these sequences take?
I guess longer than 10 usec.

Correct. You're in the 100's of msecs magnitude. Disabling interrupts would be a bad idea\<tm>.

Now on to the issue...

Looking at the original data, i.e. the entry that starts with:
```
Encoding  : UNKNOWN
Code      : 36342C81 (132 bits)
Raw Timing[263]:
```

There is garbage at the beginning of the message and at the end of the message, but using my human eye, nothing altering the timing of the actual message. e.g. the first two entries in the array are "bad" then there appears a valid looking (to my eye) Fujitsu message, followed by two more "bad" entries
If an interrupt did fire during transmission, I'd expect the "space" pulse timings during the message to be longer than usual **OR** for a "mark" pulse to be split by a gap of no transmission. I see no signs of that.

There seems to be semi-regular "noise" as indicated by the capture log:
```
Encoding  : UNKNOWN
Code      : 811FFBAE (7 bits)
Raw Timing[13]:
   +   102, - 48106,    +   330, - 43476,    +   330, -  1996,    +   462, - 48352, 
   +   104, - 20246,    +   104, - 41046,    +   270
```
e.g. a "pulse" of approx 102 usecs, and then a gap of 48106 usecs .. and then a similar scenario of 330 & 43476 and then 462 & 48352. That's awfully close to 50msecs. Is anything happening at that frequency? i.e. ~20Hz.

That's sort of similar to what is at the start of the "Code      : 36342C81 (132 bits)" message.

I'd recommend reducing the capture timeout time for the IR receiving to 25msec (25000usec) or similar .. or even less like 15msecs. or try to find out what is causing that noise.

The TL;DR: of what I am saying and what the data shows me (without formally verifying it in code) is I see no evidence that there is some foreign interrupt causing the transmitted IR message to be garbled.
If it was, I'd expect the timing sequences to be "off" during the message. The data shows something is producing IR "data"/noise before (and after) the library is sending the message.

Now, that doesn't exclude the possibility that ESPEasy could be making some programming mistake before or after the IR message is sent via the library, but I think your hypothesis of "some interrupt" is happening during message transmission and that's causing a problem is probably not the root cause.

I'd suggest you try to repeat the debugging in a IR quiet location. i.e. at night, in a cool environment and no other devices. i.e. sunlight, fluorescent tubes, or other IR is bad. It could also be noise in your circuit too. If the same symptoms persist, then I'd look at the specific code "around" calling the library just in case it could be causing some erroneous IR messages to be sent etc.

TD-er, 2019-08-26:
> e.g. a "pulse" of approx 102 usecs, and then a gap of 48106 usecs .. and then a similar scenario of 330 & 43476 and then 462 & 48352. That's awfully close to 50msecs. Is anything happening at that frequency? i.e. ~20Hz.

I don't know about the schematics of that board, nor the quality of the received signal from the IR receiver.
But there is WiFi activity about every 102.4 msec (half the noted 20 Hz), when the node does try to listen to WiFi beacon intervals.
I am not sure how autonomous these are being handled in the background, but I do know that some received data may cause memory allocations on the heap.
For example received UDP traffic will be kept in the heap and should be handled ASAP (or else memory will fill up).
So it could be that such background stuff is either slowing down the ESP or sudden change in power consumption could maybe flip a bit on a signal line?
But I do think the non-realtime operations of the ESP are very likely to be the cause here.

You could try to disable the ESPeasy p2p traffic for that node (set UDP port to 0) to make sure it has less UDP traffic to handle.

TD-er, 2019-08-27:
> The library's IR transmission doesn't use any heap allocation, so I think that's a red herring.

What I meant to say was that memory allocation alone does take time and can cause timing issues if timing really needs to be strict.

crankyoldgit, 2019-08-27:
> I am not sure how autonomous these are being handled in the background, but I do know that some received data may cause memory allocations on the heap.
> For example received UDP traffic will be kept in the heap and should be handled ASAP (or else memory will fill up).

The library's IR transmission doesn't use any heap allocation, so I think that's a red herring.

> So it could be that such background stuff is either slowing down the ESP or sudden change in power consumption could maybe flip a bit on a signal line?

That is a possibility, but a low probability, hence my feeling is it is best to "debug" other areas first. To produce accidental "noise" in the IR signal like that would confuse the receiver's IR demodulator, it would have to happen at around a freq. of 38kHz to confuse the hardware. e.g. It's detecting a "mark" signal of 100+usecs, that's approx 4 pulses (26.3us PWM period) at 38kHz. More unlikely the longer the "mark" signal. I feel that's unlikely to be triggered by pure happen-stance. Also the user has effectively decoupled sending and receiving by using two different ESP modules.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

crankyoldgit, 2023-09-03:
Just adding a note. We (IRremoteESP8266 library) have noticed some issues with _single_ CPU ESP32 chips and WIFI interrupts causing timing issues in IR transmissions from the ESP.

The real fix is for our library to use the RMT feature of the ESP32s when transmitting but that is a lot of work and a low-ish priority. i.e. I'd like to fix it, but I don't have the spare time.

tonhuisman, 2023-09-03:
@crankyoldgit Thanks for the update 👍

TD-er, 2023-09-04:
The reason it doesn't affect ESP8266 could be because ESP8266 is using NONOS, while ESP32 variants use RTOS.
RTOS does perform scheduling for RTOS tasks of which the "sketch" is just one of those tasks.
On dual core ESP32-variants, the RTOS task running WiFi is running on the other core and the other tasks running on the same core as the "sketch" is running are less likely to cause interrupts.

When considering using RMT for sending IR data (which I think is the best solution), please be aware not all ESP32-variants seem to support it.
See: https://espeasy.readthedocs.io/en/latest/ESPEasy/ESPchips.html
At least the ESP32-C2 (ESP8684) doesn't seem to support it.
I don't understand the use case of this C2 as it is by far the least interesting one. But it probably will be cheap and thus likely we will find it in lots of commercially made devices.

Anyway, I think we can indeed close this issue, as further fixing should be done in the IRremoteESP8266 library and not in ESPEasy.

closed 2023-09-04 by TD-er