letscontrolit/ESPEasy #2548: Add documentation for IRTX, fix upper/lower case

s0170071, 2019-08-15:
at least add a comment to the code that the command is like this. 

http://192.168.1.33/control?cmd=IRSENDAC,{"power":"1","temp":"20","protocol":"FUJITSU_AC","model":"3","mode":"FAN", "swingv":"OFF", "swingh":"OFF", "turbo":"OFF", "econo":"OFF", "fanspeed":"MIN"}

At this time, the command parameters need to be lower case, the values upper case. 
Please add toUpperCase(); to IRSENDAC branch.

jimmys01, 2019-08-15:
The upper case stuff are suposed to be handled by the IR library. The "power":"1" should be the same like "power":"on" and with "power":"ON". I will test this myself the following days.

jimmys01, 2019-08-15:
By the way if you use the IR RX plugin and a IR receiver you can deduse the IRSENDAC command for your remote. This is in my knowledge the only fast way to find out if your AC needs the "model" parameter to the command as well.

jimmys01, 2019-08-15:
To scratch my itcing curiosity i tried this:

        tempstr = "on";
        Serial.println(IRac::strToBool(tempstr.c_str()));
        tempstr = "ON";
        Serial.println(IRac::strToBool(tempstr.c_str()));
        tempstr = "oN";
        Serial.println(IRac::strToBool(tempstr.c_str()));

Its output was:

```
1
1
1
```

So at least the "power" should be fine even with "oN" as an input.

jimmys01, 2019-08-28:
this can be closed

closed 2019-08-28 by TD-er