letscontrolit/ESPEasy #2550: problem with Generic HTTP Advanced [TESTING]

maxmib, 2019-08-17:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [* ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [* ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [* ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ *] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [* ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ *] Tested previous/other build (mention which one already tested)
- [* ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

After ESP_Easy_mega-20190311 version, the "http method" setting in Generic HTTP Advanced [TESTING] doesn't work anymore. "Http method " setting can only be "get" . Changes about this setting can't be saved.
### Expected behavior

Change http method from "get" to "post" or any other option.

### Actual behavior

http method in Generic HTTP Advanced [TESTING] can't be changed .

### Steps to reproduce

1. edit Generic HTTP Advanced [TESTING controller 
2. change http method from get to post
3. submit

### System configuration

Hardware:
esp8266 nodemcu

ESP Easy version: 
Build:⋄
20103  - Mega
System Libraries:⋄
ESP82xx Core 2_5_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.1.2 PUYA support
Git Build:⋄
mega-20190817
Plugins:⋄
81 [Normal] [Testing] [Development]
Build Time:⋄
Aug 17 2019 02:40:11
Binary Filename:⋄
ESP_Easy_mega-20190817_dev_ESP8266_4M.bin

ESP Easy settings/screenshots: 

### Rules or log data

```

```

popaserge, 2019-08-27:
@TD-er I confirm the bug
Seems that there is a regression from previous version from march(see #1970 )
I used
Firmware
Build:⋄	20103 - Mega
System Libraries:⋄	ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support
Git Build:⋄	mega-20190315
Plugins:⋄	77 [Normal] [Testing] [Development]
Build Md5:	8a634edb71d6a875645851fb62f4b6f3
Md5 check:	passed.
Build Time:⋄	Mar 15 2019 03:29:42
Binary Filename:⋄	ESP_Easy_mega-20190315_dev_ESP8266_4M.bin

This gave following web page for the data of the controller
[image]

I made an update with last release
Firmware
Build:⋄	20103 - Mega
System Libraries:⋄	ESP82xx Core 2_5_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.1.2 PUYA support
Git Build:⋄	mega-20190823
Plugins:⋄	81 [Normal] [Testing] [Development]
Build Md5:	fdc320d17ab522e1ccdde1e0b7f543de
Md5 check:	passed.
Build Time:⋄	Aug 23 2019 02:35:31
Binary Filename:⋄	ESP_Easy_mega-20190823_dev_ESP8266_4M.bin

The web page has changed to
[image]

No possibility to change the type of request which always is GET.
My actual requests were POST and do no more give any result to the server.
The fields for Header and Body are only one line.
It is very inconvenient if your body has several lines: you cannot have a look on the entire body to check or modify something.
Of course this behaviour is the same if you try to do your own build with only few plugins.

TD-er, 2019-08-27:
Ah found the problem.
The plugin_load function is assuming a String object ("GET", "POST" etc.) but if you try to save it, it is only sending an integer value.  (see parameter `P011httpmethod` )

[image]

So it does not save the method correctly.

TD-er, 2019-08-27:
Seems like it is broken in this change of mine: https://github.com/letscontrolit/ESPEasy/commit/89a49a3355fc6901d57c7e2465e22175277d2892#diff-1a50d04dd766628c7effc50f21f0cf63

popaserge, 2019-08-29:
@TD-er
I am sorry but I could not do anything with that information because I am not acquainted with the structure of EspEasy and with coding, but would be very interested to try a solution.

TD-er, 2019-08-29:
@popaserge Last few days I had not yet time to fix this one.
These comments I made were just a note to myself (or one of the few others sometimes fixing stuff) about what's going on.

popaserge, 2019-08-29:
@TD-er
OK understood Thanks

uzi18, 2019-08-29:
@popaserge no worry, fix is ready

PR #2582 (2019-09-02): - [C011] fix method save #2550
@TD-er tested and ready to merge, it fix method save #2550

Fixes #2550
closed 2019-09-02 by TD-er