letscontrolit/ESPEasy #2554: Microsoft Visual Studio Code. (MS VS-Code)

fly74, 2019-08-17:
why the binaries build with this doc [https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html) are bigger than in your zip file without any code change ?

TD-er, 2019-08-17:
What binaries are you comparing?
You have seen that there are lots of environment options to chose when building?
You can also check on a running version how many plugins are available in the build.

I do use exactly that platformio.ini file as in the mega branch to create a nightly build.
Only thing that may be different is that the nightly builds are made on a Linux system.

Another thing that may have an effect is the directory in which you are building.
Not all debug information is stripped from the builds, so the full path of the source files is also present in the build.
You may also want to test a simple build from a very short path like C:\test\.
This difference should be one or two kByte max. I believe, unless you're using a very long path.

fly74, 2019-08-17:
ESP_Easy_mega-20190817_custom_ESP8266_4M

[image]

VS Plugins:
[image]

Difference 60 KB:
[image]

fly74, 2019-08-17:
ESP_Easy_mega-20190817_normal_ESP8266_4M have the same size, maybe only custom_4M.

closed 2019-08-17 by fly74