letscontrolit/ESPEasy #2556: Support of ADS1235

R1F800, 2019-08-19:
Hi.
in Test Version support of an HX711 is included. 
Instead of this I want to use established ADC for an beehive weightcell.
But there is only support of the ADS1115 ADC.

Is there any chance to include a new ADS1235 for a higher bit resolution?
THX

uzi18, 2019-08-19:
do you have more details?
how you want to connect it to weightcell
both ADSxxxx have different interfaces

R1F800, 2019-08-19:
HI. Sorry made a mistake !!  It is ADC1234 or 1231.
I am not shure, what / how i May give you more Information.

A BOSCHE H40A-C3-0150  sould be connected to Analog IN an ( equal as ADS1115 but with higher Resolution 24 bit instead of 16)

uzi18, 2019-08-19:
where to find/buy such module? can't find documentation for it.

R1F800, 2019-08-19:
Datasheet:
**ADS1234**
[](http://www.ti.com/product/ADS1234)

**ADS1232**
[](https://www.ti.com/lit/ds/symlink/ads1232.pdf)

[](https://www.aliexpress.com/item/32895565811.html?spm=a2g0o.productlist.0.0.41186bedEAeclb&algo_pvid=95825122-833a-46bb-a870-78bfd6883ebf&algo_expid=95825122-833a-46bb-a870-78bfd6883ebf-14&btsid=645b0bc6-1c31-41e5-94eb-fd621728cde5&ws_ab_test=searchweb0_0,searchweb201602_7,searchweb201603_52)

Maybe availability of 1232 is better than 1234

uzi18, 2019-08-19:
what about ADS1256

R1F800, 2019-08-19:
Well I think it is a GAIN issue.  ADS1234 / 1232 has an GAIN of 128 (MAX)

ADS1256 GAIN of 64 (MAX)

and I think we don´t Need 30k SPS instead of GAIN ..

R1F800, 2019-12-04:
Are there any new Information About support of ADC1231 in ESPeasy  instead/beside ADS1115 ?

R1F800, 2019-12-04:
@TD-er
PLEASE Change TYPE:QUESTION >>> TYPE:FEATURE REQUEST

TD-er, 2019-12-04:
I have not done anything for this request, simply because other fixes and stability issues were taken all of my time.

uzi18, 2019-12-05:
First of all weight cell is not a good idea for long term measurments.
If can suggest, please check this project: http://www.openhivescale.org/ and write here what do you think about it.

R1F800, 2019-12-05:
Because of other implemented Services at my home Automation this kind of technical solution will not fit.
I am working at a Project www.hiveeyes.org. So there are as many solutions as requirements...

My requirement is to integrate this data with ESPeasy to my implemented home Automation (FHEM / ESPeasy - data generating layer)   
ETL will be next step at grafana or something else....

uzi18, 2019-12-05:
```
Why a mechanical scale, and not using load cells?

Because load cell are not supposed to be under continuous load

    Typical use case implies a tare between each measurement, not possible in hive monitoring scenario
    A creep phenomen make measurement unstable after some time, this is well documented, that we check on our first prototype
    Mechanical scale is not subject to this phenomen. Even if structure creeps, measurement wouldn’t change.

https://dipot.ulb.ac.be/dspace/bitstream/2013/146415/1/RAM_HBM.pdf

“Transducer manufacturers provide only information on the short-time stability like maximum creep after 30 minutes or maximum 1 hour under full load, but appear very reticent to provide figures on the long-term stability of their products……

Recording strains over long periods of time without having the opportunity to unload the test specimen or the structure and check zero reference is considered to be one of the most difficult applications of strain-gauges….”

http://www.analog.com/en/analog-dialogue/articles/a-reference-design-for-weigh-scales.html

Because we don’t depend on any complicated part that we must outsource on the mechanical part (motor, optical endstop)

    This makes possible to keep cost low
    Easier to understand what could go wrong and repair
    DIY-ers friendly

Electronic is much simpler : no analog layer on which measurement quality depends
```

In my opinion is just firmware adaptation.

About ADS1231/32/34/35 they have SPI  like interface where ADS1115 is I2C. 
So ... we need another plugin for this or we need some unification in core to support it in one, right @TD-er ? Ideas?

TD-er, 2019-12-05:
My first idea is indeed that change of communication layer (I2C/SPI/UART/etc) is potentially a big change so it will make a plugin quite complex to support both.
Thus if it is using a different comm. layer, I would suggest to start a new plugin for it.

A lot of chips do support SPI and I2C, just by pulling some pins low or high using resistors.
If the chip you need can be set to I2C and still meets your requirements regarding data speed etc. then it might be possible to extend support for the other chips in the same plugin.
As long as the chips do share a lot of the logic to interface with them (e.g. same kind of registers to read, similar way of interpreting the data etc.), then it makes sense to have a plugin support multiple chips.  For example the Bosch BME280 and BMP280 are 2 different chips, but they share a lot of the same logic to read and interpret their data.
The same with SenseAir S8 and S11/Sunrise. They are completely different chips, but still are combined in the same plugin.  (S11 in UART mode, not I2C. For exact the reason explained above)

uzi18, 2019-12-06:
here it is imposible to switch to I2C