letscontrolit/ESPEasy #2557: Feature request: plugin sets web based generator

s0170071, 2019-08-19:
What if there was a web-hosted ESPEasy plugin sets generator ? 

Just check your plugins: 
[x] SHT Temperature
[x] IR-TX
[x] MQTT
[ ] MT681

or predefined sets,

and it spits out the plugin definition as header file.   Of course it also shows the memory usage as you order your dish.

It should also parse / grap all available plugins from the main repository, the plugin playground and maybe the local disk if thats possible?
Bonus: instead of downloading the header to the compile directory, is there a compiler directive to fetch it on the fly ?

uzi18, 2019-08-19:
one step more ... choose and send to build release.
like in lua for microcontrolles

jimmys01, 2019-08-19:
+1 from me

TD-er, 2019-08-19:
There is an option to run a Python file before the build.
See how I did the custom build in platform.ini.
So my plan was to write some Python to read a build definition from a database.

But I have not finished it yet.
Also @Grovkillen and I were planning to run such a build service as extra for the patreon users.
Thing is I have no clue how popular it would be, so how much compiler power is needed for such a service.

Misiu, 2019-08-21:
@TD-er if I understand correctly you want to create a service similar to what https://esphome.io/ is doing inside Home Assistant- You specify yaml with needed features and only those features are included inside the compiled binary.

Personally, I don't need an online service for compiling ESPEasy. I use Visual Studio Code and PlatformIO, so an online tool that would allow us to select board we have and plugins wee need and would generate an environment that we could add into `platformio.ini` would be enough. This way you won't need a powerful server to compile ESPEasy, but a simple site that will generate the config (even full `platformio.ini`) What do you think?

TD-er, 2019-08-21:
@Misiu Well I was thinking about doing both.
There is a number of users capable of building their own, but one of the many returning questions is how to build it.
And loading a yaml file into one directory is not too much effort, if you have the build environment setup, but that's where a lot of users are struggling.

But first to make such a build setup I have to do some work, regardless if it is for someone else to use it or a build server to do it.
But I think the current "custom" build is already making it very easy to define your own build.
Just extending it to parse some file if present would already make it quite simple to define your own setup.

happytm, 2019-08-21:
Another example of this feature is at link below:

https://quickiots.com/quick-iots-desktop/

Thanks