letscontrolit/ESPEasy #2558: Not telling IP address on serial line

hadmut, 2019-08-19:
Hi,
I just installed ESPEasy_R120_1024.bin on an ESP-01, which worked well, but after entering the password and SSID for the WLAN to connect to, there was no trivial way to find it, I had to use a tool like fing on the tablet computer in order to identify it. 

On the serial line it just says:
WIFI : Connecting... 1
WIFI : Connected!

Wouldn't  use of it, life, just everything be so much easier if it told SSID and IP address and not just "connected!" ?

TD-er, 2019-08-19:
The R120 build is about 2 years old.
Since then we have made quite some progress.
Not sure if it was already present in the r120 build, but you may want to try running `settings` as command from the serial port.
It should show the current settings.
I think `ip` is also a so called get/set command, so without parameters it should show the settings.

uzi18, 2019-08-19:
Settings and IP commands are correct.

hadmut, 2019-08-21:
> The R120 build is about 2 years old.

In that case it's somewhat confusing that this is called the _current stable release_, e.g. on 

https://www.letscontrolit.com/wiki/index.php?title=ESPEasy

hadmut, 2019-08-21:
But it solves the problem... thank you.

TD-er, 2019-08-21:
> > The R120 build is about 2 years old.
> 
> In that case it's somewhat confusing that this is called the _current stable release_, e.g. on

I agree., but sadly it hasn't really been stable the last year and it took quite some time to find out what was making it unstable. :(

closed 2019-08-21 by TD-er