letscontrolit/ESPEasy #2560: Emoncms Plugin, hard coded root folder

nixis, 2019-08-25:
Hi everyone,

I found a little problem in https://github.com/letscontrolit/ESPEasy/blob/d23b6272cfc5a9290718a7d05e4aafcc1ddcf6cf/src/_C007.ino 

There is a hard coded url with a subfolder "emoncms"
`74 String url = F("/emoncms/input/post.json?node=");`

In older versions and installation guides is the root url EMONCMSSERVER/emoncms but in newer versions and in its setup script, Emoncms does have its root in / and not in /emoncms/.
This causes the problem the Emoncms plugin cat reach the input of Emoncms server, and end with this error (on Emoncms Server):
`2019-08-25 12:36:12.237|WARN|index.php|406 Not Acceptable|emoncms/input/post`

Is there a chance to get the option to set the subfolder manually for Emoncms in the next releases? 

Build:⋄
20103  - Mega
System Libraries:⋄
ESP82xx Core 2_5_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.1.2 PUYA support
Git Build:⋄
mega-20190813
Plugins:⋄
48 [Normal]
Build Time:⋄
Aug 13 2019 02:16:54
Binary Filename:⋄
ESP_Easy_mega-20190813_normal_ESP8266_4M.bin

TD-er, 2019-08-26:
I've looked through the Git Blame of that URL and it looks like it is using the current syntax for the last 4 years.
See: https://github.com/letscontrolit/ESPEasy/blame/5f029c59d25e3ebb09bdaff05b20942dac6fe110/src/_C007.ino#L64

Did EmonCMS change anything here?

nixis, 2019-08-26:
Hi,

I used the official install script https://github.com/openenergymonitor/EmonScripts/tree/master/install
after execution Emoncms is reachable under [IP]/ 

If I follow this manual installation guide
 https://github.com/emoncms/emoncms/blob/master/docs/LinuxInstall.md) Emoncms will be available unter [IP]/emoncms.

The second url will be correct for the ESPeasy Emoncms Plugin, the first url is the "new" one without /emoncms/ subfolder.

Maybe it is possible to set the "subfolder" as an optional input field like:

**Existing Inputs**
Protocol: Emoncms
IP / Hostname: IP or Hastname
Port: 80
Controller Password: V3ryT0p53cr3t:)
**New Input**
Controller_Subfolder: /emoncms 

`tring postDataStr = F("GET $Controller_Subfolder/input/post.json?node=");`

TD-er, 2019-08-26:
Well when I'm going to change it, I'm not going to make only a sub part customizable, but I will make the entire path customizable.
Otherwise someone else may need another slight change and then we're stuck again :)

nixis, 2019-08-26:
Hi @TD-er thank you!

Think I can close this "issue".

TD-er, 2019-08-26:
> Hi @TD-er thank you!
> 
> Think I can close this "issue".

If you want things to remain like they are, then you can close it.
Otherwise it should be open

nixis, 2019-08-27:
@TD-er If you need a tester, let me know i will test it and give you feedback here.

TD-er, 2019-08-27:
Re-open this issue, or else it will be forgotten and not dealt with.

tonhuisman, 2024-06-22:
@nixis I know it's been a long time going, but I've made the url for Emoncms configurable in ESPEasy. Not sure if you still need that, but if you're available, can you please test this feature? Testbuilds available from [this GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/9628021903)

PR #4770 (2024-12-30): [MQTT] Controller improvements
Resolves #2341 
Resolves #4864 
Resolves #2560 
Resolves #4459 

Affected controllers:
- [C002] Domoticz MQTT
- [C005] Home Assistant (openHAB) MQTT
- [C006] PiDome MQTT
- [C014] Homie MQTT convention (V3/V4)
- [C007] Emoncms

Features:
- MQTT Controllers: generic support for `/cmd` and `/set` topics where applicable
- String/resource usage reductions (most of that in Homie controller)
- Added generic `CPlugin_helper_mqtt` module with MQTT support functions
- [C004] Small code improvements, add documentation and examples
- [C007] Add configurable url, stored in ` [... cut]
closed 2024-12-30 by TD-er