letscontrolit/ESPEasy #2561: Sonoff RF error in wifi setup

norimikan, 2019-08-25:
I install ESP_Easy_mega-20190823_normal_core_241_ESP8266_1M.bin to Sonoff RF.

But error in setup wifi after select AP.

Log is
>debug 3

Ok
37086 : AP Mode: Client connected: F8:1E:DF:D6:C8:0C Connected devices: 1
64385 : WD   : Uptime 1 ConnectFailures 0 FreeMem 6296 WiFiStatus 6
67177 : WIFI : Credentials Changed, retry connection. SSID: 299/103
67335 : WIFI : Set WiFi to AP+STA
67372 : WIFI : Connecting ******* attempt #0
67375 : WIFI : WiFi.status() = WL_DISCONNECTED  SSID: 299/103
67488 : AP Mode: Client disconnected: F8:1E:DF:D6:C8:0C Connected devices: 0
71683 : WIFI : Connected! AP: ******* (60:38:E0:8D:67:52) Ch: 5 Duration: 4109 ms
73023 : AP Mode: Client connected: F8:1E:DF:D6:C8:0C Connected devices: 1
73231 : WIFI : DHCP IP: 192.168.1.36 (ESP-Easy-0) GW: 192.168.1.1 SN: 255.255.255.0   duration: 1655 ms
73236 : SaveToFile: free stack: 3680
73854 : FILE : Saved config.dat
73855 : SaveToFile: free stack after: 3680
73858 : SaveToFile: free stack: 3680
74220 : FILE : Saved security.dat
74221 : SaveToFile: free stack after: 3680
74223 : firstLoopConnectionsEstablished
90292 : SaveToFile: free stack: 3376
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
csum err
ets_main.c 

I try to install 20190823 to 20190805 all release.
But same ploblem in wifi setting.

Best Regards.

TD-er, 2019-08-25:
That's a bug that will be saved as soon as [this one](https://github.com/letscontrolit/ESPEasy/pull/2559) is merged.

TD-er, 2019-08-25:
I made a [test build](https://www.dropbox.com/s/qb88s06cx9plyqw/ESPEasy_mega-20190823-5-PR_2559.zip?dl=0), could you test it?

TD-er, 2019-08-26:
Do you have any special characters in the SSID or password?

norimikan, 2019-08-26:
Sorry for late reply.

I'm Try test build ESP_Easy_mega-20190823-5-PR_2559_normal_core_241_ESP8266_1M.bin

But error in setting Wifi again.

attach log file.

Best Regards.
[sonoff rf error.txt](https://github.com/letscontrolit/ESPEasy/files/3541547/sonoff.rf.error.txt)

norimikan, 2019-08-27:
My SSID use "/" . like 000/000 .
Password is only alphabet and number.

I'm use esptool 2.7 to flash sonoff rf.

yesterday change to 2.6, not error same 2.7.

wifi setup page error is same, but reboot sonoff rf is connect AP.

Best Regards.

TD-er, 2019-08-27:
> wifi setup page error is same, but reboot sonoff rf is connect AP.

So at least I fixed the bug that makes the node crash during setup and it now connects to the AP?

I noticed you had exceptions while the unit tries to start the AP mode.
Could you add come capacitor to the 3V3 line?
I think this may be related to a power issue, since an exception is not likely to happen with the more recent builds in that phase of the boot sequence.
Sonoff is known for under powering their more recent units.  (Sonoff Basic R2 is a notorious example of having a barely sufficient power supply)

norimikan, 2019-08-27:
Yes, I can connect AP.

After set AP and error, disconnect USB(Poweroff) and re-connect(Poweron)
connect AP.

Today I try install ESPEasy_MEGA_0827 and 0823.

0827 is not error but web connection is very slow.
0823 is error but web connection is fast.

Sorry for poor english and poor technical report.

Best Regrads.
[sonoff rf mega0827.txt](https://github.com/letscontrolit/ESPEasy/files/3546151/sonoff.rf.mega0827.txt)
[sonoff rf mega0823.txt](https://github.com/letscontrolit/ESPEasy/files/3546152/sonoff.rf.mega0823.txt)

PR #2559 (2019-08-26): [WiFi] Improve WiFi reconnect stability and fix initial setup
Simplified the WiFi connect state machine.
Fixed setup of initial WiFi configuration.

AP mode is now kept as long as client is connected.
AP mode will be started immediately if there are no valid WiFi credentials, or after 20 seconds of no wifi connection.
AP mode will be disabled 60 seconds after starting it, or after the last client disconnected from the AP.

Fixes #2561 
Fixes #2338 
Fixes #2203 
Fixes #1987 
Fixes #1737
Fixes #1640 
Fixes #1565
Fixes #1479 
Fixes #1098
Fixes #558
closed 2023-09-03 by TD-er