letscontrolit/ESPEasy #2564: Try to always use BSSID/channel when connecting to AP, it is a lot faster.

TD-er, 2019-08-26:
Currently when performing a WiFi reconnect, the first attempt is to use the last known channel + BSSID.
This takes roughly 1 second to reconnect.
Such a speed-up can also be very beneficial for battery powered nodes, but then the BSSID and last used channel must be kept.

This can be stored in RTC, to speed-up a reboot from deep sleep or crash, but this may have impact on some nodes using under powered power supplies.
The power consumption may be higher, right after boot, which may crash the unit if the capacitors did not have enough time to get charged.

To use this from cold boot, the data should be stored somewhere. Preferably not on flash due to the number of writes it may cause.

ShardanX, 2019-08-26:
@TD-er 
"The power consumption may be higher, right after boot, which may crash the unit if the capacitors did not have enough time to get charged."

Easy workaround for that:
~10 KOhm from Vcc to CH_PD (that should be there always) and a capacitor from CH_PD to ground.
Usually i use 10,,,22 µF but it can be made bigger.
This keeps the chip down for a short time after powering up.
Time is depending on resistor and capacitor of course.

More the professional way:
For selfmade PCBs a voltage regulator with a "power good" output can be used.
"power good" is set when the regulator is working stable after power on.

Both ways should work.

TD-er, 2019-08-26:
@ShardanX Yep, that's how you make your prints and how I should make mine :)
But the reality is that a significant percentage of the modules used to run ESPeasy are selected from Ali Express and Banggood, sorted by price and those have the CH_PD pin hard wired, or next to impossible to reach (Sonoff/Blitzwolf/etc.)
And if you look at the modules labelled with "Wemos" (I don't call them "Wemos modules" since it is very unlikely there are original Wemos modules available today), those often have only 150 mA voltage regulators. So they are very likely to crash if you have only a few simple sensors powered by the module 3v3.

The amount of boards on the market with crappy components is incredible. So we'd better support it or get lots of reported issues :)
Maybe I can also make the delay to start WiFi more dynamic, so you don't even have to tweak a setting. It will just increase the delay until a boot is successful.

tonhuisman, 2022-04-18:
This behavior has been extensively worked on and massively improved, so can probably be closed now.

TD-er, 2022-04-18:
Yep, can now be closed. The recent fixes for this have been merged, but not yet included in a nightly build.

closed 2022-04-18 by TD-er