letscontrolit/ESPEasy #2569: DHT22 not working with ESP Easy on Wemos D1

TheAppService, 2019-08-26:
Hello alltogehter, 

I have connected my DHT22 properly to my Wemos D1 but can´t get it to work with ESP Easy. Many "normal" program scripts to output the temperature and humidity to the serial monitor also didn´t work, but this here finally did the trick! : https://chewett.co.uk/blog/1476/using-t ... i-esp8266/

There is also a modified library involved, related to the 3,3V limitation of the Wemos D1´s digital GPIOS. The EspEasy displaying me in the webinterface "nan" may be also caused by the "normal" library used by EspEasy, calculating the values with 5,5V. I´d be very grateful if someone knew how to resolve this problem. 

 Best regards  Daniel  

EDIT: I figured out, that "nan" is only displayed when I set the setting "IDX/Variable". Otherwise simply 0.0 is shown.

uzi18, 2019-08-26:
for this sensor none of libraries calculate with 5V or 3V3 because it is a digital sensor, so you receive value from sensor itself

TD-er, 2019-08-26:
OK, so apparently it is suggested to use [this DHTesp library](https://github.com/beegee-tokyo/DHTesp)

I have quickly browsed through the code of that library, and some of the code does look familiar, so maybe that library does have some fixes compared to the version we use.

But I don't remember a lot of issues reported with this plugin, so I would also like to know what exact sensor your are running and more info about your board and how it is connected.

uzi18, 2019-08-26:
will try to compare it with our implementation

uzi18, 2019-08-26:
@TheAppService lets start with debug log from reading your sensor under espeasy please

@TD-er I see initial timings are a little bit different, but lest check debug log so. we will know maybe what is realy wrong
We support bunch of DHT like sensors, so mayby just DHT22 is wrong

TD-er, 2019-08-26:
I know there was an issue about it about half a year ago (or maybe longer?) that was about changing the timing of some of the DHT sensors, but not all.

uzi18, 2019-08-26:
you have added some error loging, so we can use it now =)

TheAppService, 2019-08-27:
Alright, well I will give you more information about my setup shortly! But first, maybe a suggestion of mine - If the DHT 22 is the only affected one on Wemos Boards, maybe replacing the plugin wouldn´t be a good idea, maybe for testing purposes adding the new plugin especially for Wemos boards or maybe "Updated ESP Compatibility Plugin" would be suitable...

Sensor Type:
"ASAIR AM 2302" - "SN: 180497FD9"
Board is a Wemos D1 Ver. 2015-08
My Circuit is exactly like this here, although I have a Wemos D1 instead of the D1 Mini
[image]
DEBUG Log will follow in a nother comment when I have found the feature in ESP Easy

TheAppService, 2019-08-27:
In the DEBUG log i can only find this: (Maybe you can give me a hint how I can exactly debug the sensor...
`66149 : WD : Uptime 1 ConnectFailures 0 FreeMem 28888
96150 : WD : Uptime 1 ConnectFailures 0 FreeMem 28760
126151 : WD : Uptime 2 ConnectFailures 0 FreeMem 28728
156152 : WD : Uptime 2 ConnectFailures 0 FreeMem 28696
186153 : WD : Uptime 3 ConnectFailures 0 FreeMem 28680
216154 : WD : Uptime 3 ConnectFailures 0 FreeMem 28664
246155 : WD : Uptime 4 ConnectFailures 0 FreeMem 28648
276156 : WD : Uptime 4 ConnectFailures 0 FreeMem 28616
306157 : WD : Uptime 5 ConnectFailures 0 FreeMem 28584
336158 : WD : Uptime 5 ConnectFailures 0 FreeMem 28568`

uzi18, 2019-08-27:
If you are using serial connection, than in advenced settings change log level for serial to debug

TD-er, 2019-08-27:
Hmm what version of ESPeasy are you running?
This looks rather old.

TheAppService, 2019-08-27:
Booting Build nr:120 (I thought this was the newest...)
I set the serial log level to 5 (Hope this is high enough) but I´m only getting this output: https://gist.github.com/TheAppService/0d47f76b3462b988c612007afeb3d9e3

TheAppService, 2019-08-27:
"please update for latest version, R120 is obsolete." -- So i should youse the mega Versions ? Maybe you should clarify, that the "current stable" on this site here https://www.letscontrolit.com/wiki/index.php/ESPEasy is rather old...

uzi18, 2019-08-27:
yes, try mega build, mega have some problems and is not rocksolid, but has lots of bugs resolved

TheAppService, 2019-08-27:
Alright, I updated it but the log still "only" says 40797 : DHT  : No Reading
60744 : DHT  : No Reading

TheAppService, 2019-08-27:
[image]
Just tested it again with the simple sketch under the exact same conditions and voila ;D

TheAppService, 2019-08-27:
My settings should be correct or do you think otherwise?
[image]
[image]
[image]
[image]

AxelMilb, 2019-08-27:
Use 4,7kOhm resistor

TheAppService, 2019-08-27:
I can try to "build" one out of the resistors I have but I won´t exactly be able to hit 4,7kOhm. Is this really neccessary ? I have one perfectly working sensor on my raspberry with a 10kOhm resistor...

AxelMilb, 2019-08-27:
I have a DHT22 on a Wemos D1 mini with 4.7kOhm resistor.
The specifications are 4.7k - 10K 
[https://www.letscontrolit.com/wiki/index.php?title=DHT11_DHT22](url)
DHT22 defect? ESP alright? Cold solder joint?

TheAppService, 2019-08-27:
> 
> 
> [image]
> Just tested it again with the simple sketch under the exact same conditions and voila ;D

Well @AxelMilb as i tested it with the exact same setup using the sketch you see above, it is generally working...

TheAppService, 2019-08-27:
>  The specifications are 4.7k - 10K

Ah, one more question, do you have it connected to 5V or 3,3V?
And I don´t know, maybe something has changed concerning this sensor / there are maybe also different types on the market... Therefore maybe the devs could/ and maybe already had a look over the library to maybe introduce a little compatiblity mode with the other library

uzi18, 2019-08-27:
You can use 2x10kOhm in parallel connection so it will be about 5kOhm

uzi18, 2019-08-27:
Assume DHT22 works for you now =)

uzi18, 2019-08-27:
@TheAppService "no reading" means no data from sensor, it could be related to initial timing or pullup resistor 

This sensor works with 3V3 and 5V.

@TD-er  some improvements here #2571

TD-er, 2019-08-27:
A [Test build](https://www.dropbox.com/s/da0vm4v41eygzuh/ESPEasy_mega-20190827-2-PR_2571.zip?dl=0) for PR #2571.

uzi18, 2019-08-27:
@TheAppService please try this test build and give us debug log, thanks

TheAppService, 2019-08-28:
Good morning, thank you for your big efforts and the test build! Unfortunateley with the flashed build :ESP_Easy_mega-20190827-2-PR_2571_normal_ESP8266_4M.bin The issue still persists- 1335678 : DHT  : No Reading

uzi18, 2019-08-28:
try change gpio boot state to default

TheAppService, 2019-08-28:
> 
> 
> try change gpio boot state to default
Did it, but unfortunateley the issue is still the same /:

Roos-AID, 2019-08-28:
Just to add, I have intermittent problems with the DHT22 on ESP01_1M.  Cannot reproduce but sometimes the DHT22 fails with same symptoms as described above.  After switching off power for couple of minutes, the function resumes after boot.  When I only do a reboot it contineous to have No reading.  It is not with all DHT22 on ESP01_1M combinations.   I do suspect a critical timing problem, so when Pull #2571 is closed I will deploy and will do some field tests.

TD-er, 2019-08-28:
@Roos-AID There is a [test build](https://github.com/letscontrolit/ESPEasy/issues/2569#issuecomment-525409583) you can try on one of the nodes.
Just to see if you're facing the same issue as well.

uzi18, 2019-08-28:
@TD-er if AM2302 is same as DHT22 so I can try to check also on my own :)

toast1234, 2019-08-28:
Hi, since happened to see this tread I can just mention that I have a Wemos D1 mini for 2 years and frequently update the build (normal 4M 8266). I have  AM2302, DHT22, Dht11 and  ds18b20 connected to it. Have not noticed any problems with the reading. I send ro Domoticz every second minute. I use 10k pullup. I connect Vcc to 5V and pullup to 3.3V. I use GPIO 0, 12,13,14.

uzi18, 2019-08-28:
@toast1234 please try test build if it work for you with your sensors

TD-er, 2019-08-28:
Even if there is some timing fluctuation among sensors (or boards, those crystals can be of varying quality) it still should recover from a sensor lockup.
Apparently the sensor should be power cycled to make it work again.

Is the sensor used in somewhat extreme environment? (e.g high humidity)

Roos-AID, 2019-08-28:
@TD-er I have loaded the test build on a node that regularly fails, it works after cold boot, so that is good. Let's wait and see if it is still getting values tomorrow as it is an intermittent issue. 
On this node, GPIO 2 is used for DHT22 and on Serial , GPIO 1 and 3,  I have MHZ19 CO2. 
Thanks for all the good work !

toast1234, 2019-08-28:
OTA upgrade ESP_Easy_mega-20190827-2-PR_2571_normal_ESP8266_4M.bin and still show the readings after 15 min up time.
Dht11 had one miss showing NaN. Dht11 are crap anyway...  I must add that I have never had reason to watch close at every reading. But it has never been any flat line or missing values in Domoticz over the past year. Its now bed time in Sweden :-)

toast1234, 2019-08-28:
My ds18b20 is in the freezer and the DHT22 is under the house with up to over 90% humidity in the summer depending on the weather.

uzi18, 2019-08-28:
@Roos-AID please provide debug log so we can have more data

TheAppService, 2019-08-28:
Just a little comment from me, I assume the timing and hereby the whole issue with the library varies from board to board... If there´d be a general flaw I think you´d already have had more complaints..
> 
> 
> @TD-er if AM2302 is same as DHT22 so I can try to check also on my own :)

Therefore, do you @uzi18 maybe also have a Wemos D1 for testing ? 
And just out of interest, what about my suggestion of adding a compatibliity version of the other library I mentioned or is this much work / Did you try to migrate something from ESPDHT library in the pull request ?

uzi18, 2019-08-28:
@TheAppService please try on other pin like D5 (GPIO14)

uzi18, 2019-08-28:
@TheAppService we do not use any lib for DHT sensors, need to check if have such wemos d1 modules.
please consider to paste more lines of log, as @Roos-AID said it sometimes work sometimes no

TheAppService, 2019-08-28:
> 
> 
> @TheAppService we do not use any lib for DHT sensors, need to check if have such wemos d1 modules.
> please consider to paste more lines of log, as @Roos-AID said it sometimes work sometimes no

Well I would for sure poste more lines but unfortunateley there is no more to post except the No reading thing.. (Or is my DEBUG level wrong ?) As I just said, I assume the issue varies from board to board and maybe the timing works for @Roos-AID´s ESP01_1M sometimes and on my Wemos D1 there is simply no function..

uzi18, 2019-08-28:
@TD-er my AM2302 (DHT22) works before and after this patch

uzi18, 2019-08-28:
@TheAppService I see problem, GPIO2 is pin D4 and GPIO4 is D2, please correct your settings.

TD-er, 2019-08-28:
@uzi18 , I don't see where that's incorrect in the sceenshots.
It was also D2 on the ArduinoIDE scetch and on the ESPeasy settings page.

But maybe it is already way past my bed time also :)

uzi18, 2019-08-28:
@TD-er here it was different https://github.com/letscontrolit/ESPEasy/issues/2569#issuecomment-525151617
some posts later it looks ok.

TheAppService, 2019-08-29:
> 
> 
> @TheAppService please try on other pin like D5 (GPIO14)
> 
Haha, i lose my mind, it works without problems on PIN D5...

toast1234, 2019-08-29:
Its more than a year ago so I dont remember why but I had a reason for using the high GPIO numbers. I probably looked at the description for the device to find the best suitable pins. I'm quite sure I at that time had issues with the lower numbers. I picked GPIO 0 for one device just because the last high one is broken.

Roos-AID, 2019-08-29:
@uzi18 My node is now running 11 hours,  it is still ok.  In the syslog I found some occurrences of EspEasy: DHT : Checksum Error,  but it recovers from it as the next read is ok. 
 
So , do not see how to reproduce now, it all seems ok for me.

uzi18, 2019-08-29:
From some time we have "warning" sign added to some problematic gpios, by @TD-er 

So we have got happy end :)

@Roos-AID how long is your DHT line? consider to use lower resistor eg. 5k or 3k3 Ohms instead

TD-er, 2019-08-29:
You may also want to take a look at this page in the docs: [best-pins-to-use-on-esp8266](https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#best-pins-to-use-on-esp8266)

uzi18, 2019-08-30:
@toast1234 please try latest official build with DHT11, we have one report about something is wrong

georgep56, 2019-08-30:
I updated my test unit from 20190827 to 20190830 and my DHT11 sensor stopped working (it showed NaN for both temperature and humidity). On the same unit a BMP180 works fine on both builds.
Hardware is a quite old Wemos Lolin which has never shown any [hardware] problems.
I downgraded to 20190827 and everything is fine again.

In both cases I used the 'normal' core 2.6.0 builds as I am looking into the WiFi related reboots.

TD-er, 2019-08-30:
Since the changes were made based on a report which later appeared to be incorrect configuration, maybe we should just revert the changes to this plugin.
Maybe the change of the return type of the function, which was also found by @uzi18  is enough to fix getting NaN values as supposed valid readings.

uzi18, 2019-08-30:
@TD-er prepared some commits for that, please wait a moment for report from @toast1234
maybe you can prepare test build, so @georgep56 could test it

georgep56, 2019-08-30:
Happy to test :-)

toast1234, 2019-08-30:
My node is remote and the OTA failed. Will update to latest build this evening an can report back

georgep56, 2019-08-30:
I am busy for some of this afternoon but if someone could give a link to
the test build when it is ready I will download and test as soon as I can.

uzi18, 2019-08-30:
If this build not resolve your problem, will use my sealae with dht11

[45 more replies]

PR #2571 (2019-08-29): [DHT] start conversion improvements
- use delays according to datasheets
- set bus high after a request for every type of sensor (this may fix problems with pullup resistors)

fixes #2569
commit 18d26a38 (2019-08-29): [DHT] start conversion improvements

- use delays according to datasheets
- set bus high after a request for every type of sensor (this may fix problems with pullup resistors)

fixes #2569
closed 2020-09-09 by TD-er