letscontrolit/ESPEasy #2570: System Info (plugin 026) does not display "indicators" anymore

giig1967g, 2019-08-27:
Hi all,
I just upgraded to latest version from April version and noted that the system info plugin is not displaying any value.
See the plugin screenshot of latest version (ESP_Easy_mega-20190827_normal_ESP8266_4M.bin):
[image]

And this is the screenshot of april version (ESP_Easy_mega-20190425_normal_ESP8266_4M.bin):
[image]

TD-er, 2019-08-27:
Have you saved the settings of this plugin again?
There is something not right in the dynamic loading of this plugin.

giig1967g, 2019-08-27:
Great! 
Saving the settings works.

giig1967g, 2019-08-27:
I think that there has been a change in the plugin that broke it:
Now in the device setting page there is a new option called "Output data type" there was not there before.

It has to be reset for the plugin to work again.

closed 2019-08-27 by giig1967g