letscontrolit/ESPEasy #2577: ESP.Easy.Flasher PB font with window resulution

LaurentG91, 2019-08-28:
Dear developpers,
there is a problem when I use the new ESP.Easy.Flasher with a windows size factor e.g. 150% on high resolution screen. The size of the font is not in accordance with the size of the box containing informations. 
[image]

Generally speaking, this new interface is not easy to use, some explanations are needed

Best regards
Laurent

Domosapiens, 2019-08-28:
Please read the blue part: install the fonts ...

TD-er, 2019-08-29:
If the problem is resolved by installing the included fonts, please close the issue.
If it has not be resolved, please show a screenshot using the included fonts.

LaurentG91, 2019-08-30:
Yes it's partially solved: with the font installed, all the HMI is coherent and data are visible, but I think it's does take into account the scale factor (e.g. 150%) and then the window is very small compare to the others in a 4K screen for example (and become unreadable specially with these narrow fonts). I understand that this HMI version is a draft version, and I think it's usable for the majority of experimented users to test new version, but it need to be improved and (self) documented if pushed "as a major realize", specially for beginners like me. 
Thanks a lot for all this very useful developments
Best regards
Laurent

closed 2019-08-30 by LaurentG91

Grovkillen, 2019-08-30:
Save the settings and open the default.ini file and change the value of DPI (last line in the file). That should change the scale.

LaurentG91, 2019-08-30:
Yes. This solve the size of the window.
Thanks a lot...
Laurent