letscontrolit/ESPEasy #2579: All my units rebooted at the same time

giig1967g, 2019-08-28:
Hi,
today I had to do some maintenance work on my PI3 that hosts OpenHab2 and the MQTT broker. So I had to shut the OpenHAB machine down.

As soon as I switched off the PI3 all my units rebooted with the following message in the sysinfo page:
```
Boot: | Manual reboot (3)
Reset Reason: | Software/System restart
Last Task: | Const Interval timer, id: 3
```

All of my units have one controller enabled: the OpenHAB machine that I had to put offline.
All my units publish the status of their sensors to the OpenHab controller regularly.

I tried to reproduce the issue, and I managed to reprodece it: just switching off the OpenHab machine the units reboot themselves.

Can this help somehow to trace the intermittent but recurring rebooting problem?

giig1967g, 2019-08-28:
I checked the settings and I have three different setups (difference is client_timeout and queue_depth):
1:
[image]
2:
[image]

3:
[image]

To be honest I don't know why they are different. Probably some test when the feature was released.
What should be the safe setting?

Message length:
1: when button is pressed (see rule below) + sysinfo every 2.4 hours (8640 seconds)
```
on p1#switch=0 do
 publish /%sysname%/p1/Switch,0
endon
```

2: every hour (see rule below) + sysinfo every 2.4 hours (8640 seconds)
```
on rules#timer=1 do
 publish /%sysname%/Relay/r1,=[PLUGIN#MCPGPIO#PINSTATE#1]
 publish /%sysname%/Relay/r2,=[PLUGIN#MCPGPIO#PINSTATE#2]
 publish /%sysname%/Relay/r3,=[PLUGIN#MCPGPIO#PINSTATE#3]
 publish /%sysname%/Relay/r4,=[PLUGIN#MCPGPIO#PINSTATE#4]
 publish /%sysname%/Relay/r5,=[PLUGIN#MCPGPIO#PINSTATE#5]
 publish /%sysname%/Relay/r6,=[PLUGIN#MCPGPIO#PINSTATE#6]
 publish /%sysname%/Relay/r7,=[PLUGIN#MCPGPIO#PINSTATE#7]
 publish /%sysname%/Relay/r8,=[PLUGIN#MCPGPIO#PINSTATE#8]
 timerSet,1,3600
endon
```

3: when button is pressed (see rule below) + sysinfo every 2.4 hours (8640 seconds)
```
on d5#switch=0 do 
 publish /%sysname%/d5/Switch,0
endon
```

TD-er, 2019-08-28:
I would set the client timeout to 100 msec.

The other settings look OK.
A queue depth of > 10 does not sound useful, but you may have a use case for that?
Any way, all nodes crashed, so that will probably not be the reason why they all crashed

TD-er, 2019-08-29:
Could it be the controller is using too much RAM to keep the buffered messages in the queue?
Do you have any idea how long one such message is and what buffer settings do you use for this controller?

giig1967g, 2019-08-29:
tonight 

>  Any way, all nodes crashed, so that will probably not be the reason why they all crashed

Infact. They crashed all the same time with the same error in the sysinfo page (timer id. 3).
What is the exact meaning of this error message?

TD-er, 2019-08-29:
It is a bit cryptic, but this was easy to encode en store in RTC what the last scheduled task was.
If you look into the code for "Const Interval", you'll see it is a `CONST_INTERVAL_TIMER`
When looking into `process_interval_timer`, you'll see a number of case statements in the switch.
`TIMER_1SEC` is nr. 3 in that list.

giig1967g, 2019-08-29:
I had already checked the error number in the GlobalVariables.h file.
But could not understand exactly the meaning.
It's interesting that all the units rebooted at the same time with the same error.

Have you aver tried to shut down your OpenHAB instance with MQTT and see if you get the same behaviour? It could be a starting point for a debug.

TD-er, 2019-08-29:
Nope, since I am not running OpenHAB :)
But I guess it may be related to the bug that was recently fixed in the latest core 2.6.0 (not sure if merged yet) which should fix an issue where the timeout is not obeyed while still the start of the connect attempt.

closed 2023-09-03 by TD-er