letscontrolit/ESPEasy #2580: mega-20190827: Oled not working after update

pwassink, 2019-08-29:
Esp12* nodemcu 
mega-20190827

This morning i did an update of a testnode with the mega-20190827, uring reboot no logo on the display, and no data afterwards, esp works, sends sensordata to domoticz, but no display on the (1306 based) i2c oled

The i looked into the devices tab of the esp, because sometimes things go inactive when faulty, 
after a submit of the sam configuration it has been working for months it displayed an i assume Red bar with Error in the top of the page :  Error: text to long for line 11

as i mentioned before i did not change anything, and even more weird, i did make a screenprint, tried to get the exact message again while entering this possible issue, no error displayed ?

display stays empty, seems there is something wrong there

pwassink, 2019-08-29:
Not sure anymore if it is related to this release, could be coincidental failing of zome i2c related hardware on that bus during the upgrade, seems i2c related, will investigate further later this day.

TD-er, 2019-08-29:
If you get a red warning when saving things, it usually is a sign something went wrong while saving the settings.
I did add a check to the entered values for this plugin, since there was a bug (or rather a missing check) that would not check for the length of an entered string.
It was possible to enter a string longer than the allowed (32?) length, if you were using Unicode characters.
What may happen then is that the plugin cannot distinguish the difference between 2 lines (0-termination is missing) and it may lead to higher memory usage or other strange issues.
It should be 'fixed' now with the newly saved settings, but it may miss one or more characters or have the wrong character at the position where the Unicode character was.

TD-er, 2019-08-29:
If it is working again after saving (and maybe reboot), will you then close the issue?

pwassink, 2019-08-29:
Ok, that clarifies some of the things i've seen, i guess . 

Sorry was not able to take a closer look today, will investigate further tomorrow with an open mind ..

pwassink, 2019-08-30:
Can confirm local error, 

i2c bus was disturbed by a broken sensor .. 

Closed issue

closed 2019-08-30 by pwassink