letscontrolit/ESPEasy #2585: DHT22 stopped working after 20190830 update

Quart666, 2019-08-30:
I updated two of my ES8266 Wemos D1 today.
Both of them have a DHT22.
And after upgrade the values says NaN on both Wemos D1

TD-er, 2019-08-30:
See: https://github.com/letscontrolit/ESPEasy/issues/2569

[Test build ESPEasy_mega-20190830-4-PR_2582](https://www.dropbox.com/s/g4q9b55seia16d7/ESPEasy_mega-20190830-4-PR_2582.zip?dl=0)

mackowiakp, 2019-09-04:
The same problem with DTH22 on nodeMCU 8266 4M test build. NaN instead readouts. 20190827 works correctly.

TD-er, 2019-09-04:
And with the latest build?

mackowiakp, 2019-09-04:
I try 20190903, Nan instead readouts.

uzi18, 2019-09-04:
debug log please

mackowiakp, 2019-09-04:
Strange. I try to load from GitHub 20190903 again to "catch" the logs but 20190903_252_test_4M disappear from zip file.

uzi18, 2019-09-04:
try any other 4M bin

mackowiakp, 2019-09-04:
But I wanted to collect logs from the device and software on which I found a malfunction. Only then will I be sure of the logs collected. And one of the connected sensors is the PMS dust sensor, supported only by the "test" version.

TD-er, 2019-09-04:
It is renamed to ...`_4M1M`
Since there is now also a test build for `_4M2M` with a different flash layout.  (currently only for IR-build)

Also a file will not be included if it is too large to be used.

mackowiakp, 2019-09-04:
OK, if _4M1M is equivalent for former "test" version for 4M flash, so still there is no in zip file named ESP_Easy_mega-20190903_normal_core_252_ESP8266_4M1M.bin

uzi18, 2019-09-04:
so now question is what image did you flashed

mackowiakp, 2019-09-04:
ESP_Easy_mega-20190830_test_core_252_ESP8266_4M.bin. But can not test any equivalent for release 20190903 because I need PMS dust sensor driver.

uzi18, 2019-09-04:
try normal build without pms, and if you need i can later prepare bin for you with pms

mackowiakp, 2019-09-04:
Be so kind to prepare such build. In my meaning there is a sens to "catch" logs in the same configuration on which I found a malfunction

uzi18, 2019-09-04:
[firmware.bin.zip](https://github.com/letscontrolit/ESPEasy/files/3575428/firmware.bin.zip)

mackowiakp, 2019-09-04:
Seams to work properly. But give me 24 h for final assumption.

skylineboy20000, 2019-09-04:
> [firmware.bin.zip](https://github.com/letscontrolit/ESPEasy/files/3575428/firmware.bin.zip)
My DHT22 also stopped responding - NaN, tried all the 4m files in the 20190903 to no avail but this custom build seems to be working so far. Will monitor and report back also.

uzi18, 2019-09-04:
@TD-er build env dependent issue?

mackowiakp, 2019-09-05:
It looks so. Maybe this is not exactly the same situation, but I installed 20190903 build on other nodeMCU with DTH11 sensor. NaN occurs. But custom FW works properly both with DTH22 and DTH11. But I think that reboots accrues more frequently comparing to 20190827. But it is too short period of time to say it for sure. Maybe it is WiFi calibration related issue, because both units use ADC for another sensor.

uzi18, 2019-09-05:
All reports about problems with DHT, was resolved so maybe your 
problem is related to something different. 
My build have just some dev plugins disabled by default (in facts it is dev_ESP8266_4M1M)

Please add printscreens of your controller and devices configuration.

TD-er, 2019-09-05:
And note the used core version.

mackowiakp, 2019-09-05:
First I have to say that configuration of sensor is a little bit complicated, so it may cause some unwanted interactions . I have connected to this unit such set of sensors:
- BME180
- DTH22
- PMS7003
- photo-resistor connected to ADC 
- generic sensor - uptime
DTH22, generic sensor and ADC sends readouts do Domoticz just from task directly. In case of PMS and BME180, readouts are sending via SendToHttp from rule.
Its because BME readouts are sometimes nonsensical like pressure 2500 hPa or temp 200 C. or both. It is already rapport as a bug in issue list. So before sending readout to Domoticz, first values are checked for correctness and then send to Domoticz. 
PMS laser diode has life time 8000 h of continuous work (less than one year). So to increase lifespan, I turn on sensor from rule every 10 min for 30 sec time period, reset, run task for readouts and SendToHTTP results from role too of course. Today PMS driver has no such facility itself. 
Core version used in custom FW is ESP82xx Core 2.6.0-dev, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Below screenshots.

mackowiakp, 2019-09-05:
Just updated above post.

mackowiakp, 2019-09-06:
Can confirm that using tis custom build, reboots accures much more frequently.

TD-er, 2019-09-06:
> Can confirm that using this custom build, reboots accures much more frequently. Revert to 20190827

Just to be clear. What custom build exactly?
[This one @uzi18 made](https://github.com/letscontrolit/ESPEasy/issues/2585#issuecomment-527968831) ?

@uzi18 How did you make it, in Linux or Windows?

uzi18, 2019-09-06:
@TD-er as always:
```
[linaja@pldmachine ESPEasy-mega_test]$ ~/bin/platformio --version
PlatformIO, version 4.0.3
[linaja@pldmachine ESPEasy-mega_test]$ git describe 
mega-20190903-1-g359fecff
[linaja@pldmachine ESPEasy-mega_test]$ uname -a
Linux pldmachine 5.2.10-1 #1 SMP Mon Aug 26 20:47:00 CEST 2019 x86_64 Intel(R)_Core(TM)_i5_CPU_______M_520__@_2.40GHz PLD Linux

```

mackowiakp, 2019-09-06:
> Just to be clear. What custom build exactly?
This one @uzi18 made ?
Yep.

skylineboy20000, 2019-09-06:
> > Just to be clear. What custom build exactly?
> > This one @uzi18 made ?
> 
> Yep.

Just to add, @uzi18's build for me has been online c.35hrs now with no issues, stable and no NaN errors.

mackowiakp, 2019-09-06:
In my case there was not NaN readouts too. But more reboots.

uzi18, 2019-09-06:
reboots could be related to different core used

mackowiakp, 2019-09-06:
But formerly, before installing custom FW, it works a lot better. But for shure formerly I use core 2.4.2 but custom FW based on 2.6.0. When I revert to 20190827 (2.5.2), reboots still occurs that is like in custom FW. Strange.

uzi18, 2019-09-06:
will try to prepare 2.4.2 build for test, time is not on my side today as we have a long trip to family

Roos-AID, 2019-09-06:
With ESP_Easy_mega-20190903_normal_core_242_ESP8266_1M.bin  my DHT22 gets NaN after roughly 1 day.  It appears that an unscheduled reboot occurred at 13:00 CET. Log attached
[All_2019-9-6-23_58_39.csv.zip](https://github.com/letscontrolit/ESPEasy/files/3585843/All_2019-9-6-23_58_39.csv.zip)

A warm reboot does not solve the problem, only after a cold boot the DHT22 responds again.

Is the fix from PR_2571 in this build ? As that fix seemed to have fixed this.

TD-er, 2019-09-06:
PR_2571 was before the change that may have caused these issues to (re)appear.

I think you're looking at several issues you just described.
- Watchdog reboots (lots of reports of them the last 18 months)
- NaN readings of the DHT22
- DHT22 not being able to recover from a fault state.

The NaN readings are cause of too critical timings needed for this plugin.
DHT22 not recovering is something I have no idea about its cause.
Watchdog reboots are caused by numerous issues and we're tackling them one at a time as soon as it is clear what is causing them. (not related to this topic)

uzi18, 2019-09-06:
@TD-er got only one idea - timings are not accurate, but need to check it with logic analyzer

maybe add +1 of first delay can resolve it

TD-er, 2019-09-06:
> maybe add +1 of first delay can resolve it, in datasheet 18ms is minimum so maybe delay(19) will. do the job

That sounds rather time critical :)

uzi18, 2019-09-06:
maybe it is and we don't know how esp clock is accurate

TD-er, 2019-09-06:
I know there is quite a range of "accuracy' among nodes.
The quality of the crystals may differ a lot.
That's also one of the reasons I added the "drift" log when the NTP updates.
I've seen updates on some nodes which were close to 1% off (> 10 msec/sec correction) and that should not happen with even really low quality crystals.

uzi18, 2019-09-06:
see my comment in code with minimum from datasheet

TD-er, 2019-09-06:
Link?

uzi18, 2019-09-06:
so guess it is start condition fail here

uzi18, 2019-09-06:
@TD-er https://github.com/letscontrolit/ESPEasy/blob/b5fd0e0ed12889cd1005c8dc5dd250cf84ba6f18/src/_P005_DHT.ino#L153-L159

Roos-AID, 2019-09-14:
Any news on 
> NaN readings of the DHT22
> DHT22 not being able to recover from a fault state.

I can easily reproduce it, when I do a Tools/Reboot then NaN readings from DHT22.  As long as there is no warm boot (either invoked manually or by watchdog) the readings are fine.

haldi4803, 2019-10-21:
I have the same issue here... Used a chinese Board which goes to GPIO-2 and was wonderwing why i only get NaN on 20191003_normal_ESP8266_1M.
So i've switched to the 20190827_normal_ESP8266_1M and without further configuration it just worked.

mackowiakp, 2019-10-21:
I use 20190928 1M4M. Works OK both with DTH11 and DTH22. Another issue I had using 20191003 was not working SentToHTTP so revert to 20190928.

dpomt, 2019-11-24:
Wemos D1 Mini with DHT22 - always getting NaN and DHT:NoReading.
Tried different firmwares (20191123, 20191108, 20190928 etc.), different resistors (3k3, 4k7, 10k) as well as +5V vs. +3.3V Always the same: NaN and DHT NoReading in log.

Any ideas?

TD-er, 2019-11-24:
@dpomt Have you also tried the version mentioned in the title of this issue, where it was supposed to be working? Just to make sure your hardware setup is not to blame here.

dpomt, 2019-11-24:
You mean the version *before* 20190830  ?

dpomt, 2019-11-24:
@TD-er: I have re-wired and retested - on a Wemos D1 Mini  and on a Wemos D1 Mini Pro.
Both work perfect with 20190827!! But does not work with actual release.
Any chance to get this fixed in an actual release?

Thanks so much for this great stuff.

Grovkillen, 2019-11-24:
Just my quick response: have you tried saving the settings to make sure it's not a settings error?

TD-er, 2019-11-24:
> Any chance to get this fixed in an actual release?

There is quite a big chance it will be fixed, but not sure when, as I'm also working on a lot of other issues myself.

uzi18, 2019-11-24:
@dpomt please try to change gpio and save new settings and/or delete dht  task and configure it one more time
waiting for feedback

uzi18, 2019-11-24:
try this build: [normal_ESP8266_4M1M.bin.zip](https://github.com/letscontrolit/ESPEasy/files/3880870/normal_ESP8266_4M1M.bin.zip)

haldi4803, 2019-11-24:
Update here... 
I've updated to the "ESP_Easy_mega-20191119_normal_ESP8266_1M.bin". No Reset, kept settings. On one of my 3 ESP8266 DHT22 sensors and it works fine.
Maybe the issue can be circumvented if you flash the old version and if it works update to a newer one. Or it just works with this version.
It's running fine now for 1day 12 hours.

BUT Same as on the old version if i press the reset button, OR use software reset via Web Interface i get NaN until i plug the Power. Not sure if this is a hardware specific issue of [this chines board](url=https://www.ebay.ch/itm/AM2302-DHT22-ESP8266-Temperature-Humidity-Sensor-Wifi-Wireless-Module-ESP-01-01S/273598617288).

uzi18, 2019-11-26:
have tested on my own with DHT22 and after reboot from website have no NaN values

TD-er, 2019-11-26:
> BUT Same as on the old version if i press the reset button, OR use software reset via Web Interface i get NaN until i replug the Power.

And what if you just press submit on the task configuration page?
This does trigger a new init of the plugin.
Maybe the sensor is brought into some undefined state it can only get out of by power cycle, or more time to initialize?

haldi4803, 2019-11-27:
Nope, pressing Submit doesn't help.
Only Powercirlce helps. Is Probably a Hardware issue and not Software related.

Roos-AID, 2019-11-27:
@uzi18 

> try this build: normal_ESP8266_4M1M.bin.zip

After loading this build, the ESP is dead. It cannot connect to Wifi anymore.   I had to flash blank_1MB.bin and an older version to get the chip up and running again. 

ESP board 
ESP Chip ID: | 2497917 (0x261D7D)
ESP Chip Frequency: | 80 MHz
ESP Board Name: | PLATFORMIO_ESP01_1M

With this ESP01S board and the DHT22 as mentioned here https://www.ebay.ch/itm/AM2302-DHT22-ESP8266-Temperature-Humidity-Sensor-Wifi-Wireless-Module-ESP-01-01S/273598617288 
the problem of NaN after Warm Boot remains with all latest firmwares. 

Only a Power off-on cycle makes it working.  Pressing the reset button on the DHT22 board does not help. 

I strongly believe this is a hardware problem with this DHT22 module. 

@uzi18  if you want I can mail the ESP01 and DHT22 board to you for your testing.  I have multiple on stock (Netherlands).

Regards, Rob

uzi18, 2019-11-27:
@Roos-AID please contact me via https://gitter.im/uzi18

Roos-AID, 2019-12-05:
I have been debugging this issue and found something interesting. 

The problem is with the GPIO 2 line on the ESP01. 

When a  (cold or warm) boot is performed, then the GPIO 2 line gives exactly the same data as the TX line for the first 36 msec.  After this the GPIO 2 line is polling the DHT22 with a 2mS low after which the DHT22 correctly responds.  
The strange thing is, that after a cold boot, the DHT22 responds correctly to the 2mS low signal. After a warmboot, the DHT22 is with the TX data set in a strange state and cannot be reset other than with a power off-on. 

So, I decided to use the GPIO0 line available on the ESP01.   This does not get the TX data when booting. 
When we use GPIO0 on the ESP01 the DHT22 gives the correct data and no NaN after warmboot. 

I hope above is clear. I can make some pics from my logic analyser show this.

[5 more replies]

closed 2020-09-09 by TD-er