letscontrolit/ESPEasy #2586: SGP 30 Adafruit I2C dont work!

joe-1, 2019-08-30:
Hello and good Day,
That,s my second attempt.

Adafruit have a new lib for the SGP 30, its the Libnumber 1.05.
https://github.com/adafruit/Adafruit_SGP30
Lady ADA say , its for the new SGP 30 and its clear the Problems with the I2C Bus.
Your Libnumber is ESPEasy/lib/Adafruit_SGP30-1.0.0.13/
I work with ESP_Easy_mega-20190830_test_ESP8266_4M_VCC.bin.
I have one BME 280 and one SGP 30 on Bord, and the BME 280 works well.
I am using an ESP 8266 E.
I work with an other Bord  only with one SGP 30.
It is the same problem.
I check the I2C Bus and found under 0x58 nothing, only I2C Addresses in use.
So i think its only the problem from the Lib.
[image]
The Device SGP 30 have TVOC the same time 0 and CO2 400.
[image]

Thank you
Joachim Vocke

TD-er, 2019-08-30:
If a device cannot be found via I2C scan, then it is a connection issue.
But it does show there is one found with address 0x58, so the device is connected properly.
The missing string in the description is not related to the used library, but something missing in ESPeasy.

I just did a quick look at the code we currently have and I think this library and/or plugin does need to be changed for sure.
It is calling the `begin()` function of I2C and **no library should call begin** on its own on a shared bus.
I have seen this often and it is really something that should be changed in a lot of other libraries too..

It is also done wrong in the new version of the library.
While we're at it, we might as well upgrade the used library.

marsic-84, 2019-11-06:
I have the same problem. All time TVOC: 0 and eCO2: 400 
When it will be fixed?

TD-er, 2019-11-07:
> I have the same problem. All time TVOC: 0 and eCO2: 400
> When it will be fixed?

Well, as I don't have the sensor myself, I first have to get one (and some time) in order to fix it.
So as of now, there is no planned time this will be fixed.

marsic-84, 2019-12-06:
Hi. How are things going?)  Has the sensor arrived?

TD-er, 2019-12-06:
Nope, apparently it was send without (working) tracking number, so have to wait. :(

marsic-84, 2019-12-31:
Happy New Year to all))) Any news about the sensor?

TD-er, 2019-12-31:
I got a few times already the notification from the mail app that they would deliver it "today".
One delivery was cancelled, so I did re-order it and that one is also stuck in the delivery for a while now.

iboolka, 2020-01-25:
Hello! Any news about the sensor?

TD-er, 2020-01-25:
It has been delivered :)
But I am also moving to a new house, so it has been a while since the last thing I programmed. (way too long if you ask me)

panecki, 2020-02-20:
Still not working :) Any news? :)

TD-er, 2020-02-20:
Have been quite busy last weeks (moved to a new house), but I promise I will have a look at it this week.
Just brought my daughter to her grandmother for a 3 day sleep-over so that's lots of nerd time (I hope)

TD-er, 2020-02-22:
I've now pulled out this sensor to look into it. (took a while to find it back, so I now have a rather clean desk)

I've ordered [this one from Ali Express](https://nl.aliexpress.com/item/4000004614708.html?spm=a2g0s.9042311.0.0.27424c4d8O3eyH) and I'm looking at the [specs](https://www.sensirion.com/en/environmental-sensors/gas-sensors/multi-pixel-gas-sensors/)
I notice it is a 1v8 one, so I may need to add some level converters and create some voltage regulator first.
Not entirely sure what the other components on the board do, so better be careful then.

[image]

TD-er, 2020-02-22:
Ah, the SOT23 package with "56k5" on it is a [1V8 voltage regulator](https://lcsc.com/product-detail/Low-Dropout-Regulators-LDO_SC6206B-65K5-1-8V_C88294.html) so that should make it possible to power it from 3V - 5V.
Will check the other component, but that's probably a level switcher and the big one with "103" on it, looks like a resistor array of 10k.

TD-er, 2020-02-22:
OK, initial results.
The reported feature set of the chip (at least my chip) is 0x22, while the AdaFruit library expects 0x20.

But apart from that my sensor does not show any values other than 0.0 for TVOC and 400 for eCO2.

So it does seem to be a different sensor from what Adafruit used to create the library.
Also the datasheets on the Senserion page are quite horrible in describing how to address these sensors and what their capabilities are (based on feature set values)

TD-er, 2020-02-22:
I'll leave the sensor running for a while and will log the values to see if it ever changes.
According to the [discussion here](https://forums.adafruit.com/viewtopic.php?f=19&t=135180) there seems to be something like a "burn in time".

TD-er, 2020-02-22:
Hmm looking in the datasheet, it looks like the plugin as it was, would never have worked at all, as it is calling the IAQ init on each read.
It should also call the IAQMeasure every second, so it deserves quite some rewrite here.

closed 2020-02-22 by TD-er

marsic-84, 2020-03-25:
Hi. How's it going? Any luck?

TD-er, 2020-03-25:
> Hi. How's it going? Any luck?

The fixed plugin was already merged about a month ago.

marsic-84, 2020-03-26:
Wow cool. Go for a test))
... yep, everything works ...thanks a lot))

marsic-84, 2020-03-26:
it only send TVOC data to MQTT broker, but no eCO2 
is it a bug or there's some additional settings to setup?

TD-er, 2020-03-26:
> it only send TVOC data to MQTT broker, but no eCO2
> is it a bug or there's some additional settings to setup?

What controller?
Domoticz?

Domoticz is a bit strange that it does need to have the values in a very specific order for the kind of device you selected in Domoticz.
The SDS011 has similar issues.

If it is also only outputting the first value on other controllers, then it is a bug for sure.

marsic-84, 2020-03-26:
No. I use Openhab. But It does not send eCO2 data even to MQTT broker, i can not see even it's topic.  I use MQTT.fx to monitor all MQTT data.

TD-er, 2020-03-26:
Then that must be a bug I guess.
I will have a look

TD-er, 2020-03-26:
Found it.

```c++
      Device[deviceCount].VType              = SENSOR_TYPE_SINGLE;
```
It should be changed into `SENSOR_TYPE_DUAL`

beicnet, 2021-09-30:
@TD-er The problems are starting when you wish to add the SGP30 sensor from AliExpress as a third or fourth i2c device!
(just received two of them and ohhh boy), you just try it with multiple i2c device.

tonhuisman, 2021-09-30:
Most likely all these sensors have pull-up resistors installed, and having 4 of these in parallel, will divide the resistance by 4, and that is probably too low.
Also interesting to know is: what is the length of the wiring to the sensors?

TD-er, 2021-09-30:
Theory of the pull-up resistors sounds plausible.

beicnet, 2021-09-30:
Yes, some sensor modules have 10k and some 4.7k pull-up resistors, the cables are ribbon 10cm with JST XH-2.54 connectors.

So, @tonhuisman , @TD-er , any solution or idea (except for desoldering the resistor array)?

TD-er, 2021-09-30:
If you place several of those parallel, you get a lower netto resistance.
For example 4 parallel with 4k7 is roughly 1k2 effectively.
I can imagine that may be too strong for some chips to pull it down.

beicnet, 2021-09-30:
Yeah, and imagine mixed with a 10k resistors in parallel too.

Any quick fix?

TD-er, 2021-09-30:
Remove the pull-up resistors from the one you try to add now.
It was working with the other modules, so this one is too much.
Either because this one may be using resistors with a lower value than the others, or it is just the drop of water that floods the bucket (to translate a Dutch expression literally :) )

beicnet, 2021-09-30:
I didn't want to mess with the resistor arrays, so, I removed the resistors from the cheapest sensor modules, BMP180 and BH1750, now there is only 2x 10k left, before was 2x 10k and 2x 4.7k, now it's everything running just fine.

Thank you for the help you gave me! :+1: :1st_place_medal:   (entire day was eating s... whit this). :hankey:  :tada:

TD-er, 2021-10-01:
The effective pull up you had was 1.6k, with 2x 4k7 and 2x 10k.
So good to know that that's too low :)
The 2x10k + 4k7 was effectively 2k4, which was still working (maybe already a bit low?)

beicnet, 2021-10-01:
Yeah, it's on the lower limit, but I just realized that the resistor array can't be removed due level shifting.

PR #2895 (2020-02-22): Fix SGP30 plugin (#2586)
Fixes: #2586 
Update of AdaFruit library for the SGP30 to accept newer sensor revisions.

Also made the plugin work, as it apparently was never tested before.