letscontrolit/ESPEasy #2587: _IRext_ESP8266_4M not compiled to .bin anymore?

ghtester, 2019-08-30:
In recent official builds only the _minimal_IRext_ESP8266_1M.bin can be found. Could it be possible to add the 4M version again please? Thanks a lot for your great work!

TD-er, 2019-08-30:
I will check the build scripts to see if some changes in the platformio.ini are also applied in the build scripts.

jimmys01, 2019-08-31:
There is no space for IR and NORMAL build. Thats why we have changed it to IR and MINIMAL build.
Unfortunately you will have to compile your own build with the plugins of your liking, as I have did as well.

TD-er, 2019-08-31:
@jimmys01 
The minimal 4M build is not present in the before_deploy script, so I will add it there.

I can also look into the requirements for the 2/2M builds

PR #2589 (2019-09-02): [build] Add Minimal IRext 4M build
Fixes: #2587
commit 64d714a4 (2019-09-02): [build] Add Minimal IRext 4M build

Fixes: #2587
closed 2019-09-02 by TD-er