letscontrolit/ESPEasy #2593: Try to make a own build with platformio gives contineusly a error

henkiejan1, 2019-09-01:
Hello all,

A few time ago i build with succes a lot of esp easy builds with platformio. Now i like to try the TTN test controller. I can´t find nowhere where i can find a prebuild or something. So i try platformio. But get contineusly errors like:

`Compiling .pio/build/dev_ESP8266_4M/src/ESPEasy.ino.cpp.o
xtensa-lx106-elf-g++: error: src/ESPEasy.ino.cpp: No such file or directory
xtensa-lx106-elf-g++: fatal error: no input files
compilation terminated.
*** [.pio/build/dev_ESP8266_4M/src/ESPEasy.ino.cpp.o] Error 1
========================= [FAILED] Took 56.26 seconds =========================`

What goes wrong?

TD-er, 2019-09-01:
If you change something to the platformio.ini file, you will notice there is a background process running which temporary creates the file ESPeasy.ino.cpp
This background process also deletes this file if it is done.

If you started a build when this background process was still active, then the file will be deleted as soon as your build process is ready to compile it.

So it is best to save the platformio.ini file and wait for this ESPEasy.ino.cpp file to be deleted again and then start running your build.
You can also try running it the second time if you like.

By the way, I think this controller is included in the nightly builds named "custom".
I still have USES_C018 set in the pre_extra_script.py file.

fluppie, 2019-09-14:
I have the same issue, I use Github Desktop (even recloned the project) on Windows 10. Both errors on Atom and Visual Studio Code.
```

Processing normal_core_252_ESP8266_4M1M (platform: espressif8266@2.2.2; board: esp12e; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 2.2.2 > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: toolchain-xtensa 2.40802.190218 (4.8.2), framework-arduinoespressif8266 2.20502.0 (2.5.2), tool-esptool 1.413.0 (4.13), tool-esptoolpy 1.20600.0 (2.6.0)
UnicodeEncodeError: 'charmap' codec can't encode characters in position 1653755-1653756: character maps to <undefined>:
  File "C:\Users\Philippe\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 126:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Philippe\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Philippe\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Philippe\.platformio\platforms\espressif8266@2.2.2\builder\main.py", line 212:
    target_elf = env.BuildProgram()
  File "C:\Users\Philippe\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Philippe\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 110:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\Philippe\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Philippe\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 288:
    env.ConvertInoToCpp()
  File "C:\Users\Philippe\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Philippe\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 198:
    out_file = c.convert(ino_nodes)
  File "C:\Users\Philippe\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 57:
    return self.process(contents)
  File "C:\Users\Philippe\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 80:
    assert self._gcc_preprocess(contents, out_file)
  File "C:\Users\Philippe\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 90:
    fp.write(contents)
  File "C:\Users\Philippe\AppData\Local\Programs\Python\Python37\lib\encodings\cp1252.py", line 19:
    return codecs.charmap_encode(input,self.errors,encoding_table)[0]
=============================================================================================== [FAILED] Took 1.05 seconds ===============================================================================================

```

TD-er, 2019-09-14:
Come to think of it, I also had something similar once.
Can you check the platformio version?
And maybe remove PlatformIO, remove the directory `.platformio` and restart your PC (or first just the editor) and install platformIO again.

It does seem to be a [platformIO issue](https://github.com/platformio/platform-espressif8266).

fluppie, 2019-09-14:
I'm running Home 2.3.3 and Core 4.0.3
I'll try removing it en doing a fresh install. Maybe it mixes up Atom PlatformIO and VS Code?

TD-er, 2019-09-14:
Have you looked at the [PlatformIO issues](https://github.com/platformio/platform-espressif8266) ?
I do think it is not an ESPeasy issue, but an PlatformIO issue.

henkiejan1, 2019-09-24:
Sorry for my really late reply. But finally i have installed visual code on Linux Mint (what i preferer) and that works like a charm. Thank you all for your support / help. I think indeed it´s a Platformio issue. Atom works ok in safe mode.

TD-er, 2019-09-24:
So this issue can be closed I guess?

closed 2019-09-24 by TD-er