letscontrolit/ESPEasy #2600: Dummy sensor issue

mackowiakp, 2019-09-05:
Build 20190827 normal, core 2.5.2.
I want to set dummy sensor as temp/hum date type and click on Submit. But it returns to default data type "Single". So I try with type "dual". But the result is the same. That means after clicking "Submit" it still returns to data type "Single". In my meaning it is issue.
Of course as workaround I can use two dummy devices and it works properly.

TD-er, 2019-09-05:
Was this dummy plugin present in the settings from a previous build, or did you add is as new plugin using only this build?

mackowiakp, 2019-09-05:
I have not used dummy sensor before at all. For the first time I try to use it today. Just from medu of build 20190827 I try to setup one.

TD-er, 2019-09-05:
OK thanks. I will look at it after diner (I know, late diner here)

It may be related to some changes I made last week to split the webserver code, so if that got broken, some other plugins may be affected too.

mackowiakp, 2019-09-05:
Enjoy your dinner  !!!

uzi18, 2019-09-05:
@TD-er leave it to me ;)

uzi18, 2019-09-05:
@mackowiakp @TD-er can not reproduce

mackowiakp, 2019-09-05:
Maybe it is configuration related problem. My configuration is:
- 7 segment LED display
- DS18b20
- generic Uptime
- two relays
- ADC
- blinking LED
- hardware press-on button for swiching display content
- buzzer

uzi18, 2019-09-05:
ok got it ...

uzi18, 2019-09-05:
@TD-er got it

uzi18, 2019-09-05:
@TD-er #2601 is ready

uzi18, 2019-09-05:
@mackowiakp 
[dev_ESP8266_4M1M.bin.zip](https://github.com/letscontrolit/ESPEasy/files/3581109/dev_ESP8266_4M1M.bin.zip)

mackowiakp, 2019-09-06:
Still problem. I set dummy sensor as temp/baro, assign names and so on and click on Submit. Device reboot itself. After boot, the situation is as on attached screenshots. All the rest working properly. 

[image]

[image]

mackowiakp, 2019-09-06:
Additional info
Dummy sensor was create at first position in place of unnecessary longer device. So I revert to build 2190827 and quad dummy sensor exist. So I try to setup it the way I want click on Submit and WEB interface shows me such error:

> FS : Error while reading/writing config.dat in 765

uzi18, 2019-09-06:
please test on attached file and show me sysinfo page on it

uzi18, 2019-09-06:
It looks like you have problem with flash, so you should erase whole and flash blank bin, next flash test image

waiting for sysinfo data

mackowiakp, 2019-09-06:
The test was done with attached custom FW. Ok, re-revert to custom FW. Below json output

> {"System":{
"Build":20103,
"Git Build":"",
"System Libraries":"ESP82xx Core 2.6.0-dev, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support",
"Plugins":71,
"Plugin description":" [Normal] [Testing] [Development]",
"Local Time":"2019-09-06 08:21:26",
"Unit Number":1,
"Unit Name":"Salon",
"Uptime":1,
"Last Boot Cause":"Cold boot",
"Reset Reason":"Software/System restart",
"Load":35.10,
"Load LC":1663,
"CPU Eco Mode":"false",
"Heap Max Free Block":15016,
"Heap Fragmentation":9,
"Free RAM":16568
},
"WiFi":{
"Hostname":"Salon-1",
"IP Config":"Static",
"IP Address":"192.168.0.29",
"IP Subnet":"255.255.255.0",
"Gateway":"192.168.0.1",
"STA MAC":"84:F3:EB:B7:45:42",
"DNS 1":"192.168.0.201",
"DNS 1":"192.168.0.201",
"SSID":"maciek24",
"BSSID":"74:DA:38:72:B0:D9",
"Channel":1,
"Connected msec":62103,
"Last Disconnect Reason":1,
"Last Disconnect Reason str":"(1) Unspecified",
"Number Reconnects":0,
"Force WiFi B/G":"false",
"Restart WiFi Lost Conn":"false",
"Force WiFi No Sleep":"false",
"Periodical send Gratuitous ARP":"false",
"Connection Failure Threshold":0,
"RSSI":-65
},
"Sensors":[
{
"DataAcquisition": [
{"Controller":1,
"IDX":0,
"Enabled":"false"
},
{"Controller":2,
"IDX":0,
"Enabled":"false"
},
{"Controller":3,
"IDX":0,
"Enabled":"false"
}],
"TaskInterval":60,
"Type":"Display - 7-segment display",
"TaskName":"Temp+Time",
"TaskDeviceNumber":73,
"TaskEnabled":"true",
"TaskNumber":2
},
{
"TaskValues": [
{"ValueNumber":1,
"Name":"Temperatura-Salon",
"NrDecimals":2,
"Value":22.25
}],
"DataAcquisition": [
{"Controller":1,
"IDX":106,
"Enabled":"true"
},
{"Controller":2,
"IDX":0,
"Enabled":"false"
},
{"Controller":3,
"IDX":0,
"Enabled":"false"
}],
"TaskInterval":64,
"Type":"Environment - DS18b20",
"TaskName":"Temp-in",
"TaskDeviceNumber":4,
"TaskEnabled":"true",
"TaskNumber":3
},
{
"TaskValues": [
{"ValueNumber":1,
"Name":"Włączenie_display_temp",
"NrDecimals":0,
"Value":0
}],
"DataAcquisition": [
{"Controller":1,
"IDX":0,
"Enabled":"false"
},
{"Controller":2,
"IDX":0,
"Enabled":"false"
},
{"Controller":3,
"IDX":0,
"Enabled":"false"
}],
"TaskInterval":0,
"Type":"Switch input - Switch",
"TaskName":"Temp",
"TaskDeviceNumber":1,
"TaskEnabled":"true",
"TaskNumber":4
},
{
"TaskValues": [
{"ValueNumber":1,
"Name":"LED_on_off",
"NrDecimals":0,
"Value":0
}],
"DataAcquisition": [
{"Controller":1,
"IDX":0,
"Enabled":"false"
},
{"Controller":2,
"IDX":0,
"Enabled":"false"
},
{"Controller":3,
"IDX":0,
"Enabled":"false"
}],
"TaskInterval":0,
"Type":"Switch input - Switch",
"TaskName":"LED_on_off",
"TaskDeviceNumber":1,
"TaskEnabled":"true",
"TaskNumber":5
},
{
"TaskValues": [
{"ValueNumber":1,
"Name":"Test-czas-pracy",
"NrDecimals":0,
"Value":0
}],
"DataAcquisition": [
{"Controller":1,
"IDX":100,
"Enabled":"true"
},
{"Controller":2,
"IDX":0,
"Enabled":"false"
},
{"Controller":3,
"IDX":0,
"Enabled":"false"
}],
"TaskInterval":141,
"Type":"Generic - System Info",
"TaskName":"Test-czas-pracy",
"TaskDeviceNumber":26,
"TaskEnabled":"true",
"TaskNumber":6
},
{
"TaskValues": [
{"ValueNumber":1,
"Name":"VOC-Salon",
"NrDecimals":0,
"Value":163
}],
"DataAcquisition": [
{"Controller":1,
"IDX":122,
"Enabled":"true"
},
{"Controller":2,
"IDX":0,
"Enabled":"false"
},
{"Controller":3,
"IDX":0,
"Enabled":"false"
}],
"TaskInterval":93,
"Type":"Analog input - internal",
"TaskName":"VOC-Salon",
"TaskDeviceNumber":2,
"TaskEnabled":"true",
"TaskNumber":7
},
{
"TaskValues": [
{"ValueNumber":1,
"Name":"Przycisk",
"NrDecimals":0,
"Value":0
}],
"DataAcquisition": [
{"Controller":1,
"IDX":0,
"Enabled":"false"
},
{"Controller":2,
"IDX":0,
"Enabled":"false"
},
{"Controller":3,
"IDX":0,
"Enabled":"false"
}],
"TaskInterval":0,
"Type":"Switch input - Switch",
"TaskName":"Przycisk",
"TaskDeviceNumber":1,
"TaskEnabled":"true",
"TaskNumber":8
}
],
"TTL":60000
}

uzi18, 2019-09-06:
it could be related to flash or core, what is your flash ID?

mackowiakp, 2019-09-06:
The sysinfo You have in post above in json format. But below part of output from Main menu:

> 

Flash Chip ID: | Vendor: 0x20 Device: 0x4016
-- | --
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash IDE Speed: | 40 MHz
Flash IDE Mode: | DOUT
Flash Writes: | 0 daily / 0 boot
Sketch Size: | 967 kB (2104 kB free)
SPIFFS Size: | 934 kB (839 kB free)

Whats concerning erase flash and flash custom FW again it is a little bit complicated for me. 
The device is installed at a height of 3 m and I need to take a ladder. And while climbing the ladder, hold the laptop at the same time. And I have a fear of space, which is a very big problem for me. Is there an easier way to do the test?

uzi18, 2019-09-06:
thanks, so turn on debug log and check if you get more save errors in log when changing options in dummy?
core 2.6.0-dev is not well tested, but some problems are resolved there, we use it now with dev_x build to test it
I'm testing now one more time every option and save/submit works as expected - so it could be hardware related.

my sysinfo:

ESP Easy | Information |
 -----|-----|
Build:⋄|20103  - Mega|
System Libraries:⋄|ESP82xx Core 2.6.0-dev, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support|
Git Build:⋄||
Plugins:⋄|71 [Normal] [Testing] [Development]|
Build Time:⋄|Sep  5 2019 20:20:53|
Binary Filename:⋄|Self built!|

Flash Chip ID: | Vendor: 0xE0 Device: 0x4016
-- | --
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash IDE Speed: | 40 MHz
Flash IDE Mode: | DIO
Flash Writes: | 33 daily / 33 boot
Sketch Size: | 967 kB (2104 kB free)
SPIFFS Size: | 934 kB (844 kB free)

mackowiakp, 2019-09-06:
OK. Its good idea. So I have to buy nodeMCU module because I have not spare one at this moment. Just order today and probably tomorrow will have it.

mackowiakp, 2019-09-06:
Sorry. Tomorrow is Saturday. So new unit I will have in Monday.

mackowiakp, 2019-09-06:
@uzi18 
Below my sysinfo and mode logs (WEB log level set to debug dev)

Build:⋄ | 20103 - Mega
-- | --
System Libraries:⋄ | ESP82xx Core 2.6.0-dev, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ |  
Plugins:⋄ | 71 [Normal] [Testing] [Development]
Build Time:⋄ | Sep 5 2019 20:20:53
Binary Filename:⋄ | Self built!

Flash Chip ID: | Vendor: 0x20 Device: 0x4016
-- | --
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash IDE Speed: | 40 MHz
Flash IDE Mode: | DOUT
Flash Writes: | 43 daily / 43 boot
Sketch Size: | 967 kB (2104 kB free)
SPIFFS Size: | 934 kB (839 kB free)
Page size: | 256
Block size: | 8192
Number of blocks: | 116
Maximum open files: | 5
Maximum path length: | 32

LOGS

> 1644144: FS : Error while reading/writing config.dat in 871
1654104: SaveToFile: free stack: 3424
1654109: FS : Error while reading/writing config.dat in 774
1654112: SaveToFile: free stack: 3360
1654175: FILE : Saved config.dat
1654175: SaveToFile: free stack after: 3360
1654186: FS : Error while reading/writing config.dat in 871
1658391: WD : Uptime 28 ConnectFailures 0 FreeMem 16432 WiFiStatus 3
1663784: SaveToFile: free stack: 3104
1663789: FS : Error while reading/writing config.dat in 774
1663794: SaveToFile: free stack: 3376
1663798: FS : Error while reading/writing config.dat in 774
1663801: SaveToFile: free stack: 3312
1663866: FILE : Saved config.dat
1663867: SaveToFile: free stack after: 3312
1663879: FS : Error while reading/writing config.dat in 871
1664329: EVENT: Clock#Time=Fri,12:52
1664760: Command: delay
1664815: ACT : 7dst,12,52
1664817: 7DGT : Show Time=12:52:0
1664824: ACT : delay,50
1664835: Command: delay
1664887: ACT : GPIO,14,0
1664890: SW : GPIO 14 Set to 0
1664894: ACT : Let,1,1
1664905: Command: let
1667691: DS: SP: 7d,1,4b,46,7f,ff,c,10,3c,OK,7
1667700: DS : Temperature: 23.81 (28-ff-f6-ba-33-18-1-6)
1667704: EVENT: Temp-in#Temperatura-Salon=23.81
1667820: ACT : Let,3,23.81
1667833: Command: let
1667983: Domoticz: Sensortype: 1 idx: 106 values: 23.81
1671108: HTTP: event,LED_off
1671201: EVENT: LED_off
1671316: ACT : GPIO,15,0
1671320: SW : GPIO 15 Set to 0
1672633: Command: delay
1672687: ACT : GPIO,0,0
1672690: SW : GPIO 0 Set to 0
1672693: ACT : Let,1,1
1672704: Command: let
1672710: ACT : timerSet,4,420
1672722: Command: timerset
1672725: ACT : timerSet,7,1800
1672737: Command: timerset
1677497: HTTP: event,set_preassure=1028
1677529: EVENT: set_preassure=1028
1677612: ACT : Let,4,1028
1677624: Command: let
1677629: ACT : timerSet,7,2000
1677640: Command: timerset
1677643: ACT : Let,8,0
1677655: Command: let
1678647: ADC : Analog value: 878 = 878.000
1678653: EVENT: VOC-Salon#VOC-Salon=146.00
1678764: ACT : Let,5,146.00
1678777: Command: let
1678967: Domoticz: Sensortype: 1 idx: 122 values: 146
1688388: WD : Uptime 28 ConnectFailures 0 FreeMem 16432 WiFiStatus 3

uzi18, 2019-09-06:
@mackowiakp 
[dev_ESP8266_4M1M_2.4.2.bin.zip](https://github.com/letscontrolit/ESPEasy/files/3583709/dev_ESP8266_4M1M_2.4.2.bin.zip)

mackowiakp, 2019-09-06:
Still problem. When I try to define dummy sensor - error FS : Error while reading/writing config.dat in 774

TD-er, 2019-09-06:
Does this also happen with changing settings of other plugins?
Can you remotely reboot the node? (power cycle)
N.B. There is a chance it may not reboot into the same state as it is now, but enter a reboot loop in which it will disable any troubled plugin or controller.

I think the current settings may have been corrupted somehow.

mackowiakp, 2019-09-06:
OK, I dismount device from 3m high and now have it on the table. But can not clear flash. I use such command:

> esptool.py --port /dev/ttyUSB1 write_flash --flash_size 4MB --flash_mode dout 0x00000 ESPEasy_mega-20190827/bin/blank_4MB.bin

But there is not communication with nodeMCU. /dev/ttyUSB1 is present (that is port which present when I connect nodeMCU to PC. But maybe problem is with GPIO`s. It is because I use all D (0-8) and RX pin as GPIO input). I removed all rules form unit config - still no communication. So i press "Flash" button before connection to USB - still no communication. What I have to do? Everything - including nodeMCU  pins - are solder. Of course I can dis-solder particular pin, but which one?

TD-er, 2019-09-06:
You have to make sure GPIO-0 can be pulled to GND to boot into flash mode.

mackowiakp, 2019-09-06:
> Does this also happen with changing settings of other plugins?
Can you remotely reboot the node? (power cycle)

No. If I try to change already existing and working device - I can do it without problems
As I wrote, I dismounted unit so I have it on the table.

TD-er, 2019-09-06:
You could try to create a new dummy plugin on another position in the device list.
Then it will use a different part of the settings file.

Also make sure you have enough free space on the file system.
SPIFFS does need at least 2 free blocks and one block does mean 16 (??) consecutive pages of 256 bytes which are not linked to any file.
At boot, the garbage collector is run (not on all core versions I think) to try and free pages no longer allocated to a file.
This garbage collector is also called after deleting a file.

But you must have either a very fragmented filesystem, or nearly any free space to hit this.
It could happen on 1M modules which only have 128k of SPIFFS.

mackowiakp, 2019-09-06:
I created dummy sensor on another position and it was possible without problems !!! My curreny FW is the custom build dev_ESP8266_4M1M_2.4.2. Do You think that previous build, based on 2.6.0 is better to use? I mean about future upgrades.

mackowiakp, 2019-09-06:
Its crazy. After power reboot, dummy sensor in device list shows 4 position. Two of them with proper description i setup, other two without descriptions. But device configuration show data tyle "Single"

TD-er, 2019-09-06:
What build are you running? One with a fix?

mackowiakp, 2019-09-06:
Yes. @uzi18 sent custom build dev_ESP8266_4M1M_2.4.2. Will try previous one, based on 2.6.0. But for sure I have to clear flash.

uzi18, 2019-09-06:
you can try to format spiffs with factory reset function in advenced tab

mackowiakp, 2019-09-06:
OK. I made factory reset without WiFi and net config. But there is the same problem. After power reboot, dummy sensor in device list shows 4 position. Two of them with proper description i setup, other two without descriptions. But device configuration show data type "Single". But no errors during setup

TD-er, 2019-09-06:
And after you change the type to something other than "single" and save, is it then showing like you expect it to be?

uzi18, 2019-09-06:
change first all task to None function or. change plugin to. other

mackowiakp, 2019-09-06:
> And after you change the type to something other than "single" and save, is it then showing like you expect it to be?

Yes, until power on/off reset

> change first all task to None function or. change plugin to. other and then to dummy

Still the same. After power reboot, dummy sensor in device list shows 4 position. Two of them with proper description i setup, other two without descriptions. But device configuration show data type "Single". But no errors during setup

uzi18, 2019-09-06:
looks like flash is some kind broken, try to flash blank_4MB.bin and one more time my custom bin

mackowiakp, 2019-09-06:
Which one? 2.6.0 or 2.4.2?

mackowiakp, 2019-09-06:
But after factory reset I was able to setup dummy sensor on position which preciously shows error in cofig.dat file. Now I use 2.6.0 custom build and all setup goes without errors. Only that one wit "Single" described above.

mackowiakp, 2019-09-06:
Still can not communicate with nodeMCU. During boot I connected D3 (GPIO 0) to GND but can not transmit anything.

TD-er, 2019-09-06:
If you're using a NodeMCU, then it should have a "flash" button, which is safer to use.
You should not connect a GPIO pin directly to GND or Vcc, but use a resistor.
This to protect the GPIO pin if you try to use it as an output pin and force high or low while pulling it strongly to GND or Vcc.

I assume GPIO2 is pulled high and GPIO-15 pulled low, or else it would not boot to ESPeasy.
With those pulled to the right state, it should boot into flash mode if you reset it.
Make sure you also reset the unit while pressing the flash button.

mackowiakp, 2019-09-06:
I try to use flash button. Result the same.

mackowiakp, 2019-09-06:
I think that there is something wrong with this paticular nodeMCU module I have. Monday, Tuesday I will have new one and then will try, Generally is working but......

mackowiakp, 2019-09-09:
Brand new device from shop. As the first I install 4M blank img. Then test buil @uzi18 dev_ESP8266_4M1M. The same error. Dummy device defined as dual data type (Temp/Baro), after power reboot is still quad data type. And data type in device config is single. The only thing I defined so fat is static IP config, NTP server and this dummy sensor. below unit details and screenshots

ESP Board
--
ESP Chip ID: | 14901938 (0xE362B2)
ESP Chip Frequency: | 80 MHz
ESP Board Name: | PLATFORMIO_ESP12E
Storage
Flash Chip ID: | Vendor: 0x68 Device: 0x4016
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash IDE Speed: | 40 MHz
Flash IDE Mode: | DOUT
Flash Writes: | 0 daily / 0 boot
Sketch Size: | 967 kB (2104 kB free)
SPIFFS Size: | 934 kB (844 kB free)
Page size: | 256
Block size: | 8192
Number of blocks: | 116
Maximum open files: | 5
Maximum path length: | 32

[image]

[image]

.

uzi18, 2019-09-09:
please find me directly on gitter.im/Lucjan

uzi18, 2019-09-09:
do you have config write errors?

mackowiakp, 2019-09-09:
No. Any kind of errors or warning was no raported.

xztraz, 2020-05-29:
This is still a problem. can't use more than one dummy device if you want different types.

TD-er, 2020-05-29:
> This is still a problem. can't use more than one dummy device if you want different types.

Yep that's still an issue.

PR #3265 (2020-09-24): Use task specific valueCount and VType instead of shared in plugin
Lots of reported issues regarding strange issues when using multiple instances of the same plugin which allows to change its output variables and value count.

Possible affected plugins:

- P026 Sysinfo
- P033 Dummy
- P052 SenseAir
- P078 Eastron
- P082 GPS
- P085 AcuDC243

N.B. this PR is not yet finished.
Still need to have a look at usage of `sensorTypeHelper_loadOutputSelector` and related functions.

Fixes: #3184 
Fixes: #2675
Fixes: #2626
Fixes: #2600
Fixes: #3045 
Fixes: #3061
closed 2020-09-24 by TD-er